zephyrproject-rtos/zephyr #20712: nRF clock_control_on() is nonblocking

mbolivar, 2019-11-14:
**Describe the bug**

Per discussion in https://github.com/zephyrproject-rtos/zephyr/issues/20708, the non-async clock enable function `clock_control_on()` should block until the clock is ready, but the nRF clock driver doesn't.

**To Reproduce**

- call clock_control_on() on an nRF clock that isn't already enabled
- observe clock_start() in nrf_power_clock.c calls clock_async_start() directly with no callback
- without a callback, note clock_async_start() returns before the clock may be ready

**Expected behavior**

Clock status should be definitely enabled by the time clock_control_on() returns.

**Impact**

Unknown.

**Environment (please complete the following information):**
 
nRF platform, possibly others.

**Additional context**

Discussion in https://github.com/zephyrproject-rtos/zephyr/issues/20708 and https://github.com/zephyrproject-rtos/zephyr/pull/20710.

ioannisg, 2019-11-27:
@nordic-krch any updates on this?

rlubos, 2020-02-03:
I was about to open an issue when I've found this one.

The nRF radio driver is hit by the nonblocking behavior of `clock_control_on` - the radio may malfunction when used before HF clock is ready, yet the radio driver itself does not take any extra action to verify that the clock is running as `clock_control_on` is supposed to block until clock is ready (and it used to work like this until  [this commit](https://github.com/zephyrproject-rtos/zephyr/commit/6700f2f1940c645c5cd1eb23baad8bd0c8041b81)).

nordic-krch, 2020-02-04:
im working right now on clock_control driver to straighten this up (and start using onoff service)

PR #24334 (2020-07-20): drivers: clock_control: nrf: Use onoff service
Ported nRF clock control driver to use onoff service.
Clock_control API no longer tracks number of users but only enables or disables clock. It should not be used directly for clocks which have multiple user.

Clock control driver implements onoff service which can be used to track clock users. Clock control driver implements error detection mechanism which detects cases when direct API and onoff API is used for the same clock subsystem.

Nordic specific API (`nrf_clock_control.h`) provides function for getting onoff manager associated with given  [... cut]
closed 2020-07-20 by carlescufi