zephyrproject-rtos/zephyr #20724: Packed pointer warning in LL Controller

joerchan, 2019-11-15:
**Describe the bug**
GCC 9 throws new warnings for possible unaligned pointer issues. On some architectures this is not supported, or can be configured to not be supported. On ARM this would throw a Hard Fault Exception.

**To Reproduce**
Steps to reproduce the behavior:
Build BT Controller with GCC v9

**Impact**
This could end in HardFault or other forms of crash.

**Screenshots or console output**
```
      [129/199] Building C object 
[142/199] Building C object zephyr/subsys/bluetooth/contr...MakeFiles/subsys__bluetooth__controller.dir/hci/hci.c.obj
/home/thomsten/dev/zephyr/subsys/bluetooth/controller/hci/hci.c: In function 'le_conn_complete':
/home/thomsten/dev/zephyr/subsys/bluetooth/controller/hci/hci.c:2776:9: warning: converting a packed 'struct pdu_data' pointer (alignment 1) to a 'struct node_rx_cc' pointer (alignment 2) may result in an unaligned pointer value [-Waddress-of-packed-member]
 2776 |  struct node_rx_cc *node_rx = (void *)pdu_data;
      |         ^~~~~~~~~~
In file included from /home/thomsten/dev/zephyr/subsys/bluetooth/controller/hci/hci.c:29:
/home/thomsten/dev/zephyr/subsys/bluetooth/controller/./ll_sw/pdu.h:511:8: note: defined here
  511 | struct pdu_data {
      |        ^~~~~~~~
In file included from /home/thomsten/dev/zephyr/subsys/bluetooth/controller/hci/hci.c:32:
/home/thomsten/dev/zephyr/subsys/bluetooth/controller/./ll_sw/ull_conn_types.h:219:8: note: defined here
  219 | struct node_rx_cc {
      |        ^~~~~~~~~~
/home/thomsten/dev/zephyr/subsys/bluetooth/controller/hci/hci.c: In function 'le_conn_update_complete':
/home/thomsten/dev/zephyr/subsys/bluetooth/controller/hci/hci.c:2906:2: warning: converting a packed 'struct pdu_data' pointer (alignment 1) to a 'struct node_rx_cu' pointer (alignment 2) may result in an unaligned pointer value [-Waddress-of-packed-member]
 2906 |  cu = (void *)pdu_data;
      |  ^~
In file included from /home/thomsten/dev/zephyr/subsys/bluetooth/controller/hci/hci.c:29:
/home/thomsten/dev/zephyr/subsys/bluetooth/controller/./ll_sw/pdu.h:511:8: note: defined here
  511 | struct pdu_data {
      |        ^~~~~~~~
In file included from /home/thomsten/dev/zephyr/subsys/bluetooth/controller/hci/hci.c:32:
/home/thomsten/dev/zephyr/subsys/bluetooth/controller/./ll_sw/ull_conn_types.h:235:8: note: defined here
  235 | struct node_rx_cu {
      |        ^~~~~~~~~~
[177/199] Building C object zephyr/subsys/bluetooth/contr...eFiles/subsys__bluetooth__controller.dir/ll_sw/ctrl.c.obj
/home/thomsten/dev/zephyr/subsys/bluetooth/controller/ll_sw/ctrl.c: In function 'isr_rx_conn_pkt_ctrl_rej_conn_upd':
/home/thomsten/dev/zephyr/subsys/bluetooth/controller/ll_sw/ctrl.c:2175:2: warning: converting a packed 'struct pdu_data' pointer (alignment 1) to a 'struct radio_le_conn_update_cmplt' pointer (alignment 2) may result in an unaligned pointer value [-Waddress-of-packed-member]
 2175 |  cp = (void *)pdu_data_rx;
      |  ^~
In file included from /home/thomsten/dev/zephyr/subsys/bluetooth/controller/ll_sw/ctrl.c:42:
/home/thomsten/dev/zephyr/subsys/bluetooth/controller/ll_sw/pdu.h:511:8: note: defined here
  511 | struct pdu_data {
      |        ^~~~~~~~
In file included from /home/thomsten/dev/zephyr/subsys/bluetooth/controller/ll_sw/ctrl.c:44:
/home/thomsten/dev/zephyr/subsys/bluetooth/controller/ll_sw/ctrl.h:144:8: note: defined here
  144 | struct radio_le_conn_update_cmplt {
      |        ^~~~~~~~~~~~~~~~~~~~~~~~~~
```

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain GCC v9
 - Commit: 2.0.0 (or current master).

dleach02, 2019-11-15:
This one does seem to be problematic:

warning: converting a packed 'struct pdu_data' pointer (alignment 1) to a 'struct radio_le_conn_update_cmplt' pointer (alignment 2) may result in an unaligned pointer value [-Waddress-of-packed-member]
 2175 |  cp = (void *)pdu_data_rx;

Generally, Arm would throw an exception when trying to do unaligned access but in the case of a packed structure with pointers and such the compiler normally knows this and will use "relaxed" rules when generating code around these. The relaxed rules result in unaligned accesses being structured with multiple byte or word accesses (what ever is allowed). But if you are assigning a pointer of different types where one structure is packed and the other is not then the compiler won't know to use the relaxed rules.

cvinayak, 2019-11-18:
I have analyzed the implementation, the packed struct pdu_data is used as place holder of pointer to the message passed from controller's ISR contect to HCI thread context. The access to data by both the producing and consuming implementation is by use of unpacked structures. The current upstream implementation does not lead to any unaligned access in the unpacked structures.

That said, I am not getting compile time warnings in my dev environ:
- Archlinux distribution with gcc version 9.2.0 (Arch Repository)
- export ZEPHYR_TOOLCHAIN_VARIANT=gnuarmemb
- export GNUARMEMB_TOOLCHAIN_PATH=/usr

Is there something I need to do/add to build script to enable the reported warnings?

thomasstenersen, 2019-11-18:
@cvinayak, here's the full output against vanilla Zephyr master branch (6ba1fcec3fe9a527ccd2a956cd6860290a0c02eb).

Compiler is
```
$ arm-none-eabi-gcc --version
arm-none-eabi-gcc (Fedora 9.2.0-1.fc30) 9.2.0
Copyright (C) 2019 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

Building `zephyr/samples/bluetooth/central_hr`

```
-- west build: build configuration:
       source directory: /home/thomsten/dev2/zephyr/samples/bluetooth/central_hr
       build directory: /home/thomsten/dev2/zephyr/samples/bluetooth/central_hr/build (created)
       BOARD: nrf52_pca10040 (origin: command line)
-- west build: generating a build system
Zephyr version: 2.1.0
-- Found PythonInterp: /home/thomsten/.virtualenv/default/bin/python (found suitable version "3.7.3", minimum required is "3.4") 
-- Selected BOARD nrf52_pca10040
-- Found west: /home/thomsten/.virtualenv/default/bin/west (found suitable version "0.6.3", minimum required is "0.6.0")
-- Loading /home/thomsten/dev2/zephyr/boards/arm/nrf52_pca10040/nrf52_pca10040.dts as base
Devicetree configuration written to /home/thomsten/dev2/zephyr/samples/bluetooth/central_hr/build/zephyr/include/generated/generated_dts_board.conf
Parsing Kconfig tree in /home/thomsten/dev2/zephyr/Kconfig
Loaded configuration '/home/thomsten/dev2/zephyr/boards/arm/nrf52_pca10040/nrf52_pca10040_defconfig'
Merged configuration '/home/thomsten/dev2/zephyr/samples/bluetooth/central_hr/prj.conf'
Configuration saved to '/home/thomsten/dev2/zephyr/samples/bluetooth/central_hr/build/zephyr/.config'
-- The C compiler identification is GNU 9.2.0
-- The CXX compiler identification is unknown
-- The ASM compiler identification is GNU
-- Found assembler: /usr/bin/arm-none-eabi-gcc
-- Cache files will be written to: /home/thomsten/.cache/zephyr
-- Configuring done
-- Generating done
-- Build files have been written to: /home/thomsten/dev2/zephyr/samples/bluetooth/central_hr/build
-- west build: building application
[1/182] Preparing syscall dependency handling

[2/182] Generating include/generated/syscall_macros.h
[3/182] Generating misc/generated/syscalls_subdirs.trigger
[4/182] Generating include/generated/driver-validation.h
[5/182] Generating include/generated/kobj-types-enum.h, include/generated/otype-to-str.h
[6/182] Generating misc/generated/syscalls.json
[7/182] Generating include/generated/syscall_dispatch.c, include/generated/syscall_list.h
[8/182] Building C object zephyr/CMakeFiles/offsets.dir/arch/arm/core/offsets/offsets.c.obj
[9/182] Generating include/generated/offsets.h
[10/182] Building C object zephyr/CMakeFiles/zephyr.dir/lib/os/crc8_sw.c.obj
[11/182] Building C object zephyr/CMakeFiles/zephyr.dir/lib/os/crc32_sw.c.obj
[12/182] Building C object zephyr/CMakeFiles/zephyr.dir/lib/os/crc16_sw.c.obj
[13/182] Building C object zephyr/CMakeFiles/zephyr.dir/lib/os/crc7_sw.c.obj
[14/182] Building C object zephyr/CMakeFiles/zephyr.dir/lib/os/dec.c.obj
[15/182] Building C object zephyr/CMakeFiles/zephyr.dir/lib/os/hex.c.obj
[16/182] Building C object CMakeFiles/app.dir/src/main.c.obj
[17/182] Linking C static library app/libapp.a
[18/182] Building C object zephyr/CMakeFiles/zephyr.dir/lib/os/sem.c.obj
[19/182] Building C object zephyr/CMakeFiles/zephyr.dir/lib/os/fdtable.c.obj
[20/182] Building C object zephyr/CMakeFiles/zephyr.dir/lib/os/timeutil.c.obj
[21/182] Building C object zephyr/CMakeFiles/zephyr.dir/lib/os/thread_entry.c.obj
[22/182] Building C object zephyr/CMakeFiles/zephyr.dir/lib/os/rb.c.obj
[23/182] Building C object zephyr/CMakeFiles/zephyr.dir/lib/os/mempool.c.obj
[24/182] Building C object zephyr/CMakeFiles/zephyr.dir/lib/os/work_q.c.obj
[25/182] Building C object zephyr/CMakeFiles/zephyr.dir/ext/lib/crypto/tinycrypt/source/utils.c.obj
[26/182] Building C object zephyr/CMakeFiles/zephyr.dir/misc/generated/configs.c.obj
[27/182] Building C object zephyr/CMakeFiles/zephyr.dir/soc/arm/nordic_nrf/validate_base_addresses.c.obj
[28/182] Building C object zephyr/CMakeFiles/zephyr.dir/ext/lib/crypto/tinycrypt/source/ecc_dh.c.obj
[29/182] Building C object zephyr/CMakeFiles/zephyr.dir/lib/os/printk.c.obj
[30/182] Building C object zephyr/CMakeFiles/zephyr.dir/ext/lib/crypto/tinycrypt/source/aes_decrypt.c.obj
[31/182] Building C object zephyr/CMakeFiles/zephyr.dir/subsys/logging/log_list.c.obj
[... 208 lines cut ...]
[163/182] Building C object zephyr/kernel/CMakeFiles/kernel.dir/pipes.c.obj
[164/182] Building C object zephyr/kernel/CMakeFiles/kernel.dir/queue.c.obj
[165/182] Building C object zephyr/kernel/CMakeFiles/kernel.dir/stack.c.obj
[166/182] Building C object zephyr/kernel/CMakeFiles/kernel.dir/system_work_q.c.obj
[167/182] Building C object zephyr/kernel/CMakeFiles/kernel.dir/version.c.obj
[168/182] Building C object zephyr/kernel/CMakeFiles/kernel.dir/thread_abort.c.obj
[169/182] Building C object zephyr/kernel/CMakeFiles/kernel.dir/sched.c.obj
[170/182] Building C object zephyr/kernel/CMakeFiles/kernel.dir/work_q.c.obj
[171/182] Building C object zephyr/kernel/CMakeFiles/kernel.dir/smp.c.obj
[172/182] Building C object zephyr/kernel/CMakeFiles/kernel.dir/thread.c.obj
[173/182] Building C object zephyr/kernel/CMakeFiles/kernel.dir/timer.c.obj
[174/182] Building C object zephyr/kernel/CMakeFiles/kernel.dir/timeout.c.obj
[175/182] Building C object zephyr/kernel/CMakeFiles/kernel.dir/poll.c.obj
[... cut ...]
Memory region         Used Size  Region Size  %age Used
           FLASH:      127991 B       512 KB     24.41%
            SRAM:       21020 B        64 KB     32.07%
        IDT_LIST:         120 B         2 KB      5.86%
[178/182] Generating linker_pass_final.cmd
[179/182] Generating isr_tables.c
[180/182] Building C object zephyr/CMakeFiles/zephyr_final.dir/misc/empty_file.c.obj
[181/182] Building C object zephyr/CMakeFiles/zephyr_final.dir/isr_tables.c.obj
[182/182] Linking C executable zephyr/zephyr.elf
```

joerchan, 2019-11-18:
> I have analyzed the implementation, the packed struct pdu_data is used as place holder of pointer to the message passed from controller's ISR contect to HCI thread context. The access to data by [...]

@cvinayak 
Zephyr is not using your distributed GCC, it is using the one in zephyr-sdk. Thomas is using the GNUARMEMB toolchain option.

cvinayak, 2019-11-18:
> Build BT Controller with GCC v9

@thomasstenersen could you give me the commandline and environment variables you have set so that I try to reproduce what you get.

@joerchan see my earlier comment, I hardly use Zephyr SDK. I may be wrong, but attaching my ninja verbose for you perusal. I piped my GCC version and "ninja --verbose".

[build.log](https://github.com/zephyrproject-rtos/zephyr/files/3858419/build.log)

joerchan, 2019-11-18:
@cvinayak Didn't see that comment. You are using arm-none-eabi-gcc from the ARCH package manager? It might not be the same build that you get from ARMs webpage. You shoud probably get that one instead.

thomasstenersen, 2019-11-18:
@cvinayak, no special environment.

```
GNUARMEMB_TOOLCHAIN_PATH=/usr/bin/..
ZEPHYR_TOOLCHAIN_VARIANT=gnuarmemb
```

Attached [build.log](https://github.com/zephyrproject-rtos/zephyr/files/3858530/build.log)

@joerchan, I'm using the one built by Fedora. So not stock ARM build. Might be that the arch build does not enable `-Waddress-of-packed-member` by default?
https://gcc.gnu.org/onlinedocs/gcc/Warning-Options.html

joerchan, 2019-11-18:
@cvinayak  @thomasstenersen AFAIK ARM maintains ARM support for GCC only on their website, not for different package managers. So I **strongly** recommend you switch to that one instead.

carlescufi, 2020-02-14:
I tested this both with Zephyr SDK 0.11.1: 
```
$ ~/bin/zephyr-sdk/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc --version
arm-zephyr-eabi-gcc (crosstool-NG 1.24.0.37-3f461da-dirty) 9.2.0
Copyright (C) 2019 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```
And with GNU Arm Embedded 9.2.0, as downloaded from [here](https://developer.arm.com/tools-and-software/open-source-software/developer-tools/gnu-toolchain/gnu-rm/downloads):
```
$ ~/bin/gcc-arm-none-eabi-9-2019-q4-major/bin/arm-none-eabi-gcc --version
arm-none-eabi-gcc (GNU Tools for Arm Embedded Processors 9-2019-q4-major) 9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599]
Copyright (C) 2019 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```
Neither show any issues, and those are the officially supported toolchains for Arm.

closed 2020-02-14 by carlescufi