zephyrproject-rtos/zephyr #20726: arm: Specifying sp register in asm's clobber list is deprecated in GCC 9

pfalcon, 2019-11-15:
This is offshot from https://github.com/zephyrproject-rtos/zephyr/issues/20582.

As said there, GCC 9 (at least one from GNU Arm Embedded package) issues warning (turned error by CI builds) about usage of "sp" in `asm` statement register clobber list. E.g., here: https://github.com/zephyrproject-rtos/zephyr/blob/master/include/arch/arm/syscall.h#L57

It might have been a quick patch to remove it, but quick history look shows that it would be a reversal of a just month-old  patch: 684f5ec12e05efe27af1c754047f2ae582f1cbdb (https://github.com/zephyrproject-rtos/zephyr/pull/19924).

There's something else strange about that patch too: there's arch_syscall_invoke0, arch_syscall_invoke1, etc. But the patch patches only arch_syscall_invoke6, and doesn't say why so in the commit message (consequently the simplest explanation is that the other functions were simply overlooked).

pfalcon, 2019-11-15:
cc: @galak

pfalcon, 2019-11-15:
> There's something else strange about that patch too

.. Also, there's no "Fixes:" reference in that PR, so one can only guess what issue it really fixed (if any).

galak, 2019-11-15:
https://github.com/zephyrproject-rtos/zephyr/pull/20410

galak, 2019-11-19:
Resolved w/merge of #20410

closed 2019-11-19 by galak