zephyrproject-rtos/zephyr #20729: Coverage reporting hangs for C++ tests on X86 qemu

vanwinkeljan, 2019-11-15:
**Describe the bug**

When C++ related tests are executed with coverage reporting on X86 Qemu, Qemu will end up in endless loop when dumping coverage results to UART.

**To Reproduce**

Steps to reproduce the behavior:
1. Remove `platform_exclude: qemu_x86_coverage` from `tests/application_development/libcxx/testcase.yaml`
2. `sanitycheck -N -p qemu_x86_coverage -s tests/application_development/libcxx/misc.app_dev.libcxx`

**Expected behavior**

C++ tests can run with coverage enabled on Qemu X86 without hanging.

**Impact**

No coverage reports can be generated for C++ test cases

**Environment (please complete the following information):**
 - OS: Linux - Ubuntu 18.04.3
 - Toolchain: Zephyr SDK 10.03
 - Commit SHA: 86461ac559a

**Additional context**

Quick look with gdb reveals that the while loop in `gcov_coverage_dump` is an endless loop

https://github.com/zephyrproject-rtos/zephyr/blob/6d9af839074ee70da0897450a62463e58a7515f9/subsys/testsuite/coverage/coverage.c#L223

carlescufi, 2020-04-30:
@nashif any ideas about this one?

nashif, 2020-09-10:
well, we never got c++ code to work with coverage, Zephyr is all C and the intention of coverage is C code primarily, so closing. If someone has time, patches to support C++ code are welcome.

closed 2020-09-10 by nashif