zephyrproject-rtos/zephyr #20734: Are cooperative threads cooperative in SMP?

carlescufi, 2019-11-15:
When scheduling cooperative threads in an SMP system, does the scheduler take into account that if 2 separate cooperative threads run in 2 different cores at the same time then the contract that one expects to hold when using cooperative threads is broken?

CC @abrodkin

andyross, 2019-11-15:
The documented scheduling behavior of cooperative threads is identical in SMP: they will not yield the CPU even if another, higher priority thread is ready to run.  The implicit **locking** behavior of cooperative scheduling (that "no other higher priority thread will be run") is a promise that can't be kept in SMP.  The same is true for k_sched_lock(), where IIRC there is some documentation of this behavior.

carlescufi, 2019-11-15:
> The documented scheduling behavior of cooperative threads is identical in SMP: they will not yield the CPU even if another, higher priority thread is ready to run. The implicit **locking** behavior [...]

I'm unsure what you mean by "CPU": a single core or the whole CPU.
So you are saying that if I spawn two cooperative threads in an SMP system, they can actually run concurrently on two separate cores?

andyross, 2019-11-15:
Yes, they'll run on any core, possibly in parallel with other cooperative threads of higher priority (i.e. something you'll never see in a single CPU environment).

closed 2020-09-19 by github-actions[bot]

LingaoM, 2024-05-11:
Don’t you think this is a potential risk for cooperative threads?