zephyrproject-rtos/zephyr #20746: Bluetooth: Mesh: Friend node Adding another Friend Update

LingoMeng, 2019-11-16:
Two friend "update message are required in the `MESH/NODE/FRND/FN/BV-20-C [Adding another Friend Update – Friend Node]`, but I did not find such processing logic in the code. But it's strange that I noticed that I have passed the certification.

In the authentication process, it is required that the first `friend_update` message should be **MD = 1**,
Only in this case `enqueue_update(frnd, 1);`
I tracked the process: `enqueue_update(frnd, 1); <--- enqueue_friend_pdu;`

However `enqueue_friend_pdu` will not be called at this stage.

If I'm right, the problem is this function `bt_mesh_friend_sec_update`:
https://github.com/zephyrproject-rtos/zephyr/blob/3cc5bda2fa10b924c04dc58ff7055fa7bb683dcc/subsys/bluetooth/mesh/friend.c#L193-L210

If we set this flag` [frnd->sec_update]` before setting it, it is obvious that the last `friend_update` will be lost.

aescolar, 2019-11-16:
CC @trond-snekvik @jhedberg

LingoMeng, 2019-11-25:
+70516 ms
Verdict Description:     Send Friend Poll. Friend Sequence Number = 1

+70532 ms
Send Event:              HCI!HCI_LE_SET_ADVERTISING_DATA{
                           advertisingDataLength=0x15,
                           advertisingData=0x142A633A636838138218FDA9B2992B453E26792E49000000000000000000
                         }

+70532 ms
Receive Event:           HCI?HCI_LE_SET_ADVERTISING_DATA_COMPLETE_EVENT{
                           status=HCI_OK
                         }

+70532 ms
Send Event:              HCI!HCI_LE_SET_ADVERTISE_ENABLE{
                           advertisingEnable=HCI_LE_ADVERTISING_ENABLE
                         }

+70532 ms
Receive Event:           HCI?HCI_LE_SET_ADVERTISE_ENABLE_COMMAND_COMPLETE_EVENT{
                           status=HCI_OK
                         }

+70547 ms
Send Event:              HCI!HCI_LE_SET_ADVERTISE_ENABLE{
                           advertisingEnable=HCI_LE_ADVERTISING_DISABLE
                         }

+70547 ms
Receive Event:           HCI?HCI_LE_SET_ADVERTISE_ENABLE_COMMAND_COMPLETE_EVENT{
                           status=HCI_OK
                         }

+70563 ms
Verdict Description:     Wait for friend message.

+70563 ms
Verdict Description:     ***** Waiting for: MeshTransport Event

+70766 ms
Verdict Description:     Received Msg: MeshTransport EventTransport Message: 
    Op Code: [2 (0x02)] FriendUpdate
    KeyRefreshFlag: [0x01]
    IVUpdateFlag: [0x01]
    FlagReserved: [0x00]
    IVIndex: [3 (0x00000003)]
    MD: [0 (0x00)]transport header Transport Header: 
    TransOpCode: [0x02]
    SEG: [0x00]MeshNetworkData:: MeshNetwork Network Header: 
    NID: [0x22]
    IVI: [0x00]
    TTL: [0x00]
    CTL: [0x01]
    SEQ: [0x000108]
    SRC: [0x0033]
    DST: [0x0035]
    TransportPDU: [0x02030000000300]

+70766 ms
Verdict Description:     ApplicationLayer Poll receive payload empty.

+70782 ms
Verdict Description:     Wait for friend message.

+70782 ms
Verdict Description:     ***** Waiting for: MeshTransport Event

+70782 ms
Verdict Description:     ***** Waiting for: MeshTransport Event

+71078 ms
Verdict Description:     FriendUpdate Command params: keyRefreshFlag = 1, ivUpdateFlag =1, IV Index = 0x00000003,  MD = 0

+74078 ms
Verdict Description:     Send Friend Poll. Friend Sequence Number = 0

+74094 ms
Send Event:              HCI!HCI_LE_SET_ADVERTISING_DATA{
                           advertisingDataLength=0x15,
                           advertisingData=0x142A6311511E8ED9E79A0C80FCD27ED76E7450AF0B000000000000000000
                         }

+74094 ms
Receive Event:           HCI?HCI_LE_SET_ADVERTISING_DATA_COMPLETE_EVENT{
                           status=HCI_OK
                         }

+74094 ms
Send Event:              HCI!HCI_LE_SET_ADVERTISE_ENABLE{
                           advertisingEnable=HCI_LE_ADVERTISING_ENABLE
                         }

+74094 ms
Receive Event:           HCI?HCI_LE_SET_ADVERTISE_ENABLE_COMMAND_COMPLETE_EVENT{
                           status=HCI_OK
                         }

+74110 ms
Send Event:              HCI!HCI_LE_SET_ADVERTISE_ENABLE{
                           advertisingEnable=HCI_LE_ADVERTISING_DISABLE
                         }

+74110 ms
Receive Event:           HCI?HCI_LE_SET_ADVERTISE_ENABLE_COMMAND_COMPLETE_EVENT{
                           status=HCI_OK
                         }

+74125 ms
Verdict Description:     Wait for friend message.

+74125 ms
Verdict Description:     ***** Waiting for: MeshTransport Event

+75141 ms
Verdict Description:     Timeout to get response on Friend Poll message.

+75438 ms
Verdict Description:     Timeout!!!

+75438 ms
Inconclusive:            Failed to receive Friend Update message with IV Update flag set to 1.

+75438 ms
Verdict Description:     Failed to receive Friend Update message with IV Update flag set to 1.

+75438 ms
Verdict Description:     ***** Waiting for: MeshTransport Event

+76453 ms
Send Event:              HCI!HCI_LE_SET_SCAN_ENABLE{
                           leScanEnable=HCI_LE_SCAN_DISABLE,
                           filterDuplicates=HCI_LE_DUPLICATE_FILTERING_DISABLE
                         }

+79016 ms
Final Verdict:           INCONC

PR #21026 (2019-12-14): Bluetooth: Mesh: Fixes Friend Add Another Update
Fixes: #20746

According MESH/NODE/FRND/FN/BV-20-C should add another
Friend Update.
commit 6e260af8 (2019-12-14): Bluetooth: Mesh: Fixes Friend Add Another Update

Fixes: #20746

According MESH/NODE/FRND/FN/BV-20-C should add another
Friend Update.
closed 2019-12-14 by jhedberg