zephyrproject-rtos/zephyr #20748: build warnings on lpcxpresso54114_m0/m4 board

hakehuang, 2019-11-16:
**Describe the bug**
there are build warnings for board lpcxpresso54114_m0 and lpcxpresso54114_m4

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=lpcxpresso54114_m4 ..
3. make
4. See error

**Expected behavior**
clean build log

**Impact**
no found yet

**Screenshots or console output**

'''
Zephyr version: 2.1.0
-- Found PythonInterp: /usr/bin/python3 (found suitable version "3.6.8", minimum required is "3.4")
-- Selected BOARD lpcxpresso54114_m0
-- Found west: /usr/local/bin/west (found suitable version "0.6.0", minimum required is "0.6.0")
-- Loading /home/shared/disk/zephyr_project/zephyr/boards/arm/lpcxpresso54114/lpcxpresso54114_m0.dts as base
lpcxpresso54114_m0.dts.pre.tmp:69.17-76.5: Warning (simple_bus_reg): /soc/gpio@0: simple-bus unit address format error, expected "4008c000"
lpcxpresso54114_m0.dts.pre.tmp:77.17-84.5: Warning (simple_bus_reg): /soc/gpio@1: simple-bus unit address format error, expected "4008c000"
lpcxpresso54114_m0.dts.pre.tmp:58.18-61.5: Warning (unique_unit_address_if_enabled): /soc/flash@0: duplicate unit-address (also used in node /soc/gpio@0)
Devicetree configuration written to /home/shared/temp/build_temp/zephyr/include/generated/generated_dts_board.conf
warning: unit-address and first reg (0x4008c000) don't match for gpio@0
warning: unit-address and first reg (0x4008c000) don't match for gpio@0
warning: unit-address and first reg (0x4008c000) don't match for gpio@0
warning: unit-address and first reg (0x4008c000) don't match for gpio@0
warning: unit-address and first reg (0x4008c000) don't match for gpio@0
warning: unit-address and first reg (0x4008c000) don't match for gpio@0
warning: unit-address and first reg (0x4008c000) don't match for gpio@0
warning: unit-address and first reg (0x4008c000) don't match for gpio@0
warning: unit-address and first reg (0x4008c000) don't match for gpio@0
warning: unit-address and first reg (0x4008c000) don't match for gpio@0
warning: unit-address and first reg (0x4008c000) don't match for gpio@1
warning: unit-address and first reg (0x4008c000) don't match for gpio@1
warning: unit-address and first reg (0x4008c000) don't match for gpio@1
warning: unit-address and first reg (0x4008c000) don't match for gpio@1
warning: unit-address and first reg (0x4008c000) don't match for gpio@1
warning: unit-address and first reg (0x4008c000) don't match for gpio@1
warning: unit-address and first reg (0x4008c000) don't match for gpio@1
warning: unit-address and first reg (0x4008c000) don't match for gpio@1
warning: unit-address and first reg (0x4008c000) don't match for gpio@1
warning: unit-address and first reg (0x4008c000) don't match for gpio@1
Parsing Kconfig tree in /home/shared/disk/zephyr_project/zephyr/Kconfig
Loaded configuration '/home/shared/disk/zephyr_project/zephyr/boards/arm/lpcxpresso54114/lpcxpresso54114_m0_defconfig'
Merged configuration '/home/shared/disk/zephyr_project/zephyr/samples/hello_world/prj.conf'
Configuration saved to '/home/shared/temp/build_temp/zephyr/.config'
-- The C compiler identification is GNU 8.3.0
-- The CXX compiler identification is GNU 8.3.0
-- The ASM compiler identification is GNU
-- Found assembler: /opt/zephyr-sdk/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc
-- Cache files will be written to: /home/ubuntu/.cache/zephyr
-- Configuring done
-- Generating done
-- Build files have been written to: /home/shared/temp/build_temp

'''

**Environment (please complete the following information):**
 - OS: (e.g. Linux)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used: v2.1.0-rc1

agansari, 2019-11-18:
@hakehuang @MaureenHelm this warning is generated by the following:
```
		gpio0:gpio@0 {
			...
			reg = <0x5008c000 0x2488>;
			...
		}
```

@0 and 0x5008c000 don't match, this is because **all gpios on LPC use the same registers**.
Using gpio@5008c000 will map all the separate ports as the last defined port.

It's a devicetree issue, @galak any advice on how to split dts ports that use the same register?
In my opinion we could just suppress the warning on this particular SoC family.

galak, 2019-11-19:
This is a long standing issue since the construct of having interleaved registers for two different devices isn't something device tree handles.

So if we have some way to suppress the DTS warning for the LPC SoCs that would be best.

ulfalizer, 2019-11-20:
Could the unit address be left out instead? Can have e.g. `gpio0` as the name.

PR #20828 (2020-02-05): DTC warning suppression
Allows boards to have a extra.cmake that allows for custom cmake commands.
We use the extra.cmake file to set EXTRA_DTC_FLAGS to suppress warnings on LPC SoC that have a single register address for all gpio ports.

Fixes #20748
closed 2020-02-05 by galak