zephyrproject-rtos/zephyr #20749: samples:sample.net.dns_resolve.mdns:frdmk64f ipv4dns handler has not result

hakehuang, 2019-11-16:
**Describe the bug**
in this sample we expect to see both ipv4 and ipv6 dns parser result, but for v2.1.0-rc1, I can only see the ipv6 handler triggerred.

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=frdm_k64f -DCONFIG_MDNS_RESOLVER=y -DCONFIG_LLMNR_RESOLVER=n -DCONFIG_NET_DHCPV4=y ..
3. make
4. make flash
5. see error

**Expected behavior**
 ishall be able to get both ipv4 and ipv6 dns result

**Impact**
net stack for DNS parser in ipv4

**Screenshots or console output**

'''
***** Booting Zephyr OS build v2.1.0-rc1 *****

[00:00:00.005,000] <inf> net_config: Initializing network
[00:00:00.005,000] <inf> net_config: IPv4 address: 192.0.2.1
[00:00:00.005,000] <inf> net_config: Running dhcpv4 client...
[00:00:00.005,000] <wrn> net_if: iface 0x2000a920 is down
[00:00:00.005,000] <err> net_if: Cannot join all nodes address ff02::1 (-1)
[00:00:00.005,000] <wrn> net_if: iface 0x2000a920 is down
[00:00:00.005,000] <err> net_if: Cannot join solicit node address ff02::1:ff00:1 (-1)
[00:00:03.001,000] <inf> eth_mcux: Enabled 100M full-duplex mode.
[00:00:03.004,000] <inf> net_dhcpv4: Received: 192.168.0.104
[00:00:03.004,000] <inf> net_config: IPv4 address: 192.168.0.104
[00:00:03.004,000] <inf> net_config: Lease time: 172800 seconds
[00:00:03.004,000] <inf> net_config: Subnet: 255.255.255.0
[00:00:03.004,000] <inf> net_config: Router: 192.168.0.1
[00:00:03.101,000] <inf> net_config: IPv6 address: <log_strdup alloc failed>
[00:00:03.101,000] <inf> net_config: IPv6 address: <log_strdup alloc failed>
[00:00:06.101,000] <inf> net_dns_resolve_client_sample: Starting DNS resolve sample
[00:00:06.101,000] <inf> net_dns_resolve_client_sample: Getting IPv4 address via DHCP before issuing DNS query
[00:00:06.101,000] <dbg> net_dns_resolve_client_sample.do_ipv6_lookup: DNS id 63664
[00:00:06.101,000] <dbg> net_dns_resolve_client_sample.do_mdns_ipv6_lookup: Doing mDNS IPv6 query
[00:00:06.102,000] <dbg> net_dns_resolve_client_sample.do_mdns_ipv6_lookup: mDNS v6 query sent
[00:00:06.137,000] <inf> net_dns_resolve_client_sample: www.zephyrproject.org IPv6 address: 2620:12a:8000::1
[00:00:06.137,000] <inf> net_dns_resolve_client_sample: www.zephyrproject.org IPv6 address: 2620:12a:8001::1
[00:00:06.137,000] <inf> net_dns_resolve_client_sample: DNS resolving finished

'''

**Environment (please complete the following information):**
 - OS: (e.g. Linux,)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used: v2.1.0-rc1

**Additional context**

if I disable the ipv6_setup in this example like below:

``
void main(void)
{
	struct net_if *iface = net_if_get_default();

	LOG_INF("Starting DNS resolve sample");

	setup_ipv4(iface);

	setup_dhcpv4(iface);

//	setup_ipv6(iface);
}

I can get below log

'''
***** Booting Zephyr OS build v2.1.0-rc1 *****

[00:00:00.005,000] <inf> net_config: Initializing network
[00:00:00.005,000] <inf> net_config: IPv4 address: 192.0.2.1
[00:00:00.005,000] <inf> net_config: Running dhcpv4 client...
[00:00:00.005,000] <wrn> net_if: iface 0x2000a900 is down
[00:00:00.005,000] <err> net_if: Cannot join all nodes address ff02::1 (-1)
[00:00:00.005,000] <wrn> net_if: iface 0x2000a900 is down
[00:00:00.005,000] <err> net_if: Cannot join solicit node address ff02::1:ff00:1 (-1)
[00:00:03.001,000] <inf> eth_mcux: Enabled 100M full-duplex mode.
[00:00:03.003,000] <inf> net_dhcpv4: Received: 192.168.0.104
[00:00:03.004,000] <inf> net_config: IPv4 address: 192.168.0.104
[00:00:03.004,000] <inf> net_config: Lease time: 172800 seconds
[00:00:03.004,000] <inf> net_config: Subnet: 255.255.255.0
[00:00:03.004,000] <inf> net_config: Router: 192.168.0.1
[00:00:03.101,000] <inf> net_config: IPv6 address: <log_strdup alloc failed>
[00:00:03.101,000] <inf> net_config: IPv6 address: <log_strdup alloc failed>
[00:00:06.101,000] <inf> net_dns_resolve_client_sample: Starting DNS resolve sample
[00:00:06.101,000] <inf> net_dns_resolve_client_sample: Getting IPv4 address via DHCP before issuing DNS query

'''

the same application behave different from former release v1.14.1 as below

'''
***** Booting Zephyr OS zephyr-v1.14.0 *****

[00:00:00.000,000] <inf> net_config: Initializing network
[00:00:00.000,000] <inf> net_config: IPv4 address: 192.0.2.1
[00:00:00.000,000] <inf> net_config: Running dhcpv4 client...
[00:00:00.110,000] <inf> net_config: IPv6 address: fe80::204:9fff:fe44:b91a
[00:00:00.110,000] <inf> net_config: IPv6 address: fe80::204:9fff:fe44:b91a
[00:00:03.120,000] <inf> net_dns_resolve_client_sample: Starting DNS resolve sample
[00:00:03.120,000] <inf> net_dns_resolve_client_sample: Getting IPv4 address via DHCP before issuing DNS query
[00:00:03.120,000] <dbg> net_dns_resolve_client_sample.do_ipv6_lookup: DNS id 40822
[00:00:05.130,000] <inf> net_dns_resolve_client_sample: DNS query was canceled
[00:00:08.170,000] <inf> net_dhcpv4: Received: 10.192.244.58
[00:00:08.170,000] <inf> net_dns_resolve_client_sample: IPv4 address: 10.192.244.58
[00:00:08.170,000] <inf> net_dns_resolve_client_sample: Lease time: 7200 seconds
[00:00:08.170,000] <inf> net_dns_resolve_client_sample: Subnet: 255.255.255.0
[00:00:08.170,000] <inf> net_dns_resolve_client_sample: Router: 10.192.244.254
[00:00:08.170,000] <inf> net_config: IPv4 address: <log_strdup alloc failed>
[00:00:08.170,000] <inf> net_config: Lease time: 7200 seconds
[00:00:08.170,000] <inf> net_config: Subnet: <log_strdup alloc failed>
[00:00:08.170,000] <inf> net_config: Router: <log_strdup alloc failed>
[00:00:08.170,000] <dbg> net_dns_resolve_client_sample.do_ipv4_lookup: DNS id 32200
[00:00:08.210,000] <inf> net_dns_resolve_client_sample: www.zephyrproject.org IPv4 address: 23.185.0.1
[00:00:08.210,000] <inf> net_dns_resolve_client_sample: DNS resolving finished
[00:00:08.210,000] <dbg> net_dns_resolve_client_sample.do_mdns_ipv4_lookup: Doing mDNS IPv4 query
[00:00:08.210,000] <dbg> net_dns_resolve_client_sample.do_mdns_ipv4_lookup: mDNS id 50262
'''

ioannisg, 2019-11-27:
@agansari do you have any updates on this?

agansari, 2019-11-27:
@ioannisg no, sorry did not have time to get to this issue

ioannisg, 2019-11-27:
Ok, just checking the status of medium priority bugs, for 2.1 release; let us know if you plan to have this bug fixed for 2.1 (that should be soon), or drop it off to next release

agansari, 2019-12-02:
@hakehuang 
I've ran multiple configurations and found the following:

After correctly configuring my local server, net-tools and the following in prj.conf:
```
...
# Use local dnsmasq server for testing
CONFIG_DNS_SERVER1="[2001:db8::100]:15353"
CONFIG_DNS_SERVER2="192.0.2.100:15353"
...
CONFIG_NET_CONFIG_PEER_IPV6_ADDR="2001:db8::100"
...
CONFIG_NET_CONFIG_PEER_IPV4_ADDR="192.0.2.100"
...
```
I have a successful running configuration with default configuration (only changed ip addresses to my local setup, see above):
```
*** Booting Zephyr OS build v2.1.0-rc2-26-ga034403440f2  ***
[00:00:00.006,000] <inf> net_config: Initializing network
[00:00:00.006,000] <inf> net_config: IPv4 address: 192.0.2.1
[00:00:00.006,000] <wrn> net_if: iface 0x2000a8c0 is down
[00:00:00.006,000] <err> net_if: Cannot join all nodes address ff02::1 (-1)
[00:00:00.006,000] <wrn> net_if: iface 0x2000a8c0 is down
[00:00:00.006,000] <err> net_if: Cannot join solicit node address ff02::1:ff00:1 (-1)
[00:00:03.001,000] <inf> eth_mcux: Enabled 100M full-duplex mode.
[00:00:03.101,000] <inf> net_config: IPv6 address: 2001:db8::1
[00:00:03.101,000] <inf> net_config: IPv6 address: 2001:db8::1
[00:00:06.101,000] <inf> net_dns_resolve_client_sample: Starting DNS resolve sample
[00:00:06.101,000] <inf> net_dns_resolve_client_sample: IPv4 address: 192.0.2.1
[00:00:06.101,000] <dbg> net_dns_resolve_client_sample.setup_ipv4: DNS id 26035
[00:00:06.101,000] <dbg> net_dns_resolve_client_sample.do_mdns_ipv4_lookup: Doing mDNS IPv4 query
[00:00:06.102,000] <dbg> net_dns_resolve_client_sample.do_mdns_ipv4_lookup: mDNS v4 query sent
[00:00:06.102,000] <dbg> net_dns_resolve_client_sample.do_ipv6_lookup: DNS id 1544
[00:00:06.103,000] <dbg> net_dns_resolve_client_sample.do_mdns_ipv6_lookup: Doing mDNS IPv6 query
[00:00:06.103,000] <dbg> net_dns_resolve_client_sample.do_mdns_ipv6_lookup: mDNS v6 query sent
[00:00:06.103,000] <inf> net_dns_resolve_client_sample: www.zephyrproject.org IPv4 address: 192.0.2.100
[00:00:06.103,000] <inf> net_dns_resolve_client_sample: DNS resolving finished
[00:00:06.104,000] <inf> net_dns_resolve_client_sample: www.zephyrproject.org IPv6 address: 2001:db8::2
[00:00:06.104,000] <inf> net_dns_resolve_client_sample: DNS resolving finished
[00:00:08.102,000] <inf> net_dns_resolve_client_sample: DNS query was canceled
[00:00:08.103,000] <inf> net_dns_resolve_client_sample: DNS query was canceled
uart:~$ 
```

Now regarding your configuration:
-DCONFIG_MDNS_RESOLVER=y -> same as default
-DCONFIG_LLMNR_RESOLVER=n -> same as default
-DCONFIG_NET_DHCPV4=y -> different
So with the configuration above + DCONFIG_NET_DHCPV4=y i get the following:
```
*** Booting Zephyr OS build v2.1.0-rc2-26-ga034403440f2  ***
[00:00:00.006,000] <inf> net_config: Initializing network
[00:00:00.006,000] <inf> net_config: IPv4 address: 192.0.2.1
[00:00:00.006,000] <inf> net_config: Running dhcpv4 client...
[00:00:00.006,000] <wrn> net_if: iface 0x2000a920 is down
[00:00:00.006,000] <err> net_if: Cannot join all nodes address ff02::1 (-1)
[00:00:00.007,000] <wrn> net_if: iface 0x2000a920 is down
[00:00:00.007,000] <err> net_if: Cannot join solicit node address ff02::1:ff00:1 (-1)
[00:00:03.001,000] <inf> eth_mcux: Enabled 100M full-duplex mode.
[00:00:03.005,000] <inf> net_dhcpv4: Received: 192.0.2.3
[00:00:03.005,000] <inf> net_config: IPv4 address: 192.0.2.3
[00:00:03.005,000] <inf> net_config: Lease time: 3600 seconds
[00:00:03.005,000] <inf> net_config: Subnet: 255.255.255.0
[00:00:03.005,000] <inf> net_config: Router: 192.0.2.100
**[00:00:03.101,000] <inf> net_config: IPv6 address: <log_strdup alloc failed>
[00:00:03.101,000] <inf> net_config: IPv6 address: <log_strdup alloc failed>**
[00:00:06.101,000] <inf> net_dns_resolve_client_sample: Starting DNS resolve sample
[00:00:06.101,000] <inf> net_dns_resolve_client_sample: Getting IPv4 address via DHCP before issuing DNSy
[00:00:06.101,000] <dbg> net_dns_resolve_client_sample.do_ipv6_lookup: DNS id 57651
[00:00:06.101,000] <dbg> net_dns_resolve_client_sample.do_mdns_ipv6_lookup: Doing mDNS IPv6 query
[00:00:06.102,000] <dbg> net_dns_resolve_client_sample.do_mdns_ipv6_lookup: mDNS v6 query sent
[00:00:06.103,000] <inf> net_dns_resolve_client_sample: www.zephyrproject.org IPv6 address: 2001:db8::2
[00:00:06.103,000] <inf> net_dns_resolve_client_sample: DNS resolving finished
**[00:00:08.102,000] <inf> net_dns_resolve_client_sample: DNS query was canceled**
```

Comment: dhcp ip v4 conflicts in application and networking subsystem, DNS is not resolved for v4.
Also v6 address allocation fails when dhcp ip v4 is enabled.

DCONFIG_NET_DHCPV4=n as default since application is starting dhcp client.

However it's not clear why v6 allocation and v4 dhcp interact.

agansari, 2019-12-02:
<log_strdup alloc failed> message is due to the fact LOG_STRDUP_BUF_COUNT is set too low, we don't have enough logging buffers (because the newly enabled subsys dhcp also uses these buffers).

Enabling ip's subsys DHCP will do the following: run dhcp client at boot and getting an ip address. The second time the dns application calls the dhcp functions, it gets the information system is ready bound and will not run DNS callbacks. This is why DCONFIG_NET_DHCPV4=n as default.

Any other issue is due to configuration. 
DNS resolve sample is limited in purpose and is designed to work with net-tools.
https://docs.zephyrproject.org/latest/samples/net/dns_resolve/README.html

Think this issue can be closed.

hakehuang, 2019-12-04:
@agansari , do you notice that in this application we expect both IPV4 and IPV6 dns resolved, however, there is only IPV6 DNS resolved, why no IPV4 DNS resolved?

hakehuang, 2019-12-04:
@anangl ,  ok, seems this application only works in dhcpv6 mode.

closed 2019-12-04 by hakehuang

agansari, 2019-12-04:
@hakehuang it works with both ipv4 and ipv6, the problem is enabling DCONFIG_NET_DHCPV4 which conflicts with sample's dhcp setup.