zephyrproject-rtos/zephyr #20770: irq locking in logging backend can cause missing interrupts

INemesisI, 2019-11-18:
**Describe the bug**
The logging subsystem uses irq_lock() in certain locations to lock all interrupts while processing data. When CONFIG_LOG_IMMEDIATE is enabled interrupts will be disabled during the whole write progess, as can be seen here:
https://github.com/zephyrproject-rtos/zephyr/blob/d58f8bee79dddb9aa1b65a2ef1624ebf1b389a4e/subsys/logging/log_backend_std.h#L75

This can cause drivers which are relying on interrupts for reading data to miss bytes while logging at the same time.

I noticed the bug while working with an external cellular modem for data transmission with an STM32L4. Bytes from the UART interface are read by an interrupt handler. I noticed seemingly randomly lost bytes every now and then.

When commenting out the disabling of interrupts while writing data at https://github.com/zephyrproject-rtos/zephyr/blob/d58f8bee79dddb9aa1b65a2ef1624ebf1b389a4e/subsys/logging/log_backend_std.h#L90 the issue disappears.

**To Reproduce**
I unfortunately don't have a sample project which demonstrates the issue at the moment. An external modem connected to the uC with net example code should already cause the issue. Logging frequently should increase the chance to cause this issue.
This especially happens while sending/recieving big chunks of data, like large packages.

**Expected behavior**
I expect peripheral drivers using interrupts for reading (like UART or I2C) to receive data reliably without the possibility to loose bytes in any case.

**Discussion**
I haven't looked at the logging backend code so far and am sure, that there is a specific reason why all interrupts must be disabled while sending data.
I would like to know why there is the need for this and how we could maybe prevent this from causing issues in other peripheral drivers.

INemesisI, 2019-11-26:
Thanks for working so quickly on this issue @nordic-krch!

I also want to point out that interrupt locking is not just happening in immediate mode.
Interrupts are also briefly locked in threaded mode when adding/removing elements from the list of logs. 
To me those operations sound like a case which is usually handled by semaphores. But again, I am not the expert here!

nordic-krch, 2019-11-26:
@INemesisI we cannot use semaphores because logs can be used in interrupt context.

wentongwu, 2019-12-04:
@INemesisI is it good for your case if configure logger in thread mode?

INemesisI, 2019-12-10:
No, The issue still persists then, although it just happens less often, since the irq is locked for a very short time. The interrupt locking itself causes the issue for me. 

The problem is, that the UART RX data on the STM32 is not buffered. Each byte is fetched through an ISR. Missing interrupts because of irq locking will result in lost data.
I don't know about other hardwares, but I can imagine that there are implementation where at least the delay introduced by locking could be not acceptable.

I see that most devs do not like the proposal to stop using irq locking. I wonder if there is another way to fix somehow.. Otherwise I would need to disable it in my own fork of zephyr, since I am not logging inside ISR's anyway.

StefJar, 2019-12-10:
I am having maybe an related issue my I2C_1 is not catching the tx complete INTR. I put this on slack an wait for some guys to confirm.
@INemesisI I recomend to use the ASYNC IO interface for the UART. If you need some working code for async uart transfers check my demo https://github.com/StefJar/zephyr_stm32_uart3_dma_driver

nordic-krch, 2019-12-11:
@INemesisI why are you using `LOG_IMMEDIATE` mode? Have you tried standard, deferred mode? In that mode log message is created quite fast (few ~5us on nrf52 @64MHz). It has to lock interrupts at least for buffer allocation and list management but only for very short periods.

PR #20902 (2019-12-18): logging: Remove long interrupt locking when immediate mode is used.
Changed behavior of the logger in immediate mode. Previously, context was lock for whole string processing and sending which resulted in log output being clean and not interleaved but at the cost of very long period of interrupt locking. It is changed to lock interrupts only for single byte sending over transport (e.g. uart). With this approach long locking time is removed but log output may be garbled since one log message can be interrupted by another.

Added test suite for log immediate mode which checks if logger is stable [... cut]
closed 2019-12-18 by andrewboie