zephyrproject-rtos/zephyr #20771: onoff_level_lighting_vnd_app mcumgr unable to connect to provisioned node

drewmlsuvo, 2019-11-18:
**Describe the bug**
I am running the onoff_level_lighting_vnd_app sample on an nrf52840_pca10056 board with the extra settings in prj_smp_svr.conf enabled. After I provisioned the board I am no longer to connect to it using mcumgr.

**To Reproduce**
Steps to reproduce the behavior:
1. west build -b nrf52840_pca10056
2. sign image for mcuboot
3. nrfjprog --program zephyr.hex --sectorerase
4. mcumgr --conntype ble --connstring ctlr_name=hci0,peer_name='Zephyr' image list

**Expected behavior**
A response from mcumgr with the current images loaded on the board.

**Impact**
annoyance

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain Zephyr SDK
 - Commit SHA c82f3d4

carlescufi, 2019-11-19:
@vikrant8051 is this something you still observe?

vikrant8051, 2019-11-19:
Please refer image in this tweet
https://twitter.com/vikrant8051/status/1096033120732639234?s=19
which contains script for entire procedure.

May need some modifications as per Zephyr 2.0

vikrant8051, 2019-11-19:
Please share your result so that we can create new shell script 
& add it in App's readme.

vikrant8051, 2019-11-19:
Steps:
1) Clone mcuboot repo under zephyrproject: https://github.com/JuulLabs-OSS/mcuboot
2) cd zephyrproject/mcuboot
3) pip3 install -r scripts/requirements.txt
4) pip3 install imgtool
5) cd zephyrproject/zephyr/samples/boards/nrf52/mesh/onoff_level_lighting_vnd_app
6) Erase full flash
7) west build -b nrf52840_pca10056 -s zephyrproject/mcuboot-master/boot/zephyr -d build-mcuboot
8) west flash -d build-mcuboot
9) west build -b nrf52840_pca10056 -s ./ -d build-app -- -DOVERLAY_CONFIG=prj_smp_svr.conf 
10) west sign -t imgtool -d build-app -- --key zephyrproject/mcuboot-master/root-rsa-2048.pem
11) west flash -d build-app --hex-file zephyr.signed.hex

drewmlsuvo, 2019-11-19:
Thanks for your help! however I'm still unable to connect to the board using mcumgr after it has been provisioned (time out error) using the peer name 'Zephyr', I am however able to connect to it using mcumgr when it is unprovisioned.

I've checked and it doesn't seem to be advertising with the name 'Zephyr' while provisioned.

vikrant8051, 2019-11-19:
Why you are use mcumgr ? First try it with DFU feature from nRF Connect App which supports mcuboot.

Modify something in onoff_level_lighting_vnd_app ...for eg. initial printk message & create it's signed copy. 
This will help to confirm after DFU that image has exchanged.

jhedberg, 2019-11-19:
> One second, as per my knowledge after provisioning Mesh NODE doesn't broadcast "Zephyr" as device name.

That's correct. If GATT Proxy support is available we'll either be advertising Node or Network Identity, and neither of those contain the device name.

vikrant8051, 2019-11-26:
@drewmlsuvo Hope issue has solved. If so kindly close it.

drewmlsuvo, 2019-11-26:
Thanks for your help, I managed to get it working last night.

closed 2019-11-26 by drewmlsuvo