zephyrproject-rtos/zephyr #20781: peripheral_hr on VEGABoard disconnects from central_hr after BT_CONN_PARAM_UPDATE_TIMEOUT

alexandru-porosanu-nxp, 2019-11-19:
**Describe the bug**
When running peripheral_hr on a VEGABoard, connected to a nrf52 central_hr, after 5s, the peripheral disconnects with error code 0x08 (timeout).

**To Reproduce**
Steps to reproduce the behavior:
1. cmake -Bbuild_vega -G"Eclipse CDT4 - Ninja" -DBOARD=rv32m1_vega_ri5cy -DCMAKE_REQUIRED_FLAGS=-Wl,-dT=/dev/null /work/zephyr/zephyr-source/samples/bluetooth/peripheral_hr
2. ninja -Cbuild_vega
3. deploy on VEGA Board

**Expected behavior**
The peripheral_hr is able to communicate endlessly with the central_hr

**Impact**
The VEGA SW LL disconnects after 5s (which is equal to BT_CONN_PARAM_UPDATE_TIMEOUT), and reconnects, which is a showstopper (unstable)

**Screenshots or console output**
[SUBSCRIBED]
[NOTIFICATION] data 0x200065f7 length 2
[NOTIFICATION] data 0x200065f7 length 2
[NOTIFICATION] data 0x200065f7 length 2
[NOTIFICATION] data 0x200065f7 length 2
[NOTIFICATION] data 0x200065f7 length 2
[UNSUBSCRIBED]
Disconnected: cb:82:d1:e1:5d:d4 (random) (reason 0x08)

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: VEGA OpenISA official toolchain
 - head/master

Reverting d1d603afc3558f31a4728d70179d1cb5d74890c5 Bluetooth: controller: split: Fix conn update to be cacheable seems to 'fix' it.

alexandru-porosanu-nxp, 2019-11-19:
Here's a capture of the packet exchanges. At 339, you can see the start of the connection update procedure. Then the disconnection happens.

[conn_update_fail.zip](https://github.com/zephyrproject-rtos/zephyr/files/3863100/conn_update_fail.zip)

joerchan, 2019-11-19:
@alexandru-porosanu-nxp You said it was 5 seconds. But from the Sniffer Capture the slave disappears at the Connection Update Instant. 

@cvinayak From the sniffer log it looks like the master is sending outside of the TX transmit window for the connection parameter update.
The master transmits 34690 us after the slave, backtracked using empty packets. that is:
34689 + 150 + 80 + 80 = 34999

The tx windows should start at 24 * 1250 + (1 + 4) * 1259 = 36295
Old conn interval = 24
TX window offset = 4

Seems to me the master transmitted too early so the slave fails to sync up with it.

George-Stefan, 2019-11-19:
Connection update is done/initiated in ull_conn_llcp() called from ull_slave_ticker_cb().

d1d603afc3558f31a4728d70179d1cb5d74890c5 breaks this somehow.
Still debugging

joerchan, 2019-11-19:
@George-Stefan Can you check if the master is sending connetion update because of the L2CAP conn param update request? Or if it initiated it itself? Or possibly both?

Or is it possible that 24 * 1250 + (1 + 4) * 1250 has turned into 24 * 1250 + 4 * 1250 instead?
My suspicion for that is that there is state confusion between connection parameter update procedure, and connection parameter request procedure.

George-Stefan, 2019-11-19:
> @George-Stefan Can you check if the master is sending connetion update because of the L2CAP conn param update request? 
Yes, the slave (Vega board) sends BT_L2CAP_CONN_PARAM_REQ.

Same scenario with and without  d1d603afc3558f31a4728d70179d1cb5d74890c5:
- the  master sends the same conn update params
- but in the  1st scenario the slave misses all the connection evens after instant
[image]
[image]
[image]
[image]

George-Stefan, 2019-11-19:
> My suspicion for that is that there is state confusion between connection parameter update procedure, and connection parameter request procedure.

It's disabled.
BT_CTLR_CONN_PARAM_REQ_SUPPORT(=n)

George-Stefan, 2019-11-19:
```
	**if (((conn->llcp_req - conn->llcp_ack) & 0x03) == 0x02) {**
		switch (conn->llcp_type) {
		case LLCP_CONN_UPD:
		{
			if (event_conn_upd_prep(conn, lazy,
						ticks_at_expire) == 0) {
				return -ECANCELED;
			}
		}
```

joerchan, 2019-11-19:
```
-	conn->llcp_type = LLCP_CONN_UPD;
-	conn->llcp_ack -= 2U;
```
Yes I saw that as well. It has been changed to `--` instead.
@cvinayak Should this  llcp_cu.ack be `-= 2U`? I think there was other places that had this change as well. I think it is fair to say that something is not right in the connection param update procedure.

George-Stefan, 2019-11-19:
conn->llcp_cu.ack--; was added but conn->llcp_cu.ack isn't checked for in  ull_conn_llcp() when CONFIG_BT_CTLR_CONN_PARAM_REQ is disabled

alexandru-porosanu-nxp, 2019-11-19:
@joerchan , I haven't been really clear in my statement: 5s after the connection is established, the slave disconnects; 5s being the conn update interval as well. Increasing this to 6s will lead to a disconnection 6s after the establishment of the connection. Sorry for not being more specific.

alexandru-porosanu-nxp, 2019-11-19:
> conn->llcp_cu.ack--; was added in conn_upd_recv() but conn->llcp_cu.ack isn't checked for in ull_conn_llcp() when CONFIG_BT_CTLR_CONN_PARAM_REQ is disabled
> 
> I believe that the bug will reproduce on a Nordic board as well when CONFIG_BT_CTLR_CONN_PARAM_REQ and CONFIG_BT_CTLR_PHY are disabled

I've disabled all the features that I could for the controller side and I've run  on the nrf52 we've got here:

# CONFIG_BT_CTLR_LE_ENC is not set
# CONFIG_BT_CTLR_CONN_PARAM_REQ is not set
# CONFIG_BT_CTLR_SLAVE_FEAT_REQ is not set
# CONFIG_BT_CTLR_PRIVACY is not set
# CONFIG_BT_CTLR_CHAN_SEL_2 is not set
# CONFIG_BT_CTLR_ADV_EXT is not set
# CONFIG_BT_CTLR_DTM_HCI is not set
# CONFIG_BT_CTLR_ADVANCED_FEATURES is not set

    the result is an identical behavior with the VEGA Board: after conn update interval, the connection gets broken. I've attached the config file that I used. I'll try to slowly add one by one till I can pinpoint which are the ones that make/break the behavior.

[config.zip](https://github.com/zephyrproject-rtos/zephyr/files/3864582/config.zip)

George-Stefan, 2019-11-19:
> I believe that the bug will reproduce on a Nordic board as well when CONFIG_BT_CTLR_CONN_PARAM_REQ and CONFIG_BT_CTLR_PHY are disabled

@alexandru-porosanu-nxp  see my commen

alexandru-porosanu-nxp, 2019-11-19:
> > I believe that the bug will reproduce on a Nordic board as well when CONFIG_BT_CTLR_CONN_PARAM_REQ and CONFIG_BT_CTLR_PHY are disabled
> 
> @alexandru-porosanu-nxp see my comment

Yep, running now w/both CONFIG_BT_CTLR_CONN_PARAM_REQ  and CONFIG_BT_CTLR_PHY disabled and see what happens. I've already run with CONFIG_BT_CTLR_CONN_PARAM_REQ=n but the issue did not appear.

alexandru-porosanu-nxp, 2019-11-19:
Another interesting fact: enabling CONFIG_BT_CTLR_SLAVE_FEAT_REQ_SUPPORT and CONFIG_BT_CTLR_SLAVE_FEAT_REQ on VEGA leads to the issue disappearing: there is no more disconnection after the connection update interval

alexandru-porosanu-nxp, 2019-11-19:
> ```
> -	conn->llcp_type = LLCP_CONN_UPD;
> -	conn->llcp_ack -= 2U;
> ```
>
> Yes I saw that as well. It has been changed to `--` instead.
> @cvinayak Should this llcp_cu.ack be `-= 2U`? I think there was other [...]

@cvinayak , @joerchan  correct me if I'm wrong, but isn't the whole concept of 'cachable conn update' to allow more than 1 LLCP PDUs to be non-ack'ed (and hence the -2)?

cvinayak, 2019-11-20:
@alexandru-porosanu-nxp the differential value (llcp_req, llcp_ack) is used as guard on the llcp context wherein the difference have the following meaning:
1. Thread is modifying the context of the LLCP. Ex: Populating parameters before ISR processes it.
2. Thread has commits the values and ISR can start processing the LLCP.
3. ISR has modified the LLCP context while thread requested to modify. The thread will revert its request.

This design was necessary earlier before making control procedures cacheable i.e. to cache local initiated control procedure when another prior local initiated procedure or a remote procedure is in progress. 

The channel map procedure has not been made cacheable, which is the only one currently modifying llcp_req value in thread. This the only one checking using the guard design. Refer to `ull_conn_llcp_req`

George-Stefan, 2019-11-20:
@cvinayak The bug comes from the fact that in ull_conn_llcp()
```
		/* check if connection update procedure is requested */
		} else if (conn->llcp_cu.ack != conn->llcp_cu.req) {
			/* switch to LLCP_CONN_UPD state machine */
			conn->llcp_type = LLCP_CONN_UPD;
			conn->llcp_ack -= 2U;

```

is under 
`#if defined(CONFIG_BT_CTLR_CONN_PARAM_REQ) || defined(CONFIG_BT_CTLR_PHY)
`
which are not defined for Vega

cvinayak, 2019-11-20:
@George-Stefan yes, I just discovered it myself... will you send a PR? or should I?

George-Stefan, 2019-11-20:
@cvinayak let me do it

alexandru-porosanu-nxp, 2019-11-20:
> @alexandru-porosanu-nxp the differential value (llcp_req, llcp_ack) is used as guard on the llcp context wherein the difference have the following meaning:
>
> 1. Thread is modifying the context of the [...]

Right, it now makes more sense. Thanks for your response @cvinayak

George-Stefan, 2019-11-20:
> Another interesting fact: enabling CONFIG_BT_CTLR_SLAVE_FEAT_REQ_SUPPORT and CONFIG_BT_CTLR_SLAVE_FEAT_REQ on VEGA leads to the issue disappearing: there is no more disconnection after the connection update interval

Doing this will prevent BT_L2CAP_CONN_PARAM_REQ being sent by the slave

closed 2019-11-20 by carlescufi

joerchan, 2019-11-20:
@alexandru-porosanu-nxp can you update the title of the issue to be more descritiptive of what the actual issue was. Nice to have for the history.

alexandru-porosanu-nxp, 2019-11-21:
> @alexandru-porosanu-nxp can you update the title of the issue to be more descritiptive of what the actual issue was. Nice to have for the history.

Sure, I'm thinking to change the tile to "Fix connection state machine if !(BT_CTLR_CONN_PARAM_REQ) || BT_CTLR_PHY)". Does it sound OK?

PR #20822 (2019-11-20): Bluetooth: controller: split: Fix check for control procedures request
Remove wrong #ifdef that guarded the control procedures request check

Fixes #20781