zephyrproject-rtos/zephyr #20800: Ready thread is not swapped in after being woken up in IRQ

KAGA164, 2019-11-19:
**Describe the bug**
Ready thread is not swapped in after being woken up in IRQ. Probably the sheduler does not do a swap after this.

**To Reproduce**
I have an application where I use only `k_poll` in the main thread. Application is polling for two signal events. I call `k_poll_signal_raise` from IRQ context. After this main thread is in ready state but it is never waken up. My application is in idle thread after that. When I modified idle thread to call explicit yield inside, the main thread which polling for signals is woken up after calling `k_poll_signal_raise`.

Idle thread:

```
while (true) {
#if SMP_FALLBACK
       k_busy_wait(100);
       k_yield();
#else
       (void)arch_irq_lock();  
       sys_power_save_idle();
       k_yield();
       //IDLE_YIELD_IF_COOP();
#endif  
}
```
**Expected behavior**
Ready thread should be swapped after being woken up in IRQ.

**Impact**
What impact does this issue have on your progress (e.g., annoyance, showstopper)

**Screenshots or console output**
If applicable, add a screenshot (drag-and-drop an image), or console logs
(cut-and-paste text and put a code fence (\`\`\`) before and after, to help
explain the issue.

**Environment (please complete the following information):**
 - nrf52840_pca10056 DK

KAGA164, 2019-11-19:
@pdunaj

pdunaj, 2019-11-19:
@carlescufi , @ioannisg

andyross, 2019-11-20:
Do you have an exercisable test case you can post?  We have lots of exercising test cases of threads enabled in an ISR correctly preempting the idle thread (literally every k_sleep() call that wakes up synchronously works like that).

The line you commented out is a little suspicious, though.  The point of that macro is that the idle thread is normally preemptible and so it just spins waiting for an interrupt, but there are some ways to configure things (e.g. have zero preemptible thread priorities) such that it looks "cooperative" and has to call yield itself.  This doesn't match your symptoms (I'd expect an infinite hang if this was wrong) but might be a place to start looking.

Can you post the .config file for your app?

Alternatively, it could be a bug in poll when called from ISR context that isn't correctly reading your thread.  Are you able to swap in another IPC primitive to do that wakeup to test with, maybe a semaphore?

andyross, 2019-11-20:
Yeah, this seems to be working as intended for me in the minimal test case below (drop on top of hello_world's main.c, add CONFIG_POLL=y to prj.conf).  You'll see it log it's progress as it enters idle, gets preempted by the timer ISR, which signals a poll event and wakes up a thread which then runs on ISR exit.

This is hitting some deeper edge case, or else might be an app bug.

```c
#include <zephyr.h>

struct k_poll_signal my_signal = K_POLL_SIGNAL_INITIALIZER(my_signal);

void my_timer_handler(struct k_timer *timer)
{
	printk("Timer handler (ISR): signaling polling thread\n");
	k_poll_signal_raise(&my_signal, 0);
}

K_TIMER_DEFINE(my_timer, my_timer_handler, NULL);

struct k_thread my_thread;
K_THREAD_STACK_DEFINE(my_stack, 1024);

void my_thread_fn(void *arg0, void *arg1, void *arg2)
{
	printk("Child thread: starting\n");

	struct k_poll_event events[] = {
		K_POLL_EVENT_INITIALIZER(K_POLL_TYPE_SIGNAL,
					 K_POLL_MODE_NOTIFY_ONLY,
					 &my_signal),
	};

	printk("Child thread: entering k_poll(), sleeping...\n");
	k_poll(events, ARRAY_SIZE(events), K_FOREVER);
	printk("Child thread: k_poll() returned, exiting\n");
}

void main(void)
{
	k_thread_create(&my_thread, my_stack, K_THREAD_STACK_SIZEOF(my_stack),
			my_thread_fn, NULL, NULL, NULL,
			k_thread_priority_get(k_current_get()) - 1,
			0, K_NO_WAIT);

	printk("Main thread: starting timer\n");
	k_timer_start(&my_timer, K_MSEC(1000), K_NO_WAIT);
	printk("Main thread: Timer started, sleeping... (ENTERING IDLE)\n");

	k_sleep(K_MSEC(2000));
	printk("Main thread: awake, exiting\n");
}
```

KAGA164, 2019-11-22:
It seems like after I rebase on on newer revision of the Zephyr. I can wake up thread from IRQ. I will do a few short tests to be sure.

ioannisg, 2019-11-22:
> It seems like after I rebase on on newer revision of the Zephyr. I can wake up thread from IRQ. I will do a few short tests to be sure.

Thanks @KAGA164 , this is blocking the release right now, so it's great if this report is a false positive, and we can get rid of it today.

KAGA164, 2019-11-22:
@ioannisg I ran the very similar test as above and also my app I do no have this problem anymore so I close this issue

closed 2019-11-22 by KAGA164