zephyrproject-rtos/zephyr #20810: Better separation between guides and API references

mbolivar, 2019-11-20:
## Introduction

The Zephyr documentation has "API Reference" and "User and Developer Guides" sections. I believe several API reference pages contain valuable information that would be better placed in the guides, and that this is a discoverability problem.

If there's consensus about that, I think the "guide" documentation should be moved out of "API Reference" pages and into "User and Developer Guides", with cross-references to link the two.

### Problem description

Assuming that:

- "guides" should contain high-level introductions suitable for readers who are unfamiliar with a subject
- "reference" pages should provide comprehensive information more targeted to readers searching for something specific

Then the API Reference contains a lot of valuable information that is more "guide" than "reference".

### Proposed change

- move all "guide" information out of "reference" and into "guides"
- provide cross-references from "guides" to "references", and vice versa (e.g. device model guide <-> device model API reference)

## Detailed RFC

This RFC covers the following pages under "API Reference":

- [driver model](https://docs.zephyrproject.org/latest/reference/drivers/index.html)
- [kernel services](https://docs.zephyrproject.org/latest/reference/kernel/index.html) **and all sub-pages** ([Threads](https://docs.zephyrproject.org/latest/reference/kernel/threads/index.html), [Scheduling](https://docs.zephyrproject.org/latest/reference/kernel/scheduling/index.html), etc.)
- [logging](https://docs.zephyrproject.org/latest/reference/logging/index.html)
- Various networking pages (I stopped looking after a while, but at least the [sockets](https://docs.zephyrproject.org/latest/reference/networking/sockets.html), [DNS](https://docs.zephyrproject.org/latest/reference/networking/dns_resolve.html), [network management](https://docs.zephyrproject.org/latest/reference/networking/net_mgmt.html), [network statistics](https://docs.zephyrproject.org/latest/reference/networking/net_stats.html), [promiscuous mode](https://docs.zephyrproject.org/latest/reference/networking/promiscuous.html), [websocket](https://docs.zephyrproject.org/latest/reference/networking/websocket.html), [network buffers](https://docs.zephyrproject.org/latest/reference/networking/net_buf.html), [net_pkt](https://docs.zephyrproject.org/latest/reference/networking/net_pkt.html), [Thread](https://docs.zephyrproject.org/latest/reference/networking/thread.html), [PPP](https://docs.zephyrproject.org/latest/reference/networking/ppp.html), [CoAP](https://docs.zephyrproject.org/latest/reference/networking/coap.html), [LWM2M](https://docs.zephyrproject.org/latest/reference/networking/lwm2m.html), and [MQTT](https://docs.zephyrproject.org/latest/reference/networking/mqtt.html) pages have "how to" or "overview" style documentation)
- [sensors](https://docs.zephyrproject.org/latest/reference/peripherals/sensor.html) (this is the only sub-page of [peripherals](https://docs.zephyrproject.org/latest/reference/peripherals/index.html) covered by this RFC; the rest seem to be more "pure" API pages)
- [power management](https://docs.zephyrproject.org/latest/reference/power_management/index.html)
- [shell](https://docs.zephyrproject.org/latest/reference/shell/index.html)
- these storage pages: [NVS](https://docs.zephyrproject.org/latest/reference/storage/nvs/nvs.html), [SDHC](https://docs.zephyrproject.org/latest/reference/storage/disk/sdhc.html), [FCB](https://docs.zephyrproject.org/latest/reference/storage/fcb/fcb.html)
- [USB device stack](https://docs.zephyrproject.org/latest/reference/usb/index.html) (though this one might want to move under Peripherals, actually...)
- [user mode](https://docs.zephyrproject.org/latest/reference/usermode/index.html)
- [settings](https://docs.zephyrproject.org/latest/reference/runtime_conf/settings/settings.html) and its parent topic of runtime configuration

### Proposed change (Detailed)

The high-level overview and theory, code examples, and "HOWTO" content of these API reference pages should be moved into pages under "User and Developer Guides". Each such page should provide a cross-reference to the corresponding detailed API page at the top. The API references should also provide a link to the corresponding guides.

### Dependencies

None

### Concerns and Unresolved Questions

I'm not sure how the current split between references and guides was decided upon, but I am concerned that the current state of affairs presents real discoverability problems for our documentation.

Consider a reader who found this information about an Optimized Device Driver Model and Devicetree Support in the Introduction:

[image]

Let's imagine where they'll go to look for more information on that from the top level index:

[image]

I would expect them to click "User and Developer Guides", staying away from "API Reference". They would then miss out on the lovely and detailed guide available starting here:

https://docs.zephyrproject.org/latest/reference/drivers/index.html#introduction

## Alternatives

Rename the "API Reference" page to something else that makes it clear there are tutorials and overviews inside.

This might require moving some information in the "User and Developer Guides" over there, though. For example, [Bluetooth](https://docs.zephyrproject.org/latest/guides/bluetooth/index.html), [Network Stack Architecture](https://docs.zephyrproject.org/latest/guides/networking/net-stack-architecture.html), etc.

In general, however, I think there should be a separation between guides and references, so I don't like this alternative so much.

mbolivar, 2019-11-20:
@dbkinder @ru-fu please review and comment on this RFC

cc @carlescufi

nashif, 2019-11-28:
Maybe the easy solution is to remove API from "API Reference" and just call that section Reference.

The current status of this structure is the result of a major overhaul that was done earlier this year where thing have been moved around, you are basically proposing reverting all of that :)

mbolivar, 2019-12-02:
> Maybe the easy solution is to remove API from "API Reference" and just call that section Reference.

The word "reference" is exactly what I am objecting to.

> The current status of this structure is the result of a major overhaul that was done earlier this year where thing have been moved around, you are basically proposing reverting all of that :)

Is that really true? I think a lot of the tutorial style content that was added to "API Reference" may have been added after the reorg.

gmarull, 2021-03-29:
I generally agree with the proposal.

Taking the electronics world as an example, one usually has:

- "datasheets": detailed reference manuals
- "application notes": how to use a particular device or peripheral (+examples, +usecases...)

I think we can use the same principles. I'd call a "reference" a page that contains the API specification together with some basic hints (think of Doxygen). Likewise, I'd call a "user guide" pages that contain contextual information such as: the design principles of an API/subsys, examples on its usage, etc.

In general, I find Zephyr documentation hard to navigate. You usually find information deeply buried in the toctree, so it's hard to discover. In order to improve "references" I propose to create a new root toctree, titled `REFERENCES` with:

- API Reference
- Kconfig Reference
- Devicetree bindings Reference

This is what Godot does (I really love their docs, IMO their structure could be used as a good reference on how to structure documentation):

[image]

nashif, 2023-04-13:
still valid? I thought we did the rework long time ago...