zephyrproject-rtos/zephyr #20811: spi driver

wentongwu, 2019-11-20:
## Introduction

with user space enabled, from the call trace: spi_transceive->z_vrfy_spi_transceive->transceive,
the parameters tx_bufs and rx_bufs passed between these functions are buffer pointers from stack like below, so it's very likely that these are not NULL, so in transceive->spi_context_buffers_setup, the tx_bufs and rx_bufs will never be NULL, so there is problem when users pass NULL in spi_transceive.

```
static inline int z_vrfy_spi_transceive(struct device *dev,
                                       const struct spi_config *config,
                                       const struct spi_buf_set *tx_bufs,
                                       const struct spi_buf_set *rx_bufs)
{
        struct spi_buf_set tx_bufs_copy;
        struct spi_buf_set rx_bufs_copy;
        struct spi_config config_copy;
        ...
        return copy_bufs_and_transceive((struct device *)dev,
                                        &config_copy,
                                        &tx_bufs_copy,
                                        &rx_bufs_copy);
}
```

```
static inline
void spi_context_buffers_setup(struct spi_context *ctx,
                               const struct spi_buf_set *tx_bufs,
                               const struct spi_buf_set *rx_bufs,
                               u8_t dfs)
{
        ...
        if (tx_bufs) {
                ctx->current_tx = tx_bufs->buffers;
                ctx->tx_count = tx_bufs->count;
                ctx->tx_buf = (const u8_t *)ctx->current_tx->buf;
                ctx->tx_len = ctx->current_tx->len / dfs;
        } else {
                ctx->current_tx = NULL;
                ctx->tx_count = 0;
                ctx->tx_buf = NULL;
                ctx->tx_len = 0;
        }

        if (rx_bufs) {
                ctx->current_rx = rx_bufs->buffers;
                ctx->rx_count = rx_bufs->count;
                ctx->rx_buf = (u8_t *)ctx->current_rx->buf;
                ctx->rx_len = ctx->current_rx->len / dfs;
        } else {
                ctx->current_rx = NULL;
                ctx->rx_count = 0;
                ctx->rx_buf = NULL;
                ctx->rx_len = 0;
        }
        ...
}
```

so I suggest to change code like below:
```
@@ -224,7 +224,7 @@ void spi_context_buffers_setup(struct spi_context *ctx,
 {
        LOG_DBG("tx_bufs %p - rx_bufs %p - %u", tx_bufs, rx_bufs, dfs);
 
-       if (tx_bufs) {
+       if (tx_bufs && tx_bufs->buffers) {
                ctx->current_tx = tx_bufs->buffers;
                ctx->tx_count = tx_bufs->count;
                ctx->tx_buf = (const u8_t *)ctx->current_tx->buf;
@@ -236,7 +236,7 @@ void spi_context_buffers_setup(struct spi_context *ctx,
                ctx->tx_len = 0;
        }
 
-       if (rx_bufs) {
+       if (rx_bufs && rx_bufs->buffers) {
                ctx->current_rx = rx_bufs->buffers;
                ctx->rx_count = rx_bufs->count;
                ctx->rx_buf = (u8_t *)ctx->current_rx->buf;

```

I'm not familiar with Zephyr spi software stack, so maybe there is something wrong, welcome any suggestions. Thanks

wentongwu, 2019-11-20:
@tbursztyka

tbursztyka, 2019-11-20:
Indeed, good catch.

That said, fixing spi_handlers.c would be better. make copy_and_check() returning a strut spi_buf_set pointer, count of buffer is 0: return NULL. 
To get then: 
`tx_bufs = copy_and_check(tx_bufs, tx_buf_copy, 0);`

And there you are: having relevant NULL pointers in the actual transceive call.

andrewboie, 2019-11-20:
> the parameters tx_bufs and rx_bufs passed between these functions are buffer pointers from stack like below, so it's very likely that these are not NULL

It looks like the code in z_vrfy_spi_transceive is wrong. When NULLs are passed in to z_vrfy_spi_transceive(), it is not passing those NULLs to the driver implementation. Instead it is passing pointers to struct spi_buf_set that have been zeroed out.

maybe something like (this is not tested in any way)
```
diff --git a/drivers/spi/spi_handlers.c b/drivers/spi/spi_handlers.c
index 6ee76668cb..b1c36e4fc1 100644
--- a/drivers/spi/spi_handlers.c
+++ b/drivers/spi/spi_handlers.c
@@ -55,11 +55,15 @@ static u32_t copy_bufs_and_transceive(struct device *dev,
                                      struct spi_buf_set *tx_bufs,
                                      struct spi_buf_set *rx_bufs)
 {
-       struct spi_buf tx_buf_copy[tx_bufs->count ? tx_bufs->count : 1];
-       struct spi_buf rx_buf_copy[rx_bufs->count ? rx_bufs->count : 1];
+       if (tx_bufs) {
+               struct spi_buf tx_buf_copy[tx_bufs->count ? tx_bufs->count : 1];
+               copy_and_check(tx_bufs, tx_buf_copy, 0);
+       }
 
-       copy_and_check(tx_bufs, tx_buf_copy, 0);
-       copy_and_check(rx_bufs, rx_buf_copy, 1);
+       if (rx_bufs) {
+               struct spi_buf rx_buf_copy[rx_bufs->count ? rx_bufs->count : 1];
+               copy_and_check(rx_bufs, rx_buf_copy, 1);
+       }
 
        return z_impl_spi_transceive((struct device *)dev, config,
                                    tx_bufs, rx_bufs);
@@ -73,6 +77,8 @@ static inline int z_vrfy_spi_transceive(struct device *dev,
        struct spi_buf_set tx_bufs_copy;
        struct spi_buf_set rx_bufs_copy;
        struct spi_config config_copy;
+       bool tx_null = false;
+       bool rx_null = false;
 
        Z_OOPS(Z_SYSCALL_MEMORY_READ(config, sizeof(*config)));
        Z_OOPS(Z_SYSCALL_DRIVER_SPI(dev, transceive));
@@ -86,7 +92,7 @@ static inline int z_vrfy_spi_transceive(struct device *dev,
                memcpy(&tx_bufs_copy, tx, sizeof(tx_bufs_copy));
                Z_OOPS(Z_SYSCALL_VERIFY(tx_bufs_copy.count < 32));
        } else {
-               memset(&tx_bufs_copy, 0, sizeof(tx_bufs_copy));
+               tx_null = true;
        }
 
        if (rx_bufs) {
@@ -98,7 +104,7 @@ static inline int z_vrfy_spi_transceive(struct device *dev,
                memcpy(&rx_bufs_copy, rx, sizeof(rx_bufs_copy));
                Z_OOPS(Z_SYSCALL_VERIFY(rx_bufs_copy.count < 32));
        } else {
-               memset(&rx_bufs_copy, 0, sizeof(rx_bufs_copy));
+               rx_null = true;
        }
 
        memcpy(&config_copy, config, sizeof(*config));
@@ -113,8 +119,8 @@ static inline int z_vrfy_spi_transceive(struct device *dev,
 
        return copy_bufs_and_transceive((struct device *)dev,
                                        &config_copy,
-                                       &tx_bufs_copy,
-                                       &rx_bufs_copy);
+                                       tx_null ? NULL : &tx_bufs_copy,
+                                       rx_null ? NULL : &rx_bufs_copy);
 }
 #include <syscalls/spi_transceive_mrsh.c>
 
```

andrewboie, 2019-11-20:
> Indeed, good catch.
>
> That said, fixing spi_handlers.c would be better. make copy_and_check() returning a strut spi_buf_set pointer, count of buffer is 0: return NULL.
> To get then:
> `tx_bufs = [...]

That should work too

wentongwu, 2019-11-20:
@tbursztyka @andrewboie Thanks, I will submit a PR for this issue.

PR #20815 (2019-11-20): spi: pass correct buffer to spi driver given NULL to spi_transceive
When pass NULL to spi_transceive with user space enabled, stack buffer
is still passed to spi driver and it will cause kinds of problems like
MPU fault, so change it to pass relevant NULL pointers in the actual
transceive call.

Also fixes a spell typo.

Fixes: #20811.
commit b82e2029 (2019-11-20): spi: pass correct buffer to spi driver given NULL to spi_transceive

When pass NULL to spi_transceive with user space enabled, stack buffer
is still passed to spi driver and it will cause kinds of problems like
MPU fault, so change it to pass relevant NULL pointers in the actual
transceive call.

Fixes: #20811.
PR #20989 (2019-12-03): [Backport 20815 to v1.14 branch]spi: pass correct buffer to spi driver given NULL to spi_transceive
When pass NULL to spi_transceive with user space enabled, stack buffer
is still passed to spi driver and it will cause kinds of problems like
MPU fault, so change it to pass relevant NULL pointers in the actual
transceive call.

Also fixes a spell typo.

Fixes: #20811.
commit 3d28c18a (2019-12-03): spi: pass correct buffer to spi driver given NULL to spi_transceive

When pass NULL to spi_transceive with user space enabled, stack buffer
is still passed to spi driver and it will cause kinds of problems like
MPU fault, so change it to pass relevant NULL pointers in the actual
transceive call.

Fixes: #20811.
closed 2019-11-20 by andrewboie