zephyrproject-rtos/zephyr #20825: stm32: dam: enable dma with peripheral using DMAMUX

erwango, 2019-11-20:
**Is your enhancement proposal related to a problem? Please describe.**
On some STM32 series, a DMAMUX IP is present to handle the peripheral DMA channels multiplexing. This new IP is not supported yet on STM32 DMA driver

**Describe the solution you'd like**
So to support use of DMA by peripherals on these series, stm32 dma driver should be updated 
to support channel configuration using DMAMUX

FRASTM, 2020-01-17:
In case the soc supports dmamux feature, for dma periph->mem or mem->periph, the dma drivers is augmented.
A new device dmamux is added and initialised
The dma configuration now includes the dmamux config. with a limited nb of parameters coming from the dma client. The dmamux channel is calculated based on the dma instance and channel (stream).
The dma start and stop are always transferred through the dmamux.
The dmamux irq handler is not yet processed in case of overrun error.

FRASTM, 2020-02-20:
see the [spi: stm32: Enable DMA client
](https://github.com/zephyrproject-rtos/zephyr/pull/22934)  for using the dma and mux for a client transfer

FRASTM, 2020-02-20:
put valid link from linux kernel in st,stm32-dma.yaml and st,stm32-dmamux.yaml

FRASTM, 2020-02-28:
Requires change in the stm32 dma driver: https://github.com/zephyrproject-rtos/zephyr/pull/22100

For stm32 with DMA MUX IP  (like stm32wb55, stm32l4r5 series ), the issue requires: 
1. the stm32 dmamux driver: https://github.com/zephyrproject-rtos/zephyr/pull/22021
2. adaptation of the client like SPI, with Device Tree configuration and application: https://github.com/zephyrproject-rtos/zephyr/pull/23155

 For stm32 with DMA (and no dmammux IP) (like stm32l476 series), the issue requires: 
1. adaptation of the client like SPI, with Device Tree configuration and application: https://github.com/zephyrproject-rtos/zephyr/pull/23157

closed 2020-05-04 by galak