zephyrproject-rtos/zephyr #20835: [Coverity CID :205797] Control flow issues in drivers/flash/spi_nor.c

zephyrbot, 2019-11-20:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/master/drivers/flash/spi_nor.c

Category: Control flow issues
Function: `delay_until_exit_dpd_ok`
Component: Drivers
CID: 205797

Please fix or provide comments in coverity using the link:

https://scan9.coverity.com/reports.htm#v32951/p12996.

dleach02, 2019-11-22:
```
104         */
   	cond_at_least: Condition since >= 0, taking true branch. Now the value of since is at least 0.
105        if (since >= 0) {
106#ifdef DT_INST_0_JEDEC_SPI_NOR_T_ENTER_DPD
107                /* Subtract time required for DPD to be reached */
108                since -= T_DP_MS;
109#endif /* T_ENTER_DPD */
110#ifdef DT_INST_0_JEDEC_SPI_NOR_DPD_WAKEUP_SEQUENCE
111                /* Subtract time required in DPD before exit */
112                since -= T_DPDD_MS;
113#endif /* DT_INST_0_JEDEC_SPI_NOR_DPD_WAKEUP_SEQUENCE */
114
115                /* If the adjusted time is negative we have to wait
116                 * until it reaches zero before we can proceed.
117                 */
   	at_least: At condition since < 0, the value of since must be at least 0.
   	dead_error_condition: The condition since < 0 cannot be true.
118                if (since < 0) {
   	
CID 205797 (#1 of 1): Logically dead code (DEADCODE)
dead_error_line: Execution cannot reach this statement: k_sleep((u32_t)-since);.
119                        k_sleep(K_MSEC((u32_t)-since));
120                }
121        }
122#endif /* DT_INST_0_JEDEC_SPI_NOR_HAS_DPD */
```

pabigot, 2019-11-22:
The code is not dead in cases where devicetree properties provide an enter delay or wakeup delay that's subtracted from the elapsed delay, potentially resulting a negative time-to-delay.

FWIW: Coverity won't let me commit the description so this will probably show up again.

dleach02, 2019-11-22:
@pabigot, before quickly closing this bug we should get some consensus. The code can clearly be dead code with the right configurations and how ever the code base was submitted to coverity it was detected as dead code. My PR should address this for compile configurations that cause this code to be dead.

Further, the detection of dead code is in coverity because there is a misra required rule that there shall be no dead code. Not sure you have access to the misra spec but the reasoning given even references compilers taking out the dead code.

pabigot, 2019-11-22:
@dleach02 I have no objections to leaving this issue open until consensus is reached.  Here's my perspective:

I was assigned this issue, so I looked into it and determined that it was a bogus result because Coverity wasn't invoked with the flags that would have reached the code.  I tried to resolve it with Coverity triage following existing Zephyr practice, but found I didn't have the permissions necessary to do so.  So I documented the result and closed the issue.

Next thing I knew I was asked to review a PR that used IMO ugly preprocessor conditionals to remove the code.  As the current primary maintainer of this module I object to putting functionally unnecessary preprocessor directives in place because the QA infrastructure fails to check the code in the ways it's used in real applications.

So, let's review some options:
* I can be given the authority (permissions) necessary to perform my assigned duties in a manner consistent with existing process.
* Somebody else can configure Coverity to ignore the error.
* The QA infrastructure could be improved so that Coverity is invoked in a way that ensures all code paths are exercised.
* We can determine that configuring Coverity to ignore this sort of diagnostic is not acceptable practice.  In that case, we probably need to review every ignore directive currently in place.
* We can leverage the fact the proposed policy of maintainer sign-off on all PRs is not yet in place, or specify that I'm not the maintainer of the module, and override my NAK.
* We can define a policy that says the desires of the maintainer are subordinate to the coding guidelines (and clearly document what "unreachable" means, because the code in this case is in fact reachable in certain circumstances).  From there we might give me the opportunity to address the issue in a way I feel is appropriate, or we could say your solution goes in just because.
* Something else might happen.

What I'm not willing to do at this point is approve #20923 in its current form.  If I had formal maintainer rights, I might try to modify it to an acceptable form, but I don't, and we don't have a process in place for transferring responsibility for PRs from submitters to (informal) maintainers.

I do have multiple versions of MISRA, for both C and C++.  Last I heard it still hasn't been agreed which rules are going to apply to what code.  I think it's reasonable to get that settled before making changes like this.

dleach02, 2019-11-22:
I understand that you didn't like the PR. I suspect the easy thing would have been to just comment on the PR the change that would have been acceptable since you are suggesting there would have been a change if you had the rights to modify my PR. I can make the change and update the PR to get your acceptance if that is an option.

Note that the "ugliness" started with the fact that there are already multiple levels of #ifdef in that file. That is not my doing. All I did was wrap the code being flagged with the combination of previous conditionals, which apparently are not defined in the particular run in coverity.

Your point about which rules in misra we want to apply is correct. But note that this applies for both the disposition of coverity warnings as it does for code changes. If we mark in coverity that this particular code is to be ignored or false positive then it will always be that way unless we make a concerted effort to review these dispositions at some future date.

pabigot, 2019-11-22:
If we impose this policy and do *not* make a concerted effort to review existing dispositions then we're checking policy boxes rather than following the spirit and intent of the policy.

It's a poor use of both your time and mine for me to explain and justify what I would want an alternative approach to look like, then for you to implement it.  When I've addressed the critical timer failures and potential race conditions in Nordic clock start/stop I'll figure out what I can live with and provide an alternative I'm willing to have merged.  I don't guarantee that'll happen for 2.1.

The simplest approach at this point is to not consider this a medium priority bug.  Without documented criteria for priority I'm not clear how it became that level; it certainly isn't a functional risk.

dleach02, 2019-11-22:
I agree it is not a functional risk. I believe the current criteria of the script being run is to create issues as medium priority except for items found in tests and samples (which get low). After triage, the priority can be adjusted up/down or the issue closed. 

I believe we are at least in agreement that this is a lower priority item so I will lower it.

PR #23392 (2020-03-19): drivers: flash: spi_nor: avoid coverity warning in untested code
Coverity doesn't invoke the build for this module with a configuration
that provides a non-zero delay for entering or releasing deep power
down, so it sees a check that a value known to be non-negative
magically became negative without being touched, and so screams about
dead code.

Refactor to make the adjustments unconditional which has the same
effect in any reasonable level of optimization while making it
slightly less obvious that the value may not change.

Fixes #20835
closed 2020-03-19 by nashif