zephyrproject-rtos/zephyr #20837: [Coverity CID :205777] Error handling issues in drivers/usb/device/usb_dc_native_posix.c

zephyrbot, 2019-11-20:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/master/drivers/usb/device/usb_dc_native_posix.c

Category: Error handling issues
Function: `usb_dc_ep_write`
Component: Drivers
CID: 205777

Please fix or provide comments in coverity using the link:

https://scan9.coverity.com/reports.htm#v32951/p12996.

dleach02, 2019-11-20:
`353        if (USBIP_EP_ADDR2IDX(ep) == 0) {
   	CID 205776: Unchecked return value from library (CHECKED_RETURN) [select issue]
354                usbip_send_common(ep, data_len);
   	
CID 205777 (#1 of 1): Unchecked return value from library (CHECKED_RETURN)
8. check_return: Calling usbip_send(ep, data, data_len) without checking return value. It wraps a library function that may fail and return an error code. [show details]
355                usbip_send(ep, data, data_len);
356        } else {`

If the underlying code doesn't care about the return value then putting a (void) before the function invocation would show that the author knows that there is a return value but that the value is not needed (and it will remove the warning from coverity).

dleach02, 2019-11-27:
@finikorg, will you be posting a PR for this issue anytime soon?

dleach02, 2019-11-27:
#20929

PR #20929 (2019-11-29): USB: Coverity fixes
Fixing coverity bugs

Fixes #20842
Fixes #20836
Fixes #20837
Fixes #20838
Fixes #20839
Fixes #20841
Fixes #20840
commit 5b22c881 (2019-11-29): usb: usb_dc_native_posix: Check return values in dc_write()

Fixes coverity issue:

Fixes #20836
Fixes #20837
closed 2019-11-29 by nashif