zephyrproject-rtos/zephyr #20843: [Coverity CID :205778] Integer handling issues in include/sys/time_units.h

zephyrbot, 2019-11-20:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/master/include/sys/time_units.h

Category: Integer handling issues
Function: `z_tmcvt`
Component: Other
CID: 205778

Please fix or provide comments in coverity using the link:

https://scan9.coverity.com/reports.htm#v32951/p12996.

andyross, 2019-11-20:
False positive.  Coverity is warning that a value is being produced with 32 bit math but then assigned to a 64 bit result, that is it doesn't like to see an expression like:

```c
u64_t z = (u32_t)x / (u32_t)y;
```

But that's exactly what's desired here.  The generator function for the time unit conversions can be used in a 64 bit or 32 bit context (with a downcast where it's used), and I want the compiler to be able to see that the result is guaranteed to be 32 bits wide to **prevent** the attempt to add excess precision.

This may take some struggling to fix.  Naively addressing the complaint in z_tmcvt() is going to produce performance regressions, and giving coverity what it wants is going to be isomorphic to fully expanding all the function variants and abandoning the trick that allows the optimizer to do it for us.

I suggest whitelisting this one for now.

nashif, 2019-11-21:
close in coverity.

closed 2019-11-21 by nashif