zephyrproject-rtos/zephyr #20844: [Coverity CID :205781] Integer handling issues in lib/os/printk.c

zephyrbot, 2019-11-20:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/master/lib/os/printk.c

Category: Integer handling issues
Function: `z_vprintk`
Component: Other
CID: 205781

Please fix or provide comments in coverity using the link:

https://scan9.coverity.com/reports.htm#v32951/p12996.

andyross, 2019-11-20:
Some quick triage if someone else wants to work this.  This is a "test is always true" kind of error that happens to be platform dependent.  The code looks like this ("lld" has a type of "long long"):

    if (lld > __LONG_MAX__ ||
        lld < ~__LONG_MAX__) {
            print_err(out, ctx);
            break;
    }

On a 32 bit platform, sizeof(long) < sizeof(long long), so it's possible to have a value out of a 32 bit range (which apparently we don't have code to handle, thus the error).  On 64 bit platforms, long and long long are the same basic type, and it's literally impossible to have a value out of bounds.  The code is doing what it's supposed to do, basically.

Fix this by putting that inside a #if that includes it only or 32 bit targets, or just whitelist the coverity case.

dleach02, 2020-07-24:
This looks like it was fixed as a side affect of change 973487fdad1b72

closed 2020-08-15 by dleach02