zephyrproject-rtos/zephyr #20869: [Coverity CID :205811] Memory - corruptions in subsys/net/lib/lwm2m/lwm2m_engine.c

zephyrbot, 2019-11-20:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/master/subsys/net/lib/lwm2m/lwm2m_engine.c

Category: Memory - corruptions
Function: `lwm2m_init_message`
Component: Networking
CID: 205811

Please fix or provide comments in coverity using the link:

https://scan9.coverity.com/reports.htm#v32951/p12996.

dleach02, 2019-11-22:
I believe this is another case of mixing 32bit with 64bit architectures and Coverity getting confused. The complaint is in the called routine where there is a memset() of the structure in question using the sizeof(structure)... don't see how that is a problem unless you take into account that a 64bit architecture will use a different alignment of the fields in the structure and the pointer field is twice as big.

nashif, 2019-11-25:
Please do not remove bug label, if this is not a bug, then it should be closed.

dleach02, 2019-11-25:
Okay. I have not been in the habit of closing these bugs. I think it would be appropriate to debug the process in which we submit projects to coverity because this is the fourth such item where it appears there is a mixing of architectures that is confusing coverity.

jukkar, 2019-11-26:
False positive.

closed 2019-11-26 by jukkar