zephyrproject-rtos/zephyr #20870: [Coverity CID :205816] Control flow issues in subsys/settings/src/settings_file.c

zephyrbot, 2019-11-20:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/master/subsys/settings/src/settings_file.c

Category: Control flow issues
Function: `read_handler`
Component: Other
CID: 205816

Please fix or provide comments in coverity using the link:

https://scan9.coverity.com/reports.htm#v32951/p12996.

dleach02, 2019-11-21:
`
        ssize_t r_len;
436        int rc;
437
438        /* 0 is reserved for reding the length-field only */
439        if (entry_ctx->len != 0) {
440                if (off >= entry_ctx->len) {
441                        *len = 0;
442                        return 0;
443                }
444
445                if ((off + *len) > entry_ctx->len) {
446                        *len = entry_ctx->len - off;
447                }
448        }
449
450        rc = fs_seek(file, entry_ctx->seek + off, FS_SEEK_SET);
451        if (rc) {
452                goto end;
453        }
454
455        r_len = fs_read(file, buf, *len);
456
   	
CID 205816 (#1 of 1): Unsigned compared against 0 (NO_EFFECT)
unsigned_compare: This greater-than-or-equal-to-zero comparison of an unsigned value is always true. r_len >= 0UL.
457        if (r_len >= 0) {
458                *len = r_len;
459                rc = 0;
460        } else {
461                rc = r_len;
462        }
463end:
`

dleach02, 2019-11-21:
@nashif, Needs to be marked as a false positive in coverity. Although I don't know why it thinks the ssize_t values are unsigned values.

nvlsianpu, 2019-11-22:
confirm, it is false-positive.

nashif, 2019-11-25:
@dleach02  Please do not remove the bug label, if it is not a bug, then close it.

dleach02, 2019-11-25:
@nashif, it is not a bug. But since we don't have a clear way to feed this information back to coverity I did not want to close this issue until we could potentially actually debug our coverity process. In the future, I will only move the priority level to ensure this doesn't become a release blocker.

dleach02, 2020-07-24:
closing due to false positive determination

closed 2020-07-24 by dleach02