zephyrproject-rtos/zephyr #20877: [Coverity CID :205823] Null pointer dereferences in tests/kernel/fifo/fifo_timeout/src/main.c

zephyrbot, 2019-11-20:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/master/tests/kernel/fifo/fifo_timeout/src/main.c

Category: Null pointer dereferences
Function: `test_multiple_threads_pending`
Component: Tests
CID: 205823

Please fix or provide comments in coverity using the link:

https://scan9.coverity.com/reports.htm#v32951/p12996.

andyross, 2019-11-20:
False positive.  This is the "volatile variable reads are side effects" case (i.e. they might be MMIO registers, etc...), coverity is seeing a line like:

    zassert_true(volatile_var == val, "");

And warning that this is illegal because if that assertion is not enabled then the read won't happen and behavior will be different.  (And yes, the volatile is needed: it's a spinlock test.)

Except it's a false positive.  This is not a compile-time-configurable assertion routine.  It just has "assert" in its name.  This macro from ztest_assert.h is **always** expanded into a call to z_zassert(), which is a real C function with a real body that causes the compiler to evaluate its arguments 100% of the time, regardless of build configuration.

I fixed a few of these last release with some temporary varaibles, but if this is going to be a regular thing we really should get a whitelist added to coverity (or, frankly, just change the name of the function).  Not sure who to give that to.

andyross, 2019-11-20:
Triage for whoever's benefit: this one is real enough.

test_multiple_threads_pending() does a k_fifo_get() to retrieve a data pointer and then dereferences it to log its fields in a TC_PRINT() at line 162.  The test is making assumptions about the data environment to assume that the get never returns a NULL, but it technically can and coverity wants that case checked.

It's probably enough to put a `z_assert_true(data != NULL, "");` ahead of the use, though it might be safer to wrap the dereferences in an if() instead.

PR #21085 (2019-12-10): tests: fifo: assert of NULL from k_fifo_get
Fix Null pointer dereferences.

Fixes #20877
Coverity CID: 205823
commit 5d94b895 (2019-12-10): tests: fifo: assert of NULL from k_fifo_get

Fix Null pointer dereferences.

Fixes #20877
Coverity CID: 205823
closed 2019-12-10 by nashif