zephyrproject-rtos/zephyr #20878: [Coverity CID :205801] Incorrect expression in tests/kernel/mp/src/main.c

zephyrbot, 2019-11-20:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/master/tests/kernel/mp/src/main.c

Category: Incorrect expression
Function: `test_mp_start`
Component: Tests
CID: 205801

Please fix or provide comments in coverity using the link:

https://scan9.coverity.com/reports.htm#v32951/p12996.

andyross, 2019-11-20:
(Pasting same answer from elsewhere).

False positive.  This is the "volatile variable reads are side effects" case (i.e. they might be MMIO registers, etc...), coverity is seeing a line like:

    zassert_true(volatile_var == val, "");

And warning that this is illegal because if that assertion is not enabled then the read won't happen and behavior will be different.  (And yes, the volatile is needed: it's a mp test.)

Except it's a false positive.  This is not a compile-time-configurable assertion routine.  It just has "assert" in its name.  This macro from ztest_assert.h is **always** expanded into a call to z_zassert(), which is a real C function with a real body that causes the compiler to evaluate its arguments 100% of the time, regardless of build configuration.

I fixed a few of these last release with some temporary varaibles, but if this is going to be a regular thing we really should get a whitelist added to coverity (or, frankly, just change the name of the function).  Not sure who to give that to.

nashif, 2019-11-29:
closed in coverity as false positive

closed 2019-11-29 by nashif