zephyrproject-rtos/zephyr #20892: our nRF52840 board power management sleep duration

atKeitaroTakahashi, 2019-11-21:
I want to make our nRF52840 board by zephyr-v2.0.0 work into low power mode for specified seconds at k_sleep.
But our board often sleeps 512 seconds longer than specified seconds if we specified time longer 512 seconds.
For example, if we specified 600 seconds, our board often sleeps 1112 seconds.
I only understand this probrem is related nRF52 RTC overflow.
What's the difference when sleeped specified seconds and sleeped 512 seconds longer?
Just in case I tried zephyr-v2.1.0-rc1, but the situation does not change.
Incidentally, consumption when our board sleeped is 5uA.

pabigot, 2019-11-21:
Does your application stay completely asleep for the entire 600/1112 second period, or does it have to wake up to do other things like wireless communication, react to external signals, get timer callbacks?  If it wakes, how frequently?

pabigot, 2019-11-22:
Confirmed: a sleep of 600 s (10:00 m:s) on Nordic nrf52840_pca10056 will not wake until 18:32 m:s external time has passed.  The Zephyr uptime clock will show 600 s elapsed.

The bug is not present on nucleo_l476rg so it's likely to be specific to the Nordic timer infrastructure.

atKeitaroTakahashi, 2019-11-22:
> Does your application stay completely asleep for the entire 600/1112 second period,

Our application stay completely asleep for the entire 600/1112 second period.

> If it wakes, how frequently?

I tried 136 times, problem happened 103 times.
103/136: wake up on 1112 seconds.
33/136: wake up on 600 seconds.

> The Zephyr uptime clock will show 600 s elapsed.

We confirmed it as well. 
I was relieved it isn't inherent probrem on our board.

carlescufi, 2019-11-22:
@nordic-krch can you check if this can be something related to our system timer driver?

nordic-krch, 2019-11-22:
i will try to investigate

pabigot, 2019-11-22:
I'm on this.  I know what the problem is, and it probably applies to other boards.

nordic-krch, 2019-11-22:
ok

PR #20940 (2019-11-26): drivers: timer: nrf_rtc_timer: fix lost ticks from unannounced elapsed
The original code assumed that limiting the tick count to the maximum
cycle value representable without wrapping would guarantee that adding
the resulting cycle offset to last_count would not lap the counter.
This is not true when elapsed time, which is also added to the cycle
offset, exceeds one tick.  Cap the maximum offset at the number of
cycles corresponding to the maximum number of ticks without wrapping.

Fixes #20892 and is the Nordic part of #20939.
closed 2019-11-26 by nashif