zephyrproject-rtos/zephyr #20915: doc: Kconfig section in board_porting.rst should be moved or removed

erwango, 2019-11-21:
**Describe the bug**

An important piece of Kconfig documentation sit under doc/guides/porting/board_porting.rst
Another part of Kconfig doc is in doc/guides/kconfig/index.rst but does not reference the other part.

I think most of Kconfig relative doc in board_porting section should be moved in actual kconfig guide. 
Some part might be cleaned as well ('Old Zephyr Kconfig behavior for defaults') for instance.

ulfalizer, 2019-11-21:
Yeah, another iffy thing at the moment is that the Kconfig tips page has turned into a dumping ground for random Kconfig documentation (like documentation for the `kconfigfunctions.py` macros). Might be nice to keep the "generic" Kconfig stuff separate.

erwango, 2019-11-22:
> Might be nice to keep the "generic" Kconfig stuff separate.

I'd say: better to put everything in one place (`guides/Kconfig`) but make use of different pages  (General, tips, scripts, ..)

aescolar, 2019-11-22:
I reclassify this as an enhancement, as it is a request for an improvement to the documentation usability, independently of how welcome it may be, and not a bug that prevents using the system or a reason to gate releases.

ulfalizer, 2019-11-25:
Working on this now.

ulfalizer, 2019-12-04:
Done now. See https://github.com/zephyrproject-rtos/zephyr/pull/21133, which moves lots of Kconfig documentation into its own dedicated section in User and Developer Guides, with links from e.g. Application Developmend the board porting guide.

commit e32686d0 (2019-12-09): doc: kconfig: Put all documentation on setting symbols on a new page

Add a new 'Setting Kconfig configuration values' page in the new Kconfig
section in 'User and Developer Guides'. Move all the information on
setting Kconfig symbols from 'Application Development' and the board
porting guide into it. The same page now covers both configuration files
and Kconfig.defconfig files.

Add links to the new page in various places to make it easy to find.
Also add some more references to the top-level Kconfig page and other
Kconfig pages.

A lot of stuff was rewritten while moving it over (the CONF_FI [... cut]
closed 2019-12-09 by carlescufi