zephyrproject-rtos/zephyr #20930: Linking static library with use of zephyr_link_libraries(${PATH}) may end with undefined reference error

ppryga, 2019-11-22:
**Describe the bug**
If one wants to add static library to be linked with Zephyrs application, she may use one of two following functions provided by Zephyrs cmake environment:
1. zephyr_link_libraries
2. zephyr_library_link_libraries

Use of first one may end with undefined reference error.
In case of use of zephyr_link_libraries, an added library is appended to list of zephyr.elf dependencies (placed at the end of the list). If that library requires some symbol from e.g. libgcc.a that is linked earlier and the symbol is not required before libgcc processing, then it is added to undefined symbols table and never resolved. 
Use of zephyr_library_link_libraries does not end with such error, because an added library is inserted into zephyr.elf link dependencies earlier. 

**To Reproduce**
1. Build static library that uses e.g. __aeabi_fmul defined in libgcc.a and exports symbol that will be used by final binary.
2. Add static library to final binary link list with use of zephyr_link_libraries. Use exported function implemented in the static library (it is required to force linker to add the library to linkage process and add e.f. __aeabi_fmul to undefined symbols list).
3. Build the final application.
4. See error in link step.

**Expected behavior**
When use of zephyr_link_libraries there should not be a link error at the end of the day.

**Impact**
At the moment it is possible to use zephyr_library_link_libraries (but it can't be used in CMakeList.txt that is added by add_subdirectory: Attempt to add link library "lib_some_library.a" to target "app" which is not built in this directory. This is allowed only when policy CMP0079 is set to NEW.) **But pay attention that it may cause to fail of builds that use static libraries if a library changes introduce use of not yet resolved symbol. It may be difficult to find out for end user because change of a static library causes problem with final binary linking.** 

**Environment (please complete the following information):**
 - Toolchain: zephyr and gnuarmeabi
 - found with Zephyr >= 2.0

aescolar, 2019-11-22:
@SebastianBoe

SebastianBoe, 2019-11-22:
When using zephyr_link_libraries to add a library to the link command every library that it depends on must be listed after it. Like we do here:

https://github.com/zephyrproject-rtos/zephyr/blob/master/lib/libc/newlib/CMakeLists.txt#L39

If for some reason this is not acceptable the user can insert the library into the whole-archive group using the 'zephyr_library_import' API.

ppryga, 2019-11-22:
Ok, if that is expected behavior then it shoudl be somwhere documented. I did not spot that information anywhere. Also I suggest to move common libraries like gcc to the end of linkage list to avoid re-linking the same library multiple times.

SebastianBoe, 2019-11-22:
> Also I suggest to move common libraries like gcc to the end of linkage list to avoid re-linking the same library multiple times.

I believe this is impossible unfortunately, but if someone were to figure out how to do so then this patch would be welcome.

> Ok, if that is expected behavior then it should be somewhere documented. 

A contribution to the documentation would be welcome.

SebastianBoe, 2019-11-22:
Re-categorizing this as an enhancement to documentation.

faizurrehman, 2024-02-22:
is there any update (possible solution) on this issue?