zephyrproject-rtos/zephyr #20931: intel_s1000_crb samples can't be built with latest master

jhedberg, 2019-11-22:
**Describe the bug**
Building the intel_s1000_crb samples fail with the latest master using the xcc toolchain (the only toolchain that's supported for this board and these samples). Building works fine with the Zephyr 2.0 release. I tried bisecting but it seems the build has broken multiple times since 2.0 in different ways. This is not really a new thing (I've filed similar bugs in the past), and not surprising, since CI doesn't exercise this toolchain. 

**To Reproduce**
Steps to reproduce the behavior:
1. Install xcc toolchain
2. ZEPHYR_TOOLCHAIN_VARIANT=xcc
3. `west build -b intel_s1000_crb samples/boards/intel_s1000_crb/dmic`

The same steps work fine with v2.0.0.

**Expected behavior**
Builds sucessfully

**Impact**
Can't use this toolchain or this board

dleach02, 2019-11-22:
is there an issue with adding this toolchain to the SDK and enabling CI on this platform?

dcpleung, 2019-11-22:
Logs?

jhedberg, 2019-11-22:
> Logs?

@dcpleung are you having trouble reproducing? I didn't provide logs since I got different errors in different places of the commit timeline between 2.0 and latest master. It should be super simple to reproduce though.

jhedberg, 2019-11-22:
@dcpleung btw, even though I assigned this to you (since you're in CODEOWNERS for xtensa boards) we might end up reassigning this to some build-system person - at least that's what happened last time when similar breakage happened.

dcpleung, 2019-11-22:
XCC no longer runs on any of my machines, as it keeps segfaulting. Having logs from you will give me some time to think about it while I setup a VM with older distro to try.

dcpleung, 2019-11-22:
> is there an issue with adding this toolchain to the SDK and enabling CI on this platform?

XCC is locked to the ethernet MAC address so it would be problematic to enable it on shippable. However, we have a GCC toolchain for that SoC in SDK v0.11. When we move to the new SDK, we can have CI building the code.

jhedberg, 2019-11-22:
> XCC is locked to the ethernet MAC address so it would be problematic to enable it on shippable. However, we have a GCC toolchain for that SoC in SDK v0.11. When we move to the new SDK, we can have CI building the code.

That'd be a nice improvement, but it still won't prevent breakage with xcc though, and there will be people needing & using xcc over gcc even when the latter is available.

jhedberg, 2019-11-22:
With current master I get this:
```
-- Configuring incomplete, errors occurred!
See also "/home/jh/src/zephyr/zephyr/build/CMakeFiles/CMakeOutput.log".
See also "/home/jh/src/zephyr/zephyr/build/CMakeFiles/CMakeError.log".
ERROR: command exited with status 1: /usr/bin/cmake -B/media/psf/Home/src/zephyr/zephyr/build -S/media/psf/Home/src/zephyr/zephyr/samples/boards/intel_s1000_crb/dmic -GNinja -DBOARD=intel_s1000_crb
```
in CMakeError.log I see this at the end:
```
xt-xcc WARNING:  -V is ignored
xt-xcc ERROR:  no source or object file given
```

Starting a `git bisect` operation fails on the first commit somewhere between 2.0 and master with this:
```
[8/123] Building C object zephyr/CMakeFiles/offsets.dir/arch/xtensa/core/offsets/offsets.c.obj
FAILED: zephyr/CMakeFiles/offsets.dir/arch/xtensa/core/offsets/offsets.c.obj 
ccache /home/jh/xtensa/XtDevTools/install/tools/RF-2016.4-linux/XtensaTools/bin/xt-xcc -DBUILD_VERSION=zephyr-v2.0.0-1096-g3423d8b2a484 -DKERNEL -D_FORTIFY_SOURCE=2 -D__PROGRAM_START -D__ZEPHYR__=1 -I../kernel/include -I/media/psf/Home/src/zephyr/zephyr/arch/xtensa/include -I../include -I../include/drivers -Izephyr/include/generated -I/media/psf/Home/src/zephyr/zephyr/soc/xtensa/intel_s1000 -I/media/psf/Home/src/zephyr/zephyr/soc/xtensa/intel_s1000/include -isystem ../lib/libc/minimal/include -Os -imacros/home/jh/src/zephyr/zephyr/build/zephyr/include/generated/autoconf.h -ffreestanding -fno-common -g -fms-extensions -imacros/media/psf/Home/src/zephyr/zephyr/include/toolchain/xcc_missing_defs.h -imacros/media/psf/Home/src/zephyr/zephyr/include/toolchain/zephyr_stdint.h -Wall -Wformat -Wformat-security -Wno-format-zero-length -Wno-main -Wno-pointer-sign -Wpointer-arith -Werror=implicit-int -fno-pic -fno-strict-overflow -ffunction-sections -fdata-sections -mlongcalls -std=c99 -MD -MT zephyr/CMakeFiles/offsets.dir/arch/xtensa/core/offsets/offsets.c.obj -MF zephyr/CMakeFiles/offsets.dir/arch/xtensa/core/offsets/offsets.c.obj.d -o zephyr/CMakeFiles/offsets.dir/arch/xtensa/core/offsets/offsets.c.obj   -c /home/jh/src/zephyr/zephyr/arch/xtensa/core/offsets/offsets.c
In file included from ../include/toolchain/xcc.h:10,
                 from ../include/toolchain.h:37,
                 from ../kernel/include/gen_offset.h:74,
                 from /home/jh/src/zephyr/zephyr/arch/xtensa/core/offsets/offsets.c:26:
../include/toolchain/gcc.h:46:2: error: #error "__BYTE_ORDER__ is not defined and cannot be automatically resolved"
ninja: build stopped: subcommand failed.
ERROR: command exited with status 1: /usr/bin/cmake --build /media/psf/Home/src/zephyr/zephyr/build
```

nashif, 2019-11-25:
not fully fixed yet

PR #20955 (2019-11-25): dts: tell compiler the name of dependency file for output
When generating the dependency file for DTS (${BOARD}.dts.pre.d),
some toolchains would use the source file for the file name stem.
So, the resulting dependency file is empty_file.d instead of one
with the board name. Fix this by passing -MF to explicitly tell
the compiler the name of the dependency file.

Partially fixes #20931
PR #21018 (2019-12-02): Fix build intel_s1000_crb samples with XCC
This fixes manually building the samples under `samples/boards/intel_s1000_crb/` with XCC.

Note that XCC is based on GCC 4.2.0 and with prebuilt libraries.

Fixes #20931
closed 2019-12-02 by nashif