zephyrproject-rtos/zephyr #20939: long duration timeouts can cause loss of time

pabigot, 2019-11-22:
Multiple timer drivers work by recording a counter value at the point ticks were last announced, and tracking elapsed time since that event to get the current time.  Deadlines for new events are calculated by adding the requested duration to the elapsed time and setting a compare event at the last announce time plus this offset.  (With rounding/conversion from ticks to cycles.)

Several driver implementations try to avoid counter overflows by limiting the maximum number of ticks to a value that can be represented without overflowing the cycle counter span.  This worked in pre-tickless builds because the elapsed time was always less than one tick.

It fails in tickless because when the elapsed time is added to the converted maximum ticks it can cause a counter wrap.

This is the problem underlying #20892.  It's also present in hpet and systick, and probably multiple other drivers.

pabigot, 2019-11-25:
Here's the deep analysis for the problem.  The background is for Nordic; the problem is cross-platform.

## Background

The Nordic timer infrastructure uses a 24-bit cycle counter that increments at 32 KiHz.  The counter has a compare interrupt; when the interrupt fires the difference in ticks between the previous fire and the current fire is added to a counter and the ticks are announced.

The implementation maintains a 32-bit value that represents the low 32 bits of an extended cycle counter that advances by a multiple of cycles-per-tick to track the cycle count corresponding to the most recently announced tick.

The elapsed time since the last announced tick is calculated by a 24-bit unsigned subtraction of the extended cycle counter.

The current time is the sum of the extended cycle counter plus the elapsed time.

## Analysis

Let T be the elapsed time since system start measured in cycles, measured objectively.

Since we're in a tickless system we actually track L (number of announced ticks) and need to measure E (number of ticks since the last announcement).  When the system is correct T=L+E.

Let C be the counter span; i.e. to the counter X == (X+C) for X in [0, C).

Let M be the maximum duration that can be accommodated when scheduling a timeout (M < C).  In a timer driver where cycle rate is equal to tick rate M = C-2.

Any duration greater than M must be reached in steps that do not exceed M cycles so we avoid losing track of a cycle counter wrap.

Let D be the desired duration of an alarm measured in cycles.

So an alarm D cycles after T should be scheduled at A = T+R, where:
* R = D if D <= M; otherwise
* R = M.

We shall see that this rule is incorrect because it does not take into account a wrap event in cases where E>0.

Let's run an example where the user wants a relative alarm to occur in D = C+1000 ticks at a point where the last tick event was at time L = 1 and there have been E = 663 unannounced ticks since that event.

The desired absolute time for the alarm is 664+C+1000 = 1664+C.  This is too long, so the first step is to delay M=C-2 ticks:

    L     E     T         D        R       A
    1     663   664       C+1000   C-2     C+662

The counter will lap and read 662 in C-2 cycles.  The interrupt handler will determine that 662-L = 661 cycles have passed.  This adjustment added to L and subtracted from D so the new D is C+339.  In this case there are no unaccounted-for ticks.

Note that at this point L undercounts T by C ticks, which is an error. What follows

    L     E     T         D        R       A
    662   0     C+662     C+339    C-2     C+660

The counter will lap and read 660 in C-2 cycles.  The interrupt handler will determine that 660-L = -2 so C-2 cycles have passed. This adjustment added to L and subtracted from D so the new D is 337.

    L     E     T         D        R       A
    C+660 0     2*C+660   337      337     997

The counter will not lap, and will read 997 in 337 cycles.  The interrupt will determine that 997-660 = 337 so 337 cycles have passed. This adjustment is added to L and subtracted from D so the new D is zero.

    L     E     T          D
    C+997 0     2*C+997    0

The alarm that was requested to fire in C+1000 ticks (C+1664 absolute) will have instead fired in 2\*C-333 ticks (2\*C+997 absolute).

## Root Cause

The implementation is expected to avoid lost overflow events by this rule: 

> So an alarm D cycles after T should be scheduled at A = T+R, where:
> * R = D if D <= M; otherwise
> * R = M.

The rule is correct, but for cases where E is not zero must be stated as:

So an alarm D cycles after T=L+E should be scheduled at A = L+R, where:
* R = E+D if E+D <= M; otherwise
* R = M.

In implementations the incoming duration is measured in ticks.  The maximum number of ticks that can be supported is whatever produces M cycles (M = ticks * cycles/tick). The original implementation assumed that capping the duration at that maximum number of ticks was sufficient to guarantee that D would not exceed M, so only the first case needed to be handled.

Instead, when E is non-zero the actual upper bound is M-E, which can't be satisfied simply by adjusting ticks on entry.  Consequently the bound must be validated against E+D.

nordic-krch, 2019-11-25:
@pabigot thanks a lot for your explanation, now i understand the issue.

PR #21017 (2019-11-27): drivers/timer: Clamp after tick adjustment, not before
Some early tickless drivers had a common pattern where they would
compute a tick maximum for the request (i.e. the maximum the hardware
counter can handle) but apply it only on the input tick value and not
on the adjusted final value, opening up the overflow condition it was
supposed to have prevented.

Fixes #20939 (Strictly it fixes the specific pattern that was
discovered in that bug.  It's not impossible that other drivers with
alternative implementations have a similar issue, though they look OK
to me via a quick audit).
commit 5f63c9d9 (2019-11-27): drivers/timer: Clamp after tick adjustment, not before

Some early tickless drivers had a common pattern where they would
compute a tick maximum for the request (i.e. the maximum the hardware
counter can handle) but apply it only on the input tick value and not
on the adjusted final value, opening up the overflow condition it was
supposed to have prevented.

Fixes #20939 (Strictly it fixes the specific pattern that was
discovered in that bug.  It's not impossible that other drivers with
alternative implementations have a similar issue, though they look OK
to me via a quick audit).
closed 2019-11-27 by carlescufi