zephyrproject-rtos/zephyr #20953: usb: nrf: usb on reel board becomes unavailable if USB cable is not connected at first

jli157, 2019-11-23:
**Describe the bug**
Running the sample samples/subsys/hid on reel_board(v1507.1), based on the master branch (head is 7aa2355af2d8b735351c5def9495e40e453c749b).  At the beginning, the USB cable is not connected to the nRF USB port. 

After booting up, endless messages come like below, and the USB can't be enumerated after plugging in the USB cable. 

```
***** Booting Zephyr OS build v2.1.0-rc1-247-g7aa2355af2d8 *****
[00:00:00.004,638] <dbg> main.composite_pre_init: HID Device: dev 0x20003798
[00:00:00.004,943] <dbg> main.main: Starting application
[00:00:00.005,889] <dbg> main.status_cb: status 2 unhandled
[00:00:00.009,002] <dbg> main.status_cb: status 4 unhandled
[00:00:00.051,208] <dbg> main.status_cb: status 2 unhandled
[00:00:00.054,321] <dbg> main.status_cb: status 4 unhandled
[00:00:00.108,459] <dbg> main.status_cb: status 2 unhandled
[00:00:00.111,572] <dbg> main.status_cb: status 4 unhandled
[00:00:00.170,928] <dbg> main.status_cb: status 2 unhandled
[00:00:00.174,041] <dbg> main.status_cb: status 4 unhandled
[00:00:00.231,689] <dbg> main.status_cb: status 2 unhandled
[00:00:00.234,802] <dbg> main.status_cb: status 4 unhandled
[00:00:00.290,710] <dbg> main.status_cb: status 2 unhandled
[00:00:00.293,823] <dbg> main.status_cb: status 4 unhandled
[00:00:00.344,055] <dbg> main.status_cb: status 2 unhandled
[00:00:00.347,167] <dbg> main.status_cb: status 4 unhandled
[00:00:00.391,906] <dbg> main.status_cb: status 2 unhandled
[00:00:00.395,019] <dbg> main.status_cb: status 4 unhandled
[00:00:00.448,364] <dbg> main.status_cb: status 2 unhandled
[00:00:00.451,446] <dbg> main.status_cb: status 4 unhandled
[00:00:00.512,634] <dbg> main.status_cb: status 2 unhandled
[00:00:00.515,747] <dbg> main.status_cb: status 4 unhandled
[00:00:00.573,120] <dbg> main.status_cb: status 2 unhandled
[00:00:00.576,232] <dbg> main.status_cb: status 4 unhandled
[00:00:00.633,056] <dbg> main.status_cb: status 2 unhandled
[00:00:00.636,169] <dbg> main.status_cb: status 4 unhandled
[00:00:00.689,270] <dbg> main.status_cb: status 2 unhandled
[00:00:00.692,382] <dbg> main.status_cb: status 4 unhandled
[00:00:00.737,762] <dbg> main.status_cb: status 2 unhandled
[00:00:00.740,875] <dbg> main.status_cb: status 4 unhandled
[00:00:00.800,170] <dbg> main.status_cb: status 2 unhandled
[00:00:00.803,283] <dbg> main.status_cb: status 4 unhandled
[00:00:00.862,945] <dbg> main.status_cb: status 2 unhandled
[00:00:00.866,027] <dbg> main.status_cb: status 4 unhandled
[00:00:00.923,217] <dbg> main.status_cb: status 2 unhandled
[00:00:00.926,300] <dbg> main.status_cb: status 4 unhandled
[00:00:00.982,849] <dbg> main.status_cb: status 2 unhandled
[00:00:00.985,931] <dbg> main.status_cb: status 4 unhandled
[00:00:01.036,132] <dbg> main.status_cb: status 2 unhandled
[00:00:01.039,245] <dbg> main.status_cb: status 4 unhandled
[00:00:01.084,716] <dbg> main.status_cb: status 2 unhandled
[00:00:01.087,829] <dbg> main.status_cb: status 4 unhandled
[00:00:01.144,317] <dbg> main.status_cb: status 2 unhandled
[00:00:01.147,399] <dbg> main.status_cb: status 4 unhandled
[00:00:01.198,730] <dbg> main.status_cb: status 2 unhandled
[00:00:01.201,843] <dbg> main.status_cb: status 4 unhandled
[00:00:01.258,911] <dbg> main.status_cb: status 2 unhandled
[00:00:01.261,993] <dbg> main.status_cb: status 4 unhandled
[00:00:01.316,772] <dbg> main.status_cb: status 2 unhandled
[00:00:01.319,854] <dbg> main.status_cb: status 4 unhandled
[00:00:01.367,279] <dbg> main.status_cb: status 2 unhandled
[00:00:01.370,391] <dbg> main.status_cb: status 4 unhandled
[00:00:01.415,649] <dbg> main.status_cb: status 2 unhandled
[00:00:01.418,762] <dbg> main.status_cb: status 4 unhandled
[00:00:01.470,428] <dbg> main.status_cb: status 2 unhandled
[00:00:01.473,541] <dbg> main.status_cb: status 4 unhandled
[00:00:01.528,869] <dbg> main.status_cb: status 2 unhandled
[00:00:01.531,982] <dbg> main.status_cb: status 4 unhandled
[00:00:01.587,219] <dbg> main.status_cb: status 2 unhandled
[00:00:01.590,301] <dbg> main.status_cb: status 4 unhandled
[00:00:01.644,744] <dbg> main.status_cb: status 2 unhandled
[00:00:01.647,857] <dbg> main.status_cb: status 4 unhandled
[00:00:01.700,775] <dbg> main.status_cb: status 2 unhandled
[00:00:01.703,887] <dbg> main.status_cb: status 4 unhandled
[00:00:01.750,030] <dbg> main.status_cb: status 2 unhandled
[00:00:01.753,112] <err> usb_nrfx: USBD event allocation failed!
[00:00:01.753,143] <err> usb_nrfx: USBD event queue full!
[00:00:01.804,290] <dbg> main.status_cb: status 2 unhandled
[00:00:01.807,373] <err> usb_nrfx: USBD event allocation failed!
[00:00:01.807,403] <err> usb_nrfx: USBD event queue full!
[00:00:01.859,680] <dbg> main.status_cb: status 2 unhandled
[00:00:01.862,762] <err> usb_nrfx: USBD event allocation failed!
[00:00:01.862,792] <err> usb_nrfx: USBD event queue full!
[00:00:01.911,712] <dbg> main.status_cb: status 2 unhandled
[00:00:01.914,794] <err> usb_nrfx: USBD event allocation failed!
[00:00:01.914,825] <err> usb_nrfx: USBD event queue full!
[00:00:01.957,611] <dbg> main.status_cb: status 2 unhandled
[00:00:01.960,693] <err> usb_nrfx: USBD event allocation failed!
[00:00:01.960,723] <err> usb_nrfx: USBD event queue full!
[00:00:02.011,260] <dbg> main.status_cb: status 2 unhandled
[00:00:02.014,343] <err> usb_nrfx: USBD event allocation failed!
[00:00:02.014,373] <err> usb_nrfx: USBD event queue full!
[00:00:02.065,093] <dbg> main.status_cb: status 2 unhandled
[00:00:02.068,176] <err> usb_nrfx: USBD event allocation failed!
[00:00:02.068,206] <err> usb_nrfx: USBD event queue full!
[00:00:02.127,563] <dbg> main.status_cb: status 2 unhandled
[00:00:02.130,645] <err> usb_nrfx: USBD event allocation failed!
[00:00:02.130,645] <err> usb_nrfx: USBD event queue full!
[00:00:02.186,187] <dbg> main.status_cb: status 2 unhandled
[00:00:02.189,270] <err> usb_nrfx: USBD event allocation failed!
[00:00:02.189,300] <err> usb_nrfx: USBD event queue full!
[00:00:02.230,346] <dbg> main.status_cb: status 2 unhandled
[00:00:02.233,428] <err> usb_nrfx: USBD event allocation failed!
[00:00:02.233,459] <err> usb_nrfx: USBD event queue full!
[00:00:02.278,839] <dbg> main.status_cb: status 2 unhandled
[00:00:02.281,921] <err> usb_nrfx: USBD event allocation failed!
[00:00:02.281,951] <err> usb_nrfx: USBD event queue full!
[00:00:02.331,634] <dbg> main.status_cb: status 2 unhandled
[00:00:02.334,716] <err> usb_nrfx: USBD event allocation failed!

```

**To Reproduce**
Steps to reproduce the behavior:
1. west build -b reel_board
2. west flash

**Expected behavior**
The USB should be available even the USB cable is not connected at the beginning. 

** similiar issues**
The issue is very similar to the issue #11171, but still a bit different. It seems the USB stack becomes unrecoverable.

jfischer-phytec-iot, 2019-11-25:
It is the same hardware issue as #11171.

jli157, 2019-11-25:
I would argue this issue is different from the issue #11171 in that the USB stack here can't be recovered even though the USB cable is connected later, resulting in unavailability of USB functionality unless rebooting the system.  It is obvious that USBD event queue is full and seems the action of clearing event queue doesn't work well at that moment. 

@finikorg Do you think the issue reported here is the same as the one you has reported?

finikorg, 2019-11-26:
@jli157 In my case the USB was recoverable after I plug the cable, but it may still be the same issue.

jfischer-phytec-iot, 2019-11-27:
There is probably a issue with the event queue in the nRF driver.

emob-nordic, 2019-12-09:
I've checked it on my reel_board (v1507.0). I hope it does not differ that much.
I can see the issue however I don not think its related to event Q in the driver.
The case is that board is generating unnecessary events. Hardware issue??

It chokes because the event Q is overflowed and if that happen driver performs flush.
It seems that events are generated that fast, that its again and again overflowed making it impossible to allocate place for events that are necessary for enumeration.

For nrf52840dk it works without any problems that's why I think its board (hw) related.

jfischer-phytec-iot, 2019-12-09:
I have a fix for that.

jfischer-phytec-iot, 2019-12-09:
@jli157 Thanks for the report and sorry that I did not take it seriously right away :-)

PR #21244 (2019-12-11): drivers: usb_dc_nrfx: free memory block if event is intended to be dropped
Do not try to flush event queue in ISR.
Free memory block if event is intended to be dropped.
Resolves: #20953
commit b22035de (2019-12-11): drivers: usb_dc_nrfx: free memory if event is intended to be dropped

Free memory block if event is intended to be dropped.

Resolves: #20953
closed 2019-12-11 by carlescufi