zephyrproject-rtos/zephyr #20960: eth_mcux.c: Is there a reason for 2 memcpys and a shared buffer for ethernet Tx and Rx?

dferbas, 2019-11-25:
Platform Kinetis MK64, we use MK64FN1M0VLL12 package on our board.

The eth_mcux.c driver is using context->frame_buf to pass data from / to user buffer to / from driver buffers. For both directions, there are memcpys.
Moreover, the context->frame_buf is used both for Tx and Rx and thus protected with irq_lock, I think this reduces performance, which we observe.

In the eth_rx(), there is net_pkt_rx_alloc_with_buffer() for buffer allocation,
ENET_ReadFrame() is used to copy data from driver to context frame buffer (1st memcpy),
then net_pkt_write() copies these to packet buffer (2nd memcpy).

---
I prefer to pass pointer of allocated packet's data buffer directly to ENET_ReadFrame().
Something like passing pkt->cursor->buf->data and then calling net_pkt_skip(pkt, frame_length) to update packet cursor.

Similar with Tx - instead of net_pkt_read() only get packet data pointer as in above and pass it to ENET_SendFrame().

---
However this requires some addition to subsys/net/ip/net_pkt.c, benefit will be better throughput with higher performance, no need for extra buffer in ethernet context and no irq locking.
Ethernet is fast interface and I think the current code is killing all buffer descriptor ring capability, which is there since Motorola days to improve performance.

---
Maybe this can also help with the  #19325.

tbursztyka, 2019-11-25:
Look at line 94, it seems to be a known issue.
The hardware can do scatter-gather DMA but not the mcux hal, something like that?
Unless the driver is misusing the mcux hal. Which is likely to be the issue is seems.
I don't think you need any net_pkt api changes for this.

dferbas, 2019-11-25:
I really apologize, I am new to Zephyr and I did not check buffer declaration, only comments in code.

I am in a position whether to move to Zephyr or not. I think Zephyr has future, but now we are unable to fulfill reliable data delivery, socket options for outgoing buffer to collect line data are missing, there are unexpected ARP packets being sent.

In the past, we spent some time with Coldfire ethernet peripheral and tuned its driver - https://github.com/dferbas/NutOs/blob/master/nut/arch/m68k/coldfire/dev/common/mcf5_fec.c. After experimenting with direct sending of buffer fragments from application (Nut/OS have separate buffers for app, transport,network and datalink layers), we stayed with non zero copy way, because of reliability with DMA transfers from external RAM.

I think there can be a temporary workaround to speed it up.
1) MK64 has 256 kB RAM, so there can be 2 separate buffers and no need for irq_lock
2) Lines 602-603: status = ENET_ReadFrame(ENET, &context->enet_handle,
				pkt->cursor->buf.data, frame_length);
Line 611: if (net_pkt_write(pkt, pkt->cursor->buf.data, frame_length))
In this case logic in net_pkt_write will check same pointers and will not actually do copy.
Or because I am not familiar with net_pkt APIs pkt->cursor->buf.len = frame_length.

tbursztyka, 2019-11-26:
> status = ENET_ReadFrame(ENET, &context->enet_handle, pkt->cursor->buf.data, frame_length);

You cannot do that, buf.data might be of any size (it's a Kconfig based configurable option). So if frame_length is bigger, it will overflow.

I don't know if you can call ENET_ReadFrame() as many times as possible to get the whole frame, but if so, you'll have to loop over it like that perhaps:  

```
while (frame_length) {
length = net_buf_simple_tailroom(pkt->cursor->buf);
if (frame_length < length) {
length = frame_length;
}

status = ENET_ReadFrame(ENET, &context->enet_handle, pkt->cursor->pos, length);
... manage status code ...

frame_length -= length;
net_pkt_skip(pkt, length);
}
```
So you could remove the lines 611 also.
And no need of any local buffer. At least for receiving.

Something like that. 

But yes, zephyr is young, and we do not handle DMA properly yet where relevant (and transparently as well)

tbursztyka, 2019-11-26:
> socket options for outgoing buffer to collect line data are missing, there are unexpected ARP packets being sent.

Please create issues for both, that would be of great help to improve the stack.

Oanerer, 2019-11-26:
> 
>there are unexpected ARP packets being sent.

I think I may have seen a similar issue! Below a repost of my message on Slack in #networking:

> Yesterday I noticed that a Zephyr node with Ethernet in our busy office network seems to falsely mimick ARP probes that are sent from another (Windows 7) PC when
> 
> `CONFIG_NET_ARP_TABLE_SIZE`
> 
> is set to it's default value of 2. This causes the Windows 7 PC to repeatedly try to join the network using a new DHCP address, resending the ARP probe (which is mimicked again and again) and ultimately giving up joining the network with event ID 4199.
> 
> Wireshark log:
> [Zephyr_ARP_probe_mimicking.pcapng.gz](https://github.com/zephyrproject-rtos/zephyr/files/3891544/Zephyr_ARP_probe_mimicking.pcapng.gz)
> 
> In the log the Zephyr node is MAC `InterayS_01:b9:74` at IP 192.168.9.196 (acquired through DHCP), the Modbus TCP master it's communicating with is MAC `Giga-Byt_5f:24:d9` at IP 192.168.9.188. The Windows 7 (W7) PC trying to join the network is MAC `HewlettP_b8:39:20`.
> 
> As you can see in the log, the W7 PC is offered IP 192.168.9.158 (Packet 7). The W7 PC then decides to ARP probe for this address to make sure no duplicate IP conflicts will occur (packet 17 and 25). At packet 31 the Zephyr node decides to send an ARP request for IP address 192.168.9.158, which is told to be sent to 192.168.9.196. This seems valid. However, what happens next does not.
> 
> At packets 32 and 33 the Zephyr node mimicks the ARP probes for address 192.168.9.158 since the reply address is set to 0.0.0.0, just as the W7 PC, for no good reason as far as I'm aware. These packets are also noticed by the W7 PC, which promply sends a DHCP decline (packet 34).
> 
> Obviously the ARP table size is set very small and increasing it seems to help, but I'm interested whether this is expected behaviour?

jukkar, 2019-11-26:
> > there are unexpected ARP packets being sent.
> 
> I think I may have seen a similar issue! Below a repost of my message on Slack in #networking:
> 

@Oanerer could you create a separate issue for this, also enable ARP debug logging and attach it and pcap file of this issue.

dferbas, 2019-11-26:
, @tbursztyka Thank you for advice, I will try to follow it. I did not realize packet can be longer then max configured size. There is an assertion in the fsl_enet driver to have BD_num * BD_size > MaxFrameLen.

As to your comment: The enet_fsl loops over received buffers to collect a packet. If next buffer does not fit, driver returns with error. Rest of packet is not read from buffer descriptors and this case is not handled by Zephyr. I think enet_fsl should be called again with buffer address set to NULL to dismiss rest of a big packet.

@tbursztyka, @jukkar: I created  #21016 for ARP packets and other unexpected behavior - there are also unexpected ping packets.

Oanerer, 2019-11-27:
> 
> 
> > > there are unexpected ARP packets being sent.
> > 
> > 
> > I think I may have seen a similar issue! Below a repost of my message on Slack in #networking:
> 
> @Oanerer could you create a separate issue for this, also enable ARP debug logging and attach it and pcap file of this issue.

Logs are attached to the issue dferbas created:  #21016

decsny, 2024-01-15:
I have a branch downstream where I am working on making the enet driver zero copy. But it is not my top priority.

decsny, 2025-01-27:
this is already a known issue, closing this issue as duplicate to more general issue in #67930, will link to this discussion from that one

closed 2025-01-27 by decsny