zephyrproject-rtos/zephyr #20974: file resources exceeded with sanitycheck

pabigot, 2019-11-25:
When running sanitycheck with device testing (8 devices connected) on `tests/kernel` I've gotten this diagnostic repeating in bursts:

```
Serial device err: [Errno 24] could not open port /dev/ttyACM1: [Errno 24] Too many open files: '/dev/ttyACM1'
```

Re-invoking with `ulimit -n 4096` (to increase the number of open files from 1024 to 4096) I instead get this diagnostic:
```
Exception in thread Thread-531:
Traceback (most recent call last):
  File "/usr/lib/python3.7/threading.py", line 917, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.7/threading.py", line 865, in run
    self._target(*self._args, **self._kwargs)
  File "/scratch/zephyr/zp1/zephyr/scripts/sanitycheck", line 649, in monitor_serial
    readable, _, _ = select.select(readlist, [], [], self.timeout)
ValueError: filedescriptor out of range in select()
```

The error becomes visible near the end of the run, where the test failures clump up.  This suggests that the problem is with not releasing a file descriptor associated with test serial output when the test has completed.

It may be necessary to explicitly close some descriptors rather than wait for them to be garbage collected.

nashif, 2019-12-18:
Cant reproduce it still..

pabigot, 2019-12-18:
@nashif How many devices have you had simultaneously enabled in the hardware file?

nashif, 2020-01-02:
> @nashif How many devices have you had simultaneously enabled in the hardware file?

I am testing with different numbers of devices ranging from 1 to 12. I think to hit this there need to be some failures and timeouts, not just passing tests. Looking into it now.

PR #21684 (2020-01-06): Sanitycheck fixes, enhancements
- close pipe file handles
- More debug messages
- add Digilent to device discovery

Fixes #20974
Fixes #21637
commit bdffb889 (2020-01-06): sanitycheck: close pipe file handles

Make sure we close the pipe file handles after we are done, otherwise we
will end up with too many open file descriptors and crash...

Fixes #20974
Fixes #21637
closed 2020-01-06 by nashif