zephyrproject-rtos/zephyr #20980: ESP32 flash error

rolandma, 2019-11-25:
After successfully build a project for ESP32, when running for west flash, there's an error:
west flash -d build/ --skip-rebuild --esp-device /dev/ttyS0
-- west flash: using runner esp32
-- runners.esp32: Converting ELF to BIN
esptool.py v2.9-dev

A fatal error occurred: Invalid segment count 17 (max 16). Usually this indicates a linker script problem.
ERROR: command exited with status 2: /usr/bin/python3 /esp-idf/components/esptool_py/esptool/esptool.py --chip esp32 elf2image /zephyr/samples/build/zephyr/zephyr.elf

rolandma, 2019-11-25:
Okay, by changing the esptool.py segments >17 will fix the flashing error, but seems to me the code didn't really run.
    def verify(self):
        if len(self.segments) > 17:
            raise FatalError('Invalid segment count %d (max 16). Usually this indicates a linker script problem.' % len(self.segments))

rolandma, 2019-12-02:
I would like to re-open this issue, not because I can't flash it onto the ESP32-wroom-32D dev kit, but because even if I change the esptool.py script to flash it, no sample project will run other than "hello-world". which is not very helpful. I compiled several samples, and all compiled successfully, but none will run on the target board. I tried at least these 4 samples, and managed to compile all of them, but when using 'west flash -d build/ --skip-rebuild --esp-device /dev/ttyS0', the 'hello-world' sample can flash and start the zephyr os, but all others won't flash.
1. samples/net/vlan
2. sample/net/mqtt_publisher
3. sample/net/lwm2m_client
4. sample/net/wifi

Skallwar, 2020-07-22:
This is due to the fact that esptool create an app image from the ELF sections. And there is more than 16 of them. This might be possible to fix this in the ESP32 linker script by merging sections

dcpleung, 2020-07-22:
Network subsystem and kernel objects utilize linker sections extensively to group data together. The definitions for those sections are common to all arch/SoCs/boards. We could re-implement them in ESP32's linker script by grouping them into fewer sections, but this will increase maintenance as changes to common scripts will need to be re-implement in ESP32's.

PR #27070 (2020-10-13): esp32: workaround esptool linker sections limit
When converting ELF to a flashable image, the esptool can only
process up to 16 ELF sections. Zephyr has been happily grouping
similar objects into linker sections which can easily go over
esptool's limit. This patch consolidates the kernel, net, log,
and shell objects into their own sections.

Also remove the app_noinit section as no one is emitting
anything into that section, and it's not being used by other
arch/SoC/boards in Zephyr.

Fixes #20980
commit 41967f99 (2020-10-13): esp32: workaround esptool linker sections limit

When converting ELF to a flashable image, the esptool can only
process up to 16 ELF sections. Zephyr has been happily grouping
similar objects into linker sections which can easily go over
esptool's limit. This patch consolidates the kernel, net, log,
and shell objects into their own sections.

Also remove the app_noinit section as no one is emitting
anything into that section, and it's not being used by other
arch/SoC/boards in Zephyr.

Fixes #20980
closed 2020-10-05 by github-actions[bot]