zephyrproject-rtos/zephyr #20981: mempool: MPU fault

paulgavrikov, 2019-11-25:
**Describe the bug**
There seems to be a bug in mempools bfree_recombine. Commit 1b193e9ece638c0c6f144ce2bee96f1ff268a160 changed `partner_alloc_bits(p, level, bn) != 0xf` to `partner_alloc_bits(p, level, bn) != 0` which results in the following MPU faults in our application:

**To Reproduce**
Unfortunately I cannot pinpoint the error source in our app to provide a minimal sample.

**Expected behavior**
No MPU fault.

**Impact**
showstopper

**Screenshots or console output**

```
E: ***** MPU FAULT *****
E:   Data Access Violation
E:   MMFAR Address: 0x0
E: r0/a1:  0x2000160c  r1/a2:  0x00000200  r2/a3:  0x00000000
E: r3/a4:  0x00000000 r12/ip:  0x00000000 r14/lr:  0x00004ec1
E:  xpsr:  0x01000000
E: Faulting instruction address (r15/pc): 0x0000e6f4
E: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
E: Current thread: 0x20000f38 (gMasterTxThreadId)
E: Halting system
```

**Environment (please complete the following information):**
 - OS: Kubuntu 18.04
 - Toolchain: Zephyr SDK 0.10.13
 - Commit SHA 57c635587f718078705be763f1aa7e9a1ed328e7

**Additional context**

carlescufi, 2019-11-25:
@paulgavrikov please add as much info as you can about how you are defining and using the mempool.

stephanosio, 2019-11-25:
If the mempool struct is, for whatever reason, not located in bss, the following change will likely cause a failure: https://github.com/zephyrproject-rtos/zephyr/commit/1b193e9ece638c0c6f144ce2bee96f1ff268a160#diff-e1987e8f6f4f53b4771db2f818b192f4L97.

Try adding `clear_alloc_bit(p, 0, i);` there and see if doing so resolves the issue; if it does, your mempool struct probably is not located in bss.

andyross, 2019-11-25:
If that's true, then we have a bug somewhere underneath sys_mem_pool_init().  Obviously it shouldn't be leaving the heap in an inconsistent state.

npitre, 2019-11-25:
On Mon, 25 Nov 2019, Stephanos Ioannidis wrote:

> Try adding `clear_alloc_bit(p, 0, i);` there and see if doing so resolves the issue; if it does, your mempool struct probably is not located in bss.

That's not sufficient for that conclusion.  Adding `clear_alloc_bit(p, 
0, i)` is no longer an equivalent replacement of the former 
`set_free_bit(p, 0, i)` here.

Instead, the following should be attempted:

```
diff --git a/lib/os/mempool.c b/lib/os/mempool.c
index efca11a203..b8f1b697f0 100644
--- a/lib/os/mempool.c
+++ b/lib/os/mempool.c
@@ -85,6 +85,7 @@ void z_sys_mem_pool_base_init(struct sys_mem_pool_base *p)
 
 	for (i = 0; i < p->n_levels; i++) {
 		int nblocks = buflen / sz;
+		int j;
 
 		sys_dlist_init(&p->levels[i].free_list);
 
@@ -95,6 +96,10 @@ void z_sys_mem_pool_base_init(struct sys_mem_pool_base *p)
 			bits += (nblocks + 31)/32;
 		}
 
+		for (j = 0; j < nblocks; j++) {
+			clear_alloc_bit(p, i, j);
+		}
+
 		sz = WB_DN(sz / 4);
 	}
 
```

paulgavrikov, 2019-11-26:
@npitre Applying the diff does not fix the issue.
@stephanosio I did a git bisect and it is definitely that commit and particularly the `partner_alloc_bits(p, level, bn) != 0` check that causes the issue.
@carlescufi We are not defining our own mempools but we are using the heap through k_malloc and define threads, FIFOs.

npitre, 2019-11-26:
On Tue, 26 Nov 2019, Paul Gavrikov wrote:

> @npitre Applying the diff does not fix the issue.

OK.

> @stephanosio I did a git bisect and it is definitely that commit and particularly the `partner_alloc_bits(p, level, bn) != 0` check that causes the issue.

Could you elaborate on that `partner_alloc_bits()` issue?

Where exactly is the code faulting? If you could convert the fault 
address into a code line number, or better yet a gdb backtrace, that 
would be very helpful.

paulgavrikov, 2019-11-26:
@npitre Don't have access to gdb now but here is the code line: https://github.com/zephyrproject-rtos/zephyr/blob/master/include/sys/dlist.h#L497

I haven't bothered to understand mempools `bfree_recombine()` but there is an if clause there that was changed with above mentioned commit (https://github.com/zephyrproject-rtos/zephyr/blob/master/lib/os/mempool.c#L17)4. It used to check `!=0xf` but was changed to `!=0`. All I can say is that if I change the check back to `!=0xf` the issue disappears. So I am wondering if checking `!=0` is the correct way.

npitre, 2019-11-26:
> @npitre Don't have access to gdb now but here is the code line: https://github.com/zephyrproject-rtos/zephyr/blob/master/include/sys/dlist.h#L497

My guess would be corruption of the free list caused by some buffer 
overflow somewhere.

> I haven't bothered to understand mempools `bfree_recombine()` but 
> there is an if clause there that was changed with above mentioned 
> commit 

I do understand that commit and code pretty well.

> It used to check `!=0xf` but was changed to `!=0`. All I can say is 
> that if I change the check back to `!=0xf` the issue disappears. So I 
> am wondering if checking `!=0` is the correct way.

Yes it is. The commit in question simply flipped 0-bits to 1 and 1-bits 
to 0, essentially inverting those bit values and keeping the code logic 
unchanged. If the actual bit value matters in your case then this might 
indicate some corruption of the mempool bitmap by something exterior to 
it.

By switching the condition back to `!=0xf` you're essentially preventing 
recombination of small free adjacent blocks into a larger one.

Let me suggest another patch to help find possible overflows:

```
diff --git a/lib/os/mempool.c b/lib/os/mempool.c
index efca11a203..fdfbf389be 100644
--- a/lib/os/mempool.c
+++ b/lib/os/mempool.c
@@ -229,6 +229,8 @@ int z_sys_mem_pool_block_alloc(struct sys_mem_pool_base *p, size_t size,
 	void *data = NULL;
 	size_t lsizes[LVL_ARRAY_SZ(p->n_levels)];
 
+	size += 2 * sizeof(uintptr_t);
+
 	/* Walk down through levels, finding the one from which we
 	 * want to allocate and the smallest one with a free entry
 	 * from which we can split an allocation if needed.  Along the
@@ -290,6 +292,15 @@ int z_sys_mem_pool_block_alloc(struct sys_mem_pool_base *p, size_t size,
 		return -ENOMEM;
 	}
 
+	{
+		uintptr_t *sentinel = data;
+		int end = (lsizes[alloc_l] / sizeof(uintptr_t)) - 1;
+		sentinel[0] = 0xdeadbeef;
+		sentinel[end] = 0xbadbea7;
+		data = sentinel + 1;
+		*data_p = data;
+	}
+
 	*level_p = alloc_l;
 	*block_p = block_num(p, data, lsizes[alloc_l]);
 
@@ -313,6 +324,16 @@ void z_sys_mem_pool_block_free(struct sys_mem_pool_base *p, u32_t level,
 		lsizes[i] = WB_DN(lsizes[i-1] / 4);
 	}
 
+	{
+		uintptr_t *sentinel = block_ptr(p, lsizes[level], block);
+		int end = (lsizes[level] / sizeof(uintptr_t)) - 1;
+		if (sentinel[0] != 0xdeadbeef || sentinel[end] != 0xbadbea7) {
+			__ASSERT(false, "memory corruption");
+			printk("memory corruption\n");
+			k_panic();
+		}
+	}
+
 	block_free(p, level, lsizes, block);
 }
 
```

paulgavrikov, 2019-11-28:
@npitre I applied the diff and do **not** run into your assert.

npitre, 2019-11-28:
On Thu, 28 Nov 2019, Paul Gavrikov wrote:

> @npitre I applied the diff and do **not** run into your assert.

OK. Then here's another one for you to test:

```
diff --git a/include/sys/mempool_base.h b/include/sys/mempool_base.h
index 714e6cbd51..dd4d749780 100644
--- a/include/sys/mempool_base.h
+++ b/include/sys/mempool_base.h
@@ -20,6 +20,7 @@ struct sys_mem_pool_lvl {
 		u32_t *bits_p;
 		u32_t bits[sizeof(u32_t *)/4];
 	};
+	uintptr_t sentinel;
 	sys_dlist_t free_list;
 };
 
@@ -34,9 +35,10 @@ struct sys_mem_pool_base {
 	s8_t max_inline_level;
 	struct sys_mem_pool_lvl *levels;
 	u8_t flags;
+	u32_t bmap_csum;
 };
 
-#define _MPOOL_MINBLK sizeof(sys_dnode_t)
+#define _MPOOL_MINBLK (sizeof(uintptr_t) + sizeof(sys_dnode_t))
 
 #define Z_MPOOL_HAVE_LVL(maxsz, minsz, l) \
 	(((maxsz) >> (2*(l))) >= MAX((minsz), _MPOOL_MINBLK) ? 1 : 0)
diff --git a/lib/os/mempool.c b/lib/os/mempool.c
index efca11a203..60470a3642 100644
--- a/lib/os/mempool.c
+++ b/lib/os/mempool.c
@@ -75,6 +75,61 @@ static int partner_alloc_bits(struct sys_mem_pool_base *p, int level, int bn)
 	return (*word >> (4*(bit / 4))) & 0xf;
 }
 
+static void *set_sentinel(void *ptr)
+{
+	uintptr_t *sentinel = ptr;
+	*sentinel = (uintptr_t)sentinel ^ 0xDEADBEA7;
+	return sentinel + 1;
+}
+
+static void *check_sentinel(void *ptr)
+{
+	uintptr_t *sentinel = ptr;
+	if (*sentinel != ((uintptr_t)sentinel ^ 0xDEADBEA7)) {
+		__ASSERT(false, "memory corruption");
+		printk("memory corruption\n");
+		k_panic();
+	}
+	return sentinel + 1;
+}
+
+static void *predec_check_sentinel(void *ptr)
+{
+	uintptr_t *sentinel = ptr;
+	check_sentinel(--sentinel);
+	return sentinel;
+}
+
+#include <sys/crc.h>
+
+static uint32_t calc_bmap_csum(struct sys_mem_pool_base *p)
+{
+	u32_t csum = 0;
+	int l;
+
+	for (l = 0; l < p->n_levels; l++) {
+		int nblocks = p->n_max * (1 << (l * 2));
+		u32_t *bits;
+		get_bit_ptr(p, l, 0, &bits);
+		csum = crc32_ieee_update(csum, (u8_t *)bits, 4 * ((nblocks + 31)/32));
+	}
+	return csum;
+}
+
+static void set_bmap_csum(struct sys_mem_pool_base *p)
+{
+	p->bmap_csum = calc_bmap_csum(p);
+}
+
+static void check_bmap_csum(struct sys_mem_pool_base *p)
+{
+	if (p->bmap_csum != calc_bmap_csum(p)) {
+		__ASSERT(false, "memory corruption");
+		printk("memory corruption\n");
+		k_panic();
+	}
+}
+
 void z_sys_mem_pool_base_init(struct sys_mem_pool_base *p)
 {
 	int i;
@@ -86,6 +141,7 @@ void z_sys_mem_pool_base_init(struct sys_mem_pool_base *p)
 	for (i = 0; i < p->n_levels; i++) {
 		int nblocks = buflen / sz;
 
+		set_sentinel(&p->levels[i].sentinel);
 		sys_dlist_init(&p->levels[i].free_list);
 
 		if (nblocks <= sizeof(p->levels[i].bits)*8) {
@@ -97,11 +153,12 @@ void z_sys_mem_pool_base_init(struct sys_mem_pool_base *p)
 
 		sz = WB_DN(sz / 4);
 	}
+	set_bmap_csum(p); 
 
 	for (i = 0; i < p->n_max; i++) {
 		void *block = block_ptr(p, p->max_sz, i);
 
-		sys_dlist_append(&p->levels[0].free_list, block);
+		sys_dlist_append(&p->levels[0].free_list, set_sentinel(block));
 	}
 }
 
@@ -143,9 +200,13 @@ static void *block_alloc(struct sys_mem_pool_base *p, int l, size_t lsz)
 {
 	sys_dnode_t *block;
 
+	check_bmap_csum(p);
+	check_sentinel(&p->levels[l].sentinel);
 	block = sys_dlist_get(&p->levels[l].free_list);
 	if (block != NULL) {
+		block = predec_check_sentinel(block);
 		set_alloc_bit(p, l, block_num(p, block, lsz));
+		set_bmap_csum(p);
 	}
 	return block;
 }
@@ -158,13 +219,17 @@ static unsigned int bfree_recombine(struct sys_mem_pool_base *p, int level,
 		int i, lsz = lsizes[level];
 		void *block = block_ptr(p, lsz, bn);
 
+		check_bmap_csum(p);
+		check_sentinel(&p->levels[level].sentinel);
+
 		/* Detect common double-free occurrences */
 		__ASSERT(alloc_bit_is_set(p, level, bn),
 			 "mempool double-free detected at %p", block);
 
 		/* Put it back */
 		clear_alloc_bit(p, level, bn);
-		sys_dlist_append(&p->levels[level].free_list, block);
+		set_bmap_csum(p);
+		sys_dlist_append(&p->levels[level].free_list, set_sentinel(block));
 
 		/* Relax the lock (might result in it being taken, which is OK!) */
 		pool_irq_unlock(p, key);
@@ -178,7 +243,7 @@ static unsigned int bfree_recombine(struct sys_mem_pool_base *p, int level,
 		for (i = 0; i < 4; i++) {
 			int b = (bn & ~3) + i;
 
-			sys_dlist_remove(block_ptr(p, lsz, b));
+			sys_dlist_remove(check_sentinel(block_ptr(p, lsz, b)));
 		}
 
 		/* Free the larger block */
@@ -208,14 +273,17 @@ static void *block_break(struct sys_mem_pool_base *p, void *block, int l,
 {
 	int i, bn;
 
+	check_bmap_csum(p);
 	bn = block_num(p, block, lsizes[l]);
 	set_alloc_bit(p, l + 1, 4*bn);
+	set_bmap_csum(p);
 
 	for (i = 1; i < 4; i++) {
 		int lsz = lsizes[l + 1];
 		void *block2 = (lsz * i) + (char *)block;
 
-		sys_dlist_append(&p->levels[l + 1].free_list, block2);
+		check_sentinel(&p->levels[l + 1].sentinel);
+		sys_dlist_append(&p->levels[l + 1].free_list, set_sentinel(block2));
 	}
 
 	return block;
```

paulgavrikov, 2019-11-29:
@npitre I run into the issue in sys_dlist_remove earlier now but do not see any additional logs.

```
[13:12:20:858] E: ***** MPU FAULT *****␍␊
[13:12:20:858] E:   Data Access Violation␍␊
[13:12:20:858] E:   MMFAR Address: 0x0␍␊
[13:12:20:867] E: r0/a1:  0x2000082c  r1/a2:  0x00000000  r2/a3:  0x00000000␍␊
[13:12:20:867] E: r3/a4:  0xfeadb68f r12/ip:  0xee636261 r14/lr:  0x0000328f␍␊
[13:12:20:875] E:  xpsr:  0x81000000␍␊
[13:12:20:875] E: Faulting instruction address (r15/pc): 0x0000329a␍␊
[13:12:20:885] E: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0␍␊
[13:12:20:885] E: Current thread: 0x20000224 (gMasterTxThreadId)␍␊
[13:12:20:897] E: Halting system␍␊
```

npitre, 2019-11-29:
On Fri, 29 Nov 2019, Paul Gavrikov wrote:

> @npitre I run into the issue in sys_dlist_remove earlier now but do not see any additional logs.

Do you mean that none of the checks from my latest patch triggered?

If so I'm afraid I'd have to be able to reproduce it locally to help you further.

Is it reproducible in a deterministic manner?

How long before the crash occurs?

Please apply the following patch and send me the output log leading to 
the crash. Maybe that will tell us something.

```
diff --git a/lib/os/mempool.c b/lib/os/mempool.c
index efca11a203..660458a013 100644
--- a/lib/os/mempool.c
+++ b/lib/os/mempool.c
@@ -293,6 +293,9 @@ int z_sys_mem_pool_block_alloc(struct sys_mem_pool_base *p, size_t size,
 	*level_p = alloc_l;
 	*block_p = block_num(p, data, lsizes[alloc_l]);
 
+	printk("mempool_alloc: buf=%p max_sz=%zd n_max=%d n_levels=%d lvl=%d blk=%d\n",
+	       p->buf, p->max_sz, p->n_max, p->n_levels, *level_p, *block_p);
+
 	return 0;
 }
 
@@ -302,6 +305,9 @@ void z_sys_mem_pool_block_free(struct sys_mem_pool_base *p, u32_t level,
 	size_t lsizes[LVL_ARRAY_SZ(p->n_levels)];
 	int i;
 
+	printk("mempool_free:  buf=%p max_sz=%zd n_max=%d n_levels=%d lvl=%d blk=%d\n",
+	       p->buf, p->max_sz, p->n_max, p->n_levels, level, block);
+
 	/* As in z_sys_mem_pool_block_alloc(), we build a table of level sizes
 	 * to avoid having to store it in precious RAM bytes.
 	 * Overhead here is somewhat higher because block_free()
```

paulgavrikov, 2019-12-02:
@npitre 

Yes, none of the checks triggered. I tested the patches separately. Should I try all together?
It happens deterministically. Some more background: basically our app is a ZTests over a custom stack we wrote on top of SPI. There's quite a lot of code and I do not have clearance to share it unfortunately. The issue happens rather quickly as you can see in the log:

```
[14:35:45:918] mempool_free:  buf=0x20000628 max_sz=32768 n_max=1 n_levels=5 lvl=4 blk=0␍␊
[14:35:45:918] mempool_alloc: buf=0x20000628 max_sz=32768 n_max=1 n_levels=5 lvl=4 blk=1␍␊
[14:35:45:918] mempool_free:  buf=0x20000628 max_sz=32768 n_max=1 n_levels=5 lvl=4 blk=1␍␊
[14:35:45:932] E: ***** MPU FAULT *****␍␊
[14:35:45:932] E:   Data Access Violation␍␊
[14:35:45:932] E:   MMFAR Address: 0x0␍␊
[14:35:45:932] E: r0/a1:  0x00000002  r1/a2:  0x200006a8  r2/a3:  0x20000628␍␊
[14:35:45:939] E: r3/a4:  0x00000080 r12/ip:  0x00000000 r14/lr:  0x00000000␍␊
[14:35:45:952] E:  xpsr:  0x81000000␍␊
[14:35:45:952] E: Faulting instruction address (r15/pc): 0x000030b2␍␊
[14:35:45:952] E: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0␍␊
[14:35:45:964] E: Current thread: 0x20000224 (gMasterTxThreadId)␍␊
[14:35:45:964] E: Halting system␍␊
```

paulgavrikov, 2019-12-02:
Actually, after applying the latest patch I get the MPU fault in the very first test while it used to be a later on before. Also, the log is not deterministic:

```
[15:06:07:201] mempool_alloc: buf=0x20000628 max_sz=32768 n_max=1 n_levels=5 lvl=4 blk=0␍␊
[15:06:07:211] mempool_free:  buf=0x20000628 max_sz=32768 n_max=1 n_levels=5 lvl=4 blk=0␍␊
[15:06:07:218] mempool_alloc: buf=0x20000628 max_sz=32768 n_max=1 n_levels=5 lvl=4 blk=64␍␊
[15:06:07:225] mempool_free:  buf=0x20000628 max_sz=32768 n_max=1 n_levels=5 lvl=4 blk=64␍␊
[15:06:07:234] mempool_alloc: buf=0x20000628 max_sz=32768 n_max=1 n_levels=5 lvl=4 blk=0␍␊
[15:06:07:242] mempool_free:  buf=0x20000628 max_sz=32768 n_max=1 n_levels=5 lvl=4 blk=0␍␊
```

Another run:
```
[15:09:40:235] mempool_alloc: buf=0x20000628 max_sz=32768 n_max=1 n_levels=5 lvl=4 blk=0␍␊
[15:09:40:252] payloadSize=200␍␊
[15:09:40:252] packetsToSend=5000␍␊
[15:09:40:252] mempool_free:  buf=0x20000628 max_sz=32768 n_max=1 n_levels=5 lvl=4 blk=0␍␊
[15:09:40:252] mempool_alloc: buf=0x20000628 max_sz=32768 n_max=1 n_levels=5 lvl=4 blk=1␍␊
[15:09:40:264] mempool_free:  buf=0x20000628 max_sz=32768 n_max=1 n_levels=5 lvl=4 blk=1␍␊
```

npitre, 2019-12-02:
Quoting your log:

```
[15:06:07:201] mempool_alloc: buf=0x20000628 max_sz=32768 n_max=1 n_levels=5 lvl=4 blk=0
[15:06:07:211] mempool_free:  buf=0x20000628 max_sz=32768 n_max=1 n_levels=5 lvl=4 blk=0
[15:06:07:218] mempool_alloc: buf=0x20000628 max_sz=32768 n_max=1 n_levels=5 lvl=4 blk=64
[15:06:07:225] mempool_free:  buf=0x20000628 max_sz=32768 n_max=1 n_levels=5 lvl=4 blk=64
```

and:

```
[15:09:40:235] mempool_alloc: buf=0x20000628 max_sz=32768 n_max=1 n_levels=5 lvl=4 blk=0
[15:09:40:252] mempool_free:  buf=0x20000628 max_sz=32768 n_max=1 n_levels=5 lvl=4 blk=0
[15:09:40:252] mempool_alloc: buf=0x20000628 max_sz=32768 n_max=1 n_levels=5 lvl=4 blk=1
[15:09:40:264] mempool_free:  buf=0x20000628 max_sz=32768 n_max=1 n_levels=5 lvl=4 blk=1
```

Something is wrong there. Normally, the block being freed should be 
available and reallocated right away by the following alloc call. Here, 
in one case, block 0 is allocated and freed, then block 64 is allocated. 
In the other case, block 0 is allocated and freed, then block 1 is 
allocated. In both cases, block 0 should have been allocated again, not 
some other random block.

I tried to reproduce your allocation pattern and got the following log 
instead:

```
mempool_alloc: buf=0x00113120 max_sz=32768 n_max=1 n_levels=5 lvl=4 blk=0
mempool_free:  buf=0x00113120 max_sz=32768 n_max=1 n_levels=5 lvl=4 blk=0
mempool_alloc: buf=0x00113120 max_sz=32768 n_max=1 n_levels=5 lvl=4 blk=0
mempool_free:  buf=0x00113120 max_sz=32768 n_max=1 n_levels=5 lvl=4 blk=0
mempool_alloc: buf=0x00113120 max_sz=32768 n_max=1 n_levels=5 lvl=4 blk=0
mempool_free:  buf=0x00113120 max_sz=32768 n_max=1 n_levels=5 lvl=4 blk=0
```

You'll have to figure out why your allocation pattern is going erratic. 
Without reproducing it myself I'm afraid that my ability to help you is 
rather limited at this point.

paulgavrikov, 2020-01-15:
No longer happening with r. 2.1

closed 2020-01-15 by paulgavrikov

MrPointer, 2020-01-15:
@paulgavrikov What timing.
In the last few days, we've been encountering **very** similar issues, with the fault address pointing to the same line in the _dlist_ module.
Thing is - We're working with v2.1 as well :worried: 

At first, we did suffer from some buffer overflows as suggested in one of the comments here, but we've solved all of them by now, and the issue persists.
Unfortunately, it's non-deterministic at all, as far as happening only after 30 minutes of normal usage, whereas sometimes after merely 1-2 minutes.
I don't have anything else to share currently, especially not code since we're developing offline.

I'm afraid the issue should be reopened.