zephyrproject-rtos/zephyr #20991: test_timer_duration_period fails with stm32 lptimer

FRASTM, 2019-11-26:
When the LPTIMER is enabled, the kernel test timer/timer_api fails in the 
"test_timer_duration_period: tdata.expire_cnt == 4 is false"

**To Reproduce**
Steps to reproduce the behavior:

1. Enable the LPTIMER in the proj.conf :
CONFIG_STM32_LPTIM_TIMER=y
CONFIG_STM32_LPTIM_CLOCK_LSE=y
2. mkdir build; cd build
3. cmake -DBOARD=board\nucleo_l4r5zi
4. make
5. See error
"Assertion failed at ZEPHYR_BASE/samples/subsys/power/stm32_pm/timer_api/src/main.c:140: test_timer_duration_period: tdata.expire_cnt == 4 is false "

FRASTM, 2020-02-11:
Due to the LPTIM clock source = LSE, the minimal tick period is set to 1000 per sec.
This makes one tick duration = about 33 lptim counting units

Note that the max possible timeout remains 2000ms , reached  when  the LPTIM counter = 0xFFFF

FRASTM, 2020-04-06:
Applying the PR

carlescufi, 2020-05-21:
@erwango It doesn't seem this actually has a PR does it?

erwango, 2020-05-22:
@carlescufi #20992 should be the one closing this.

closed 2020-05-29 by carlescufi