zephyrproject-rtos/zephyr #20993: Spin lock validation may fail

nordic-krch, 2019-11-26:
**Describe the bug**
When using spin lock in a following code:

```
struct k_spinlock l;
k_spinlock_key_t k;

k = k_spin_lock(&l);
```
I'm hitting assert in `z_spin_lock_valid`.
https://github.com/zephyrproject-rtos/zephyr/blob/21e0d4875045a5f26812f34eb9c230398914fd23/include/spinlock.h#L79

`k_spin_lock` is validating lock structure which might not be initialize and validation includes checking one of the fields of uninitialized structure. I'm not sure then if `struct k_spinlock` should be somehow initialized before being used? If so, i don't see any init function. What about single core case?

andyross, 2019-11-26:
Spinlocks are valid when zero-initialized.  The normal use case is to put it in bss, but if you want to use a local you can do "`struct k_spinlock l = {};`".  There probably should be a K_SPINLOCK_INITIALIZER macro defined.

nordic-krch, 2019-11-28:
seems like spin lock is in general missing any documentation. no doxygen for  `k_spin_lock` and `k_spin_unlock`: https://github.com/zephyrproject-rtos/zephyr/blob/21e0d4875045a5f26812f34eb9c230398914fd23/include/spinlock.h#L67

nashif, 2019-12-17:
@andyross can you please add missing documentation?

closed 2020-05-08 by carlescufi

thomasthorsen, 2023-09-14:
I don't think the documentation added in #25092 actually addresses the main issue here:
"I'm not sure then if struct k_spinlock should be somehow initialized before being used?"

There is no init API and there is no part of the added documentation that states the memory must be initialised in any way (as far as I can tell). I encountered this in a spin lock that was allocated by k_malloc, which had actually worked for some time in our system, and then when some code was moved around, it started failing oddly with this, because the pre-existing memory contents happened to be different for the new location. In the old location it just happened to be zero, and nothing bad happened.

PR #25092 (2020-05-08): include: Add documentation for spinlocks
The kernel spinlock API didn't have proper API documentation.  Fix
that.

Fixes #20993