zephyrproject-rtos/zephyr #21000: Bluetooth: L2CAP: Clean up fixed & dynamic channel abstractions

jhedberg, 2019-11-26:
There's a whole bunch of confusion between the `bt_l2cap_chan` and `bt_l2cap_le_chan` abstractions. Almost all of the stuff in `bt_l2cap_le_chan` is only relevant for connection-oriented channels, yet the L2CAP APIs creation of this struct also for fixed channels. This seems to have been done to allow using as much common code as possible for the two channels, however that can be solved by moving the common parts to `bt_l2cap_chan` and using `bt_l2cap_le_chan` strictly for dynamic channels (possibly with a name change to make this clear).

jhedberg, 2019-11-26:
@joerchan FYI

joerchan, 2020-02-14:
@jhedberg Moved this out of the 2.2 release since it is an enhancement.

jhedberg, 2020-02-14:
> @jhedberg Moved this out of the 2.2 release since it is an enhancement.

@joerchan sounds good, thanks. I also removed myself from the assignees since @Vudentz will cover this as part of his other L2CAP patches (for Bluetooth 5.2).

joerchan, 2020-07-20:
@Vudentz Should this be closed? Did you do this as part of your Bluetooth Host 5.2 features PR?

carlescufi, 2021-02-04:
@Vudentz can we close this or does this still apply?

Vudentz, 2021-02-08:
> @Vudentz can we close this or does this still apply?

This is still open, we might want to take coc specifics out though while refactoring it.

jhedberg, 2024-02-13:
@carlescufi @jori-nordic there was actually an existing enhancement issue about what we discussed last week.