zephyrproject-rtos/zephyr #21010: drivers: CAN: call can_state_change_isr on 96 warning

alexanderwachter, 2019-11-26:
The can_state_change_isr is only called on bus-state changes, but there is also a warning level defined in the CAN standard.
My suggestion is to call the ISR when one of the error counters reaches 96, but not introduce a new state, because there is no warning state defined in the standard.
Tho user can check the warnings by comparing the error counter warnings.

alexanderwachter, 2019-11-26:
ping @henrikbrixandersen @karstenkoenig @nixward

henrikbrixandersen, 2019-11-26:
Why not simply introduce a warning state in the Zephyr CAN API?

alexanderwachter, 2019-11-26:
Because "warning" is not a state, it does not change the behavior of the node.

henrikbrixandersen, 2019-11-27:
> Because "warning" is not a state, it does not change the behavior of the node.

And yet you propose to issue an ISR when the condition occurs?

alexanderwachter, 2019-11-28:
> And yet you propose to issue an ISR when the condition occurs?

Exactly. Issue an ISR, but don't change the state. The user can then compare the counters to identify that it is a warning. It should work for CANopen, I guess.

henrikbrixandersen, 2019-11-28:
> Exactly. Issue an ISR, but don't change the state. The user can then compare the counters to identify that it is a warning. It should work for CANopen, I guess.

In my opinion that makes the API a bit odd. For some ISR invocations, I need to check the state, for others I need to check the counters. If we issue a state ISR I think we should have a state for it.

alexanderwachter, 2019-11-28:
The counters are not representing a state. The state is the state defined by the state machine in the CAN standard. This state is read directly from the controller and should, in my opinion at least, not be changed in the software, depending on the counter values.

alexanderwachter, 2019-11-29:
@henrikbrixandersen We could introduce some error counter states and helpers to get them.
These are to ones from Linux, for example:
```C
/* error status of CAN-controller / data[1] */
#define CAN_ERR_CRTL_UNSPEC      0x00 /* unspecified */
#define CAN_ERR_CRTL_RX_OVERFLOW 0x01 /* RX buffer overflow */
#define CAN_ERR_CRTL_TX_OVERFLOW 0x02 /* TX buffer overflow */
#define CAN_ERR_CRTL_RX_WARNING  0x04 /* reached warning level for RX errors */
#define CAN_ERR_CRTL_TX_WARNING  0x08 /* reached warning level for TX errors */
#define CAN_ERR_CRTL_RX_PASSIVE  0x10 /* reached error passive status RX */
#define CAN_ERR_CRTL_TX_PASSIVE  0x20 /* reached error passive status TX */
```

They don't represent the controller state, but at least we are compatible to Linux then.
Splitting the states into CAN_ERR_CRTL_RX_* and CAN_ERR_CRTL_TX_* is weird IMO.
CAN_ERR_CRTL_TX_OVERFLOW can't happen in our drivers and CAN_ERR_CRTL_RX_OVERFLOW would be easy to introduce.

henrikbrixandersen, 2019-12-03:
Perhaps we could just add defines for the warning and error levels to header file? The Linux API for this is not the best either, I agree.

PR #41917 (2022-01-19): drivers: can: add error-warning state
Add CAN controller error-warning state and document the relationship between CAN controller states and CAN controller RX/TX error counter values. This new state corresponds to `CAN_STATE_ERROR_WARNING` in the Linux kernel (see https://elixir.bootlin.com/linux/latest/source/include/uapi/linux/can/netlink.h#L71).

Implement support for CAN controller error-warning state in the following drivers:
- NXP FlexCAN
- ST STM32 bxCAN
- Bosch M_CAN
- Microchip MCP2515
- Renesas RCAR (thanks to @julien-massot)

~I have not added support for error-warning to the Renesa [... cut]
commit f5f93e5f (2022-01-19): drivers: can: add error-warning state

Add CAN controller error-warning state and document the relationship
between CAN controller states and CAN controller RX/TX error counter
values.

Fixes: #21010
closed 2022-01-19 by carlescufi