zephyrproject-rtos/zephyr #21016: Unexpected ethernet network traffic after power up

dferbas, 2019-11-26:
New issue to separate it from #20960.
Platform Kinetis MK64, our own board with MK64FN1M0VLL12 package (similar to frdm_k64f).

TCP stack produces packets against RFC.
After it requires its IP settings from DHCP server, it bothers network with unsolicited ARP and ICMP echo requests.

---
I thought we have a bad Kconfig or something else, but as Oanerer commented similar behavior,
here is my observation with captured packets from Wireshark. I selected only dhcp or packets with board MAC address.

[Zephyr_stack_on_kinetis_mk64.zip](https://github.com/zephyrproject-rtos/zephyr/files/3893933/Zephyr_stack_on_kinetis_mk64.zip)

Problem is when a Windows PC acquires with DHCP same IP address, Zephyr sends an ARP request for it, and Windows think this IP address is duplicated on a network. Unfortunately if such an IP address is static, Windows PC asks for an IP address again and again. As I never saw such a behavior, it took a while until we found a source of problem - Zephyr stack.

---
I think it is better to stop development with Zephyr until this is fixed.

---
 - OS: Windows)
 - ZEPHYR_TOOLCHAIN_VARIANT=gnuarmemb
 - 7_2018-q2-update, 7.3.1 20180622

jukkar, 2019-11-27:
Could you provide ARP debug log (set CONFIG_NET_ARP_LOG_LEVEL_DBG=y) that matches the pcap file, so that it is easier to match the sent and received ARP packets. Also please provide your .config file.

Oanerer, 2019-11-27:
[Zephyr_ARP_debuglog_probe_mimicking.zip](https://github.com/zephyrproject-rtos/zephyr/files/3895625/Zephyr_ARP_debuglog_probe_mimicking.zip)

The Zephyr node consists of a SiLabs EFM32GG11 dev board (not upstreamed yet). MAC: `InterayS_01:b9:74` at IP 192.168.9.135. The Windows 7 PC trying to join the network has MAC `HewlettP_b8:39:20`. 

The Wireshark capture contains only ARP and DHCP packets for these nodes. The Zephyr log was cut to where the W7 PC tries to join the network using IP 192.168.9.198 it was offered by the DHCP server, but it later declines due to the Zephyr node mimicking the ARP probes sent by the W7 PC. Also my .config is included. 

I have recently increased `CONFIG_NET_ARP_TABLE_SIZE` to 32 in my prj.conf and that seemed to make the problem go away (or it may be masked or something, dunno). I hope you can make sense of it all. Let me know if you need additional info.

jukkar, 2019-11-27:
Does the issue go away if you disable gratuitous ARP by setting `CONFIG_NET_ARP_GRATUITOUS=n`? Please try this with default ARP table size (2).

jukkar, 2019-11-27:
Also if possible, please enable `CONFIG_NET_ARP_LOG_LEVEL_DBG=y` and `CONFIG_NET_L2_ETHERNET_LOG_LEVEL_DBG=y` and attach the output here.

Oanerer, 2019-11-27:
[003_Zephyr_ARP_probe_mimicking.zip](https://github.com/zephyrproject-rtos/zephyr/files/3897397/003_Zephyr_ARP_probe_mimicking.zip)

`CONFIG_NET_ARP_GRATUITOUS=n`
`CONFIG_NET_ARP_LOG_LEVEL_DBG=y`
`CONFIG_NET_L2_ETHERNET_LOG_LEVEL_DBG=y`

Disabling gratuitous ARPs does not seem to fix the problem. 

It seems it is impossible to capture one of these occurrences without messages being dropped, but I hope you can still get enough information out of this.

jukkar, 2019-11-28:
I did not spot anything obviously wrong in the log files. Some lines in the log were skipped and were missing.

Anyway, then I tried to replicate the system using same IP addresses etc. and then sending similar packets to zephyr.
I used `samples/net/sockets/echo_server` application with following config

**overlay-arp-test.conf**
```
CONFIG_NET_CONFIG_MY_IPV4_ADDR="192.168.9.189"
CONFIG_NET_CONFIG_MY_IPV4_NETMASK="255.255.255.0"
CONFIG_NET_CONFIG_MY_IPV4_GW="192.168.9.250"
CONFIG_NET_ARP_TABLE_SIZE=2
CONFIG_NET_ARP_GRATUITOUS=y
CONFIG_NET_ARP_LOG_LEVEL_DBG=y
CONFIG_NET_L2_ETHERNET_LOG_LEVEL_DBG=y
CONFIG_LOG_STRDUP_BUF_COUNT=50
CONFIG_LOG_BUFFER_SIZE=16024
```
Then compiled echo-server like this `cmake -Bbuild -DOVERLAY_CONFIG="overlay-e1000.conf overlay-arp-test.conf" . -DBOARD=qemu_x86`

Created a host network config file `zeth.arp.test.conf` so that host will act as a gateway
```
INTERFACE="$1"
HWADDR="01:1a:04:01:b9:75"
IPV6_ADDR_1="2001:db8::2"
IPV6_ROUTE_1="2001:db8::/64"
IPV4_ADDR_1="192.168.9.250/24"
IPV4_ROUTE_1="192.168.9.0/24"
ip link set dev $INTERFACE up
ip link set dev $INTERFACE address $HWADDR
ip -6 address add $IPV6_ADDR_1 dev $INTERFACE nodad
ip -6 route add $IPV6_ROUTE_1 dev $INTERFACE
ip address add $IPV4_ADDR_1 dev $INTERFACE
ip route add $IPV4_ROUTE_1 dev $INTERFACE > /dev/null 2>&1
```
Started zeth like this: `net-tools/net-setup.sh -c zeth.arp.test.conf`
The zeth interface looks like this now:
```
zeth: flags=4099<UP,BROADCAST,MULTICAST>  mtu 1500
        inet 192.168.9.250  netmask 255.255.255.0  broadcast 0.0.0.0
        inet6 2001:db8::2  prefixlen 128  scopeid 0x0<global>
        ether 86:c8:c2:6d:a8:01  txqueuelen 1000  (Ethernet)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
```

And then start zephyr: `make -C build run`
The network setup inside zephyr looks like this:
```
uart:~$ net iface

Interface 0x145ce0 (Ethernet) [1]
===================================
Link addr : 52:54:00:12:34:56
MTU       : 1500
Ethernet capabilities supported:
	10 Mbits
	100 Mbits
	1 Gbits
IPv6 unicast addresses (max 3):
	fe80::5054:ff:fe12:3456 autoconf preferred infinite
	2001:db8::1 manual preferred infinite
IPv6 multicast addresses (max 4):
	ff02::1
	ff02::1:ff12:3456
	ff02::1:ff00:1
IPv6 prefixes (max 2):
	<none>
IPv6 hop limit           : 64
IPv6 base reachable time : 30000
IPv6 reachable time      : 21898
IPv6 retransmit timer    : 0
IPv4 unicast addresses (max 1):
	192.168.9.189 manual preferred infinite
IPv4 multicast addresses (max 1):
	<none>
IPv4 gateway : 192.168.9.250
IPv4 netmask : 255.255.255.0
```

After this the networking setup should resemble what is described in the issue report (except the MAC address of zephyr is not the same).

Then I used following script to send ARP messages and seeing what Zephyr thinks about them:
```
#!/usr/bin/env python3

from scapy.all import *

import time

# Gratuitous arp for some random IP address
arp = Ether(
    src="d4:85:64:b8:39:20", dst="ff:ff:ff:ff:ff:ff", type=0x0806)/\
    ARP(hwtype=1, ptype=0x0800, hwlen=6, plen=4, op=1,
        hwsrc="d4:85:64:b8:39:20", hwdst="ff:ff:ff:ff:ff:ff",
        psrc="169.254.169.134", pdst="169.254.169.134")

sendp(arp, iface="zeth")

# Initial query
arp = Ether(
    src="d4:85:64:b8:39:20", dst="ff:ff:ff:ff:ff:ff", type=0x0806)/\
    ARP(hwtype=1, ptype=0x0800, hwlen=6, plen=4, op=1,
        hwsrc="d4:85:64:b8:39:20", hwdst="00:00:00:00:00:00",
        psrc="0.0.0.0", pdst="192.168.9.198")

sendp(arp, iface="zeth")

# Some random query
arp = Ether(
    src="d4:85:64:b8:39:20", dst="ff:ff:ff:ff:ff:ff", type=0x0806)/\
    ARP(hwtype=1, ptype=0x0800, hwlen=6, plen=4, op=1,
        hwsrc="d4:85:64:b8:39:20", hwdst="00:00:00:00:00:00",
        psrc="192.168.9.198", pdst="192.168.9.250")

sendp(arp, iface="zeth")

# Some random query
arp = Ether(
    src="d4:85:64:b8:39:20", dst="ff:ff:ff:ff:ff:ff", type=0x0806)/\
    ARP(hwtype=1, ptype=0x0800, hwlen=6, plen=4, op=1,
        hwsrc="d4:85:64:b8:39:20", hwdst="00:00:00:00:00:00",
        psrc="192.168.9.198", pdst="192.168.9.250")

sendp(arp, iface="zeth")

# Initial query
arp = Ether(
    src="d4:85:64:b8:39:20", dst="ff:ff:ff:ff:ff:ff", type=0x0806)/\
    ARP(hwtype=1, ptype=0x0800, hwlen=6, plen=4, op=1,
        hwsrc="d4:85:64:b8:39:20", hwdst="00:00:00:00:00:00",
        psrc="0.0.0.0", pdst="192.168.9.198")

sendp(arp, iface="zeth")
```

Everything looks normal after these steps and I was not able to replicate the issue. Zephyr did not send any random ARP queries in my tests.

Oanerer, 2019-11-28:
[004_Zephyr_ARP_probe_mimicking.zip](https://github.com/zephyrproject-rtos/zephyr/files/3902293/004_Zephyr_ARP_probe_mimicking.zip)

This time all log messages are there. Retried with a fresh Zephyr installation and the following configuration options:
```
CONFIG_NET_ARP_GRATUITOUS=y
CONFIG_NET_ARP_TABLE_SIZE=2
CONFIG_NET_ARP_LOG_LEVEL_DBG=y
CONFIG_NET_L2_ETHERNET_LOG_LEVEL_DBG=y
CONFIG_LOG_STRDUP_BUF_COUNT=100
CONFIG_LOG_BUFFER_SIZE=16024
```
The strange thing is: I can *instantly* reproduce the issue. The Windows 7 PC reports an IP conflict as soon as I disable/enable the network adapter and Windows is trying to get a new IP through DHCP.

From what you were trying to achieve with your Python script I would've expected that the 'initial query' would show exactly what me and dferbas are also seeing. Let me know if I need to run some more tests for you if that helps.

dferbas, 2019-11-28:
@jukkar: Did you try the test with IP v6 disabled?

jukkar, 2019-12-05:
> Did you try the test with IP v6 disabled?

No, IPv6 was enabled all the time but that does not matter here as ARP is IPv4 only protocol.

Oanerer, 2019-12-05:
@jukkar : could you please connect your SAM E70 Xplained board with `CONFIG_NET_ARP_TABLE_SIZE=2` to a busy company network (lots of ARP traffic etc flying around). Next please observe through Wireshark if you see similar patterns as dferbas and me encountered when a Windows 7 system is trying to get an IP address through DHCP (and fails to do so because of duplicate IPs/ARP probe mimicking).

dferbas, 2019-12-05:
@jukkar: I know ARP is IPv4, but idea is to have as same as possible configurations. Because a part of zephyr stack can influence other part. Of course probability of finding this as a reason is minimal, my experience says it makes sense to duplicate configuration for testing.

PierreAronnax, 2020-02-21:
I am a colleague of @Oanerer and can also easily reproduce this problem.

I don’t think its related to `CONFIG_NET_ARP_TABLE_SIZE`, as I can reproduce this problem even before the table is full. At best increasing `CONFIG_NET_ARP_TABLE_SIZE` only delays the problem, but it is still there.

I have tracked the problem down to `net_arp_prepare`
https://github.com/zephyrproject-rtos/zephyr/blob/master/subsys/net/l2/ethernet/arp.c#L349
```
	/* If the destination address is already known, we do not need
	 * to send any ARP packet.
	 */
	entry = arp_entry_find_move_first(net_pkt_iface(pkt), addr);
	if (!entry) {
		struct net_pkt *req;

		entry = arp_entry_find_pending(net_pkt_iface(pkt), addr);
		if (!entry) {
			/* No pending, let's try to get a new entry */
			entry = arp_entry_get_free();
			if (!entry) {
				/* Then let's take one from table? */
				entry = arp_entry_get_last_from_table();
			}
		} else {
			/* There is a pending already */
			entry = NULL;
		}

		req = arp_prepare(net_pkt_iface(pkt), addr, entry, pkt,
				  current_ip);

		if (!entry) {
			/* We cannot send the packet, the ARP cache is full
			 * or there is already a pending query to this IP
			 * address, so this packet must be discarded.
			 */
			NET_DBG("Resending ARP %p", req);
		}

		return req;
	}
```
The problem occurs when there is a pending query found, and `entry` is set to NULL. The logic goes into resending the ARP. When I return NULL instead of `req` the problem goes away.
As I am not familiar with ARP I cannot oversee the complete impact. Consider this a piece of the puzzle.

> 
> 
> @jukkar: Did you try the test with IP v6 disabled?

I also tried with ipv6 disabled, this makes no difference.

jukkar, 2020-02-21:
> When I return NULL instead of `req` the problem goes away.
> As I am not familiar with ARP I cannot oversee the complete impact.

If this function returns null, it will cause the TX packet to be dropped i.e., not sent. See ethernet.c `ethernet_send() -> ethernet_ll_prepare_on_ipv4() -> net_arp_prepare()`. That is probably not the intention here.

PierreAronnax, 2020-02-21:
Yes I understand, but that is how I interpret that dropping was intentional:
> 			/* We cannot send the packet, the ARP cache is full
> 			 * or there is already a pending query to this IP
> 			 * address, so this packet must be discarded.
> 			 */

jukkar, 2020-02-21:
I am trying to replicate the issue, so far no luck. I am bombing zephyr with ARP requests but zephyr so far is only replying to requests that are directly sent to it. Have you figured out an ARP packet that is causing this issue?

PierreAronnax, 2020-02-24:
Just get any PC to request a DHCP lease. A Windows 7 PC will give an error that the IP is already in use (I only have a Dutch translation of the error), a Windows 10 will accept, but the message sequence look the same to me (attached below).
```
ipconfig /release
ipconfig /renew
```

I have added `NET_INFO` to `arp_entry_register_pending`, `arp_entry_find_pending` and `arp_entry_get_pending` for easy filtering; for `_find_` and `_get_` only in case `arp_entry_find` actually finds a result.

You can see multiple calls to `arp_entry_find_pending`. During the DHCP request `arp_entry_find_pending` finds a pending packet and `entry` is set to NULL.

I have tracked the issue down to [net: IPv4 link local support]( https://github.com/zephyrproject-rtos/zephyr/commit/00e37cbb5b2be0fde9951335100b844d0a2e3363) where the value of `my_addr` when entry == NULL has been changed from `&NET_IPV4_HDR(pending)->src` to `current_ip`.
I have added `NET_INFO` to show that they are not the same. When I change the line back to 
```
my_addr = &NET_IPV4_HDR(pending)->src;
```
the problem goes away.
`NET_IPV4_HDR(pending)->src` is 192.168.9.255 is the broadcast address of my 192.168.9.0/24 network.
`current_ip` is 56.78.0.32 is unknown to me.

**To reproduce my test:**
```
CONFIG_NET_LOG=y
CONFIG_NET_ARP_LOG_LEVEL_DBG=y
CONFIG_LOG_BUFFER_SIZE=10240
CONFIG_LOG_STRDUP_BUF_COUNT=128
```
```
 subsys/net/l2/ethernet/arp.c | 16 ++++++++++++++--
 1 file changed, 14 insertions(+), 2 deletions(-)

diff --git a/subsys/net/l2/ethernet/arp.c b/subsys/net/l2/ethernet/arp.c
index cac51dec4f..21be72296a 100644
--- a/subsys/net/l2/ethernet/arp.c
+++ b/subsys/net/l2/ethernet/arp.c
@@ -102,9 +102,15 @@ static inline
 struct arp_entry *arp_entry_find_pending(struct net_if *iface,
 					 struct in_addr *dst)
 {
+	struct arp_entry *entry;
+
 	NET_DBG("dst %s", log_strdup(net_sprint_ipv4_addr(dst)));
 
-	return arp_entry_find(&arp_pending_entries, iface, dst, NULL);
+	entry = arp_entry_find(&arp_pending_entries, iface, dst, NULL);
+	if (entry) {
+		NET_INFO("arp_entry_find_pending dst %s", log_strdup(net_sprint_ipv4_addr(dst)));
+	}
+	return entry;
 }
 
 static struct arp_entry *arp_entry_get_pending(struct net_if *iface,
@@ -117,6 +123,8 @@ static struct arp_entry *arp_entry_get_pending(struct net_if *iface,
 
 	entry = arp_entry_find(&arp_pending_entries, iface, dst, &prev);
 	if (entry) {
+		NET_INFO("arp_entry_get_pending dst %s", log_strdup(net_sprint_ipv4_addr(dst)));
+
 		/* We remove the entry from the pending list */
 		sys_slist_remove(&arp_pending_entries, prev, &entry->node);
 	}
@@ -165,6 +173,7 @@ static struct arp_entry *arp_entry_get_last_from_table(void)
 static void arp_entry_register_pending(struct arp_entry *entry)
 {
 	NET_DBG("dst %s", log_strdup(net_sprint_ipv4_addr(&entry->ip)));
+	NET_INFO("arp_entry_register_pending dst %s", log_strdup(net_sprint_ipv4_addr(&entry->ip)));
 
 	sys_slist_append(&arp_pending_entries, &entry->node);
 
@@ -299,7 +308,10 @@ static inline struct net_pkt *arp_prepare(struct net_if *iface,
 	       sizeof(struct net_eth_addr));
 
 	if (!entry || net_pkt_ipv4_auto(pkt)) {
-		my_addr = current_ip;
+		my_addr = &NET_IPV4_HDR(pending)->src;
+		NET_INFO("my_addr %s != %s",
+				log_strdup(net_sprint_ipv4_addr(current_ip)),
+				log_strdup(net_sprint_ipv4_addr(&NET_IPV4_HDR(pending)->src)));
 	} else {
 		my_addr = if_get_addr(entry->iface, current_ip);
 	}
```

**My test results:**
[arp_without_fix.txt](https://github.com/zephyrproject-rtos/zephyr/files/4244787/arp_without_fix.txt)
Arp debug with added NET_INFO without reverting my_addr.
A unkown pc is requesting IP 192.168.9.147 every 2 seconds.

[arp_win7_with_fix.txt](https://github.com/zephyrproject-rtos/zephyr/files/4244788/arp_win7_with_fix.txt)
Arp debug with added NET_INFO and reverted my_addr
A Win 7 pc is requesting IP 192.168.9.104

[arp_win10_with_fix.txt](https://github.com/zephyrproject-rtos/zephyr/files/4244789/arp_win10_with_fix.txt)
Arp debug with added NET_INFO and reverted my_addr
A Win 10 pc is requesting IP 192.168.9.130

PierreAronnax, 2020-02-25:
My information about `current_ip` is 56.78.0.32 was false. current_ip is actually NULL and happens to print 56.78.0.32.

dferbas, 2020-08-13:
Was this issue somehow handled? If yes, please give a link to a solving issue. Otherwise I still treat Zephyr as not ready for IP communication.

I think Zephyr has future, but it needs to fix basic communication issues.
With our project on MK64, we returned to the Nut/OS (www.ethernut.de, also on sourceforge.net and in my github repo), which has streams for both files and sockets, ioctl for TCP and other communication.

If someone wants to cooperate to improve Zephyr, we have an experience and knowledges to fix these issues, but we have no budget to do it on our own. You can email me to dferbas@solarmonitor.cz or dferbas@etech.cz.

jukkar, 2020-08-13:
> Was this issue somehow handled? If yes, please give a link to a solving issue. Otherwise I still treat Zephyr as not ready for IP communication.

There has not been any progress with this as I was not able to reproduce the issue. I understand your frustration with this.

> If someone wants to cooperate to improve Zephyr, we have an experience and knowledges to fix these issues, but we have no budget to do it on our own.

Not sure if I understand you here. Do you mean that you would be able to fix the issue if some company would hire you to do so?

dferbas, 2020-08-14:
I think, we can fix it. See my profile on [Linkedin](https://www.linkedin.com/in/du%C5%A1an-ferbas-809b2424/). I have experience with embedded TCP stack development since 1997. I started with fixing modem code in assembler on Motorola 68k, then implemented snmp for a rack with 32 modems with Pacific Softworks TCP stack and Philips' uC OS I, TCP stack was then sold to Net Silicon and then to Unicoi Systems. Over 60.000 units were sold under Allen-Bradley Stratix 6000 ([1783-EMS08T](https://www.rockwellautomation.com/en-us/products/hardware/allen-bradley/discontinued-products/stratix-6000-fixed-managed.html)), e.g. whole Caterpillar was using it. Then we implemented several systems with Nut/OS, which started with Adam Dunkel's code (I think lwip started from the same sources). Finally we worked with Yocto and implemented a BSP and watchdog driver for a custom board with Layerscape Qoriq LS1020A (2x 32b 1.2GHz ARM Cortex-A7, 256 MB flash, 1GB 32b RAM, 3x 1Gb LAN, used in Allen-Bradley [9300-ENA](https://www.rockwellautomation.com/en-us/products/hardware/allen-bradley/networks-and-communications/ethernet-networks/9300-nat.html)).

PierreAronnax, 2020-08-15:
For what it is worth, I currently have the following patch in my production software. Not saying this is a fix, but it works for me for fix the issue described in https://github.com/zephyrproject-rtos/zephyr/issues/21016#issuecomment-590330145.

```
 subsys/net/l2/ethernet/arp.c | 2 +-
 1 file changed, 1 insertion(+), 1 deletion(-)

diff --git a/subsys/net/l2/ethernet/arp.c b/subsys/net/l2/ethernet/arp.c
index cac51dec4f..482ad1d756 100644
--- a/subsys/net/l2/ethernet/arp.c
+++ b/subsys/net/l2/ethernet/arp.c
@@ -299,7 +299,7 @@ static inline struct net_pkt *arp_prepare(struct net_if *iface,
 	       sizeof(struct net_eth_addr));
 
 	if (!entry || net_pkt_ipv4_auto(pkt)) {
-		my_addr = current_ip;
+		my_addr = current_ip ? current_ip : &NET_IPV4_HDR(pending)->src;
 	} else {
 		my_addr = if_get_addr(entry->iface, current_ip);
 	}

```

leokrueger, 2020-09-16:
> 
> 
> I am trying to replicate the issue, so far no luck. I am bombing zephyr with ARP requests but zephyr so far is only replying to requests that are directly sent to it. Have you figured out an ARP packet that is causing this issue?

I think this issue is not about Zephyr answering to ARP requests, but Zephyr sending out ARP requests about an IP address which it saw in a broadcasted DHCP Request paket.

leokrueger, 2020-09-17:
Correcting myself here. We continued our investigation. Between the (broadcasted) DHCP Request of a Windows machine and the DHCP Decline (also broadcasted) we see, the respective Windows machine already sends out a couple (3 it semms) NetBios messages with the IP as source (192.168.100.5 in our test setup) it will shortly after decline. These NetBios messages are directed to the network's broadcast address (192.168.100.255 in our testnetwork). Strangely, Zephyr reacts to these broadcast messages with an ARP request asking for the MAC of the source address used by Windows (192.168.100.5 as mentioned above). Somehow, this leads Windows to believe the address is already taken, so sending out a DHCP decline.

So it is not Zephyr reacting to broadcasted DHCP requests, but Zephyr reacting to Broadcasts (to the network's broadcast address) in general by sending out an ARP request for the source of the broadcast it just received. As far as I know (and as far as I saw on our network), Zephyr is the only operating system reacting to those broadcasts in this way. From that, I conclude that this migth not be correct behaviour, especially since Zephyr does not need to react to those broadcasts and therefore does not need to do an ARP request.

Here you can see an excerpt from tcpdump showing the broadcast "emulating" the mentioned NetBios broadcasts and the successive ARP request from Zephyr. As long as Zephyr does not have anything in its ARP table, it will respond to the broadcast with the ARP request.

````
11:57:43.176413 IP (tos 0x0, ttl 128, id 3078, offset 0, flags [none], proto UDP (17), length 96)
    192.168.100.5.137 > 192.168.100.255.137: [udp sum ok] UDP, length 68

11:57:43.176674 ARP, Ethernet (len 6), IPv4 (len 4), Request who-has 192.168.100.5 tell 192.168.100.100, length 50
````

@jukkar does that help in replicating the issues?

PierreAronnax, 2020-09-18:
> @PierreAronnax can you let us know the Zephyr version you are using? With your fix, we see that Zephyr sends three successive ARP requests, where the first has the correct source and the following requests have the broadcast address as source. Which, at least in our opinion, is also not the correct behaviour.

I am using Zephyr 2.1.0
Again, this is not a fix. This is a workaround that works good enough for me until this is fixed properly.

jukkar, 2020-09-19:
@leokrueger thanks for the info! Would you happen to have a pcap file of the packet that caused the issue?

leokrueger, 2020-09-21:
@jukkar please find attached a pcap (in a zip, thanks GitHub...) with the NetBios UDP broadcast from a Raspberry Pi and the resulting ARP requests (which, as explained before, should not be there in our opinion).

[2020-09-21-NetBios-UDP-Broadcast-and-ARP-Requests.zip](https://github.com/zephyrproject-rtos/zephyr/files/5254204/2020-09-21-NetBios-UDP-Broadcast-and-ARP-Requests.zip)

jukkar, 2020-09-21:
@leokrueger fix submitted, please try.

leokrueger, 2020-09-21:
@jukkar we testedt your patch on top of the current master and we see that the board running your patch does not send out the ARP request in reaction to a broadcast. So: Looks good! Thanks!

PR #28563 (2020-09-21): net: conn: Ignore unhandled IPv4 broadcast packets
If there is no handler for IPv4 broadcast packet, then ignore it
instead of trying to send an ARP message to resolve the senders
address.

Fixes #21016
commit 8afaadd2 (2020-09-21): net: conn: Ignore unhandled IPv4 broadcast packets

If there is no handler for IPv4 broadcast packet, then ignore it
instead of trying to send an ARP message to resolve the senders
address.

Fixes #21016
closed 2020-09-21 by MaureenHelm