zephyrproject-rtos/zephyr #21025: sam_e70_xplained reboots after 35secs

jukkar, 2019-11-27:
**Describe the bug**
Noticed that sam_e70_xplained board reboots after 35 seconds.

**To Reproduce**
Steps to reproduce the behavior:
1. `cd samples/net/gptp`
2. `cmake -Bbuild -DBOARD=sam_e70_xplained .`
3. `make -C build flash`

**Expected behavior**
The device does not boot.

**Screenshots or console output**
```
[00:00:00.000,000] <inf> i2c_sam_twihs: Device I2C_0 initialized
[00:00:00.001,000] <inf> eth_sam: MAC: fc:c2:3d:11:c1:47
[00:00:00.001,000] <inf> eth_sam: Queue 0 activated
[00:00:00.001,000] <inf> eth_sam: Queue 1 set to idle
[00:00:00.001,000] <inf> eth_sam: Queue 2 set to idle
[00:00:00.001,000] <inf> eth_sam_phy: Soft Reset of ETH PHY
[00:00:00.091,000] <inf> eth_sam_phy: PHYID: 0x221561 at addr: 0
uart:~$ ***** Booting Zephyr OS build v2.1.0-rc1-288-gea2324dd2a3a *****
[00:00:02.556,000] <inf> eth_sam_phy: common abilities: speed 100 Mb, full duplex
[00:00:02.564,000] <inf> net_config: Initializing network
[00:00:02.564,000] <inf> net_config: IPv4 address: 192.0.2.1
[00:00:02.656,000] <inf> net_config: IPv6 address: fe80::fec2:3dff:fe11:c147
[00:00:02.664,000] <inf> net_config: IPv6 address: fe80::fec2:3dff:fe11:c147
[00:00:03.562,000] <wrn> net_gptp: Reset Pdelay requests
[00:00:04.568,000] <wrn> net_gptp: Reset Pdelay requests
[00:00:05.574,000] <wrn> net_gptp: Reset Pdelay requests
[00:00:06.580,000] <wrn> net_gptp: Reset Pdelay requests
[00:00:07.586,000] <wrn> net_gptp: Reset Pdelay requests
[00:00:08.592,000] <wrn> net_gptp: Reset Pdelay requests
[00:00:09.598,000] <wrn> net_gptp: Reset Pdelay requests
[00:00:10.604,000] <wrn> net_gptp: Reset Pdelay requests
[00:00:11.610,000] <wrn> net_gptp: Reset Pdelay requests
[00:00:12.616,000] <wrn> net_gptp: Reset Pdelay requests
[00:00:13.622,000] <wrn> net_gptp: Reset Pdelay requests
[00:00:14.628,000] <wrn> net_gptp: Reset Pdelay requests
[00:00:15.634,000] <wrn> net_gptp: Reset Pdelay requests
[00:00:16.640,000] <wrn> net_gptp: Reset Pdelay requests
[00:00:17.646,000] <wrn> net_gptp: Reset Pdelay requests
[00:00:18.652,000] <wrn> net_gptp: Reset Pdelay requests
[00:00:19.658,000] <wrn> net_gptp: Reset Pdelay requests
[00:00:20.664,000] <wrn> net_gptp: Reset Pdelay requests
[00:00:21.670,000] <wrn> net_gptp: Reset Pdelay requests
[00:00:22.676,000] <wrn> net_gptp: Reset Pdelay requests
[00:00:23.682,000] <wrn> net_gptp: Reset Pdelay requests
[00:00:24.688,000] <wrn> net_gptp: Reset Pdelay requests
[00:00:25.694,000] <wrn> net_gptp: Reset Pdelay requests
[00:00:26.700,000] <wrn> net_gptp: Reset Pdelay requests
[00:00:27.706,000] <wrn> net_gptp: Reset Pdelay requests
[00:00:28.712,000] <wrn> net_gptp: Reset Pdelay requests
[00:00:29.718,000] <wrn> net_gptp: Reset Pdelay requests
[00:00:30.724,000] <wrn> net_gptp: Reset Pdelay requests
[00:00:31.730,000] <wrn> net_gptp: Reset Pdelay requests
[00:00:32.736,000] <wrn> net_gptp: Reset Pdelay requests
[00:00:33.742,000] <wrn> net_gptp: Reset Pdelay requests
[00:00:34.748,000] <wrn> net_gptp: Reset Pdelay requests
[00:00:35.754,000] <wrn> net_gptp: Reset Pdelay requests
uart:~$ 

[00:00:00.000,000] <inf> i2c_sam_twihs: Device I2C_0 initialized
...
```

**Environment (please complete the following information):**
- latest master at ea2324dd2a3a148b164f004382b025406b0ff577 but have seen this earlier too

ioannisg, 2019-11-27:
Have you tried to bisect this @jukkar ?

jukkar, 2019-11-27:
> Have you tried to bisect this @jukkar ?

Not yet, I can try that tomorrow.

jukkar, 2019-11-28:
The issue has been long there, the reboot is happening also in 2.0. I was not able to bisect it much further because it would require using older SDK and I do not want to start messing with that.

jukkar, 2019-11-28:
Setting `CONFIG_WATCHDOG=n` and `CONFIG_WDT_SAM=n` did not change anything, the device still reboots.

stephanosio, 2019-12-02:
I can confirm that the same problem occurs when running long-running tests like `tests/kernel/sched/schedule_api/kernel.sched`.

stephanosio, 2019-12-02:
Something is wrong with `sam_e70_xplained`.

The following tests fail at the moment:
```
tests/subsys/logging/log_core/logging.log_core
tests/lib/mem_alloc/libraries.libc.newlib
tests/kernel/sched/schedule_api/kernel.sched
tests/kernel/sched/schedule_api/kernel.sched.multiq
```

**Test failure messages**
```
starting test - test_calloc

    Assertion failed at ZEPHYR_BASE/tests/lib/mem_alloc/src/main.c:70: test_calloc: ((cptr) is NULL)
calloc failed, errno: 4217356
FAIL - test_calloc

starting test - test_memalloc_all

    Assertion failed at ZEPHYR_BASE/tests/lib/mem_alloc/src/main.c:168: test_memalloc_all: ((clc_ptr) is NULL)
calloc failed, errno: 4217356
FAIL - test_memalloc_all

starting test - test_log_backend_runtime_filtering

    Assertion failed at ZEPHYR_BASE/tests/subsys/logging/log_core/src/log_core_test.c:65: put: (log_msg_timestamp_get(msg) not equal to exp_timestamp)
Unexpected message index
FAIL - test_log_backend_runtime_filtering

...
```

stephanosio, 2019-12-03:
@jukkar Regarding the `samples/net/gptp` reset, I added `CONFIG_WDT_DISABLE_AT_BOOT=y` to `samples/net/gptp/prj.conf` and, just as with all the other issues involving reset, it no longer resets.

Can you try this and confirm if this works for you too?

jukkar, 2019-12-03:
> Can you try this and confirm if this works for you too?

Confirmed, the device starts to behave well after setting `CONFIG_WDT_DISABLE_AT_BOOT=y`. I wonder why I saw reboots when I disabled watchdog earlier. Anyway, thanks for the hint, that was really helpful.

stephanosio, 2019-12-03:
> I wonder why I saw reboots when I disabled watchdog earlier.

For SAM 7 series SoC, watchdog is enabled by default on reset (`WDT_MR.WDDIS=0`), so unless you explicitly disable it by enabling the watchdog driver and explicitly setting `CONFIG_WDT_DISABLE_AT_BOOT=y`, watchdog will reset the SoC.

This is extremely un-intuitive and should be fixed.

A related discussion can be found here:
https://github.com/zephyrproject-rtos/zephyr/pull/20607#issuecomment-560992779

pabigot, 2019-12-03:
Whether it's intuitive or not depends on perspective.  A case can be made either way (all boards behave the same; all boards behave in the way each vendor designed them to behave).  I completely agree it's annoying.

If you @stephanosio want  create an issue to discuss this I think that's reasonable, but since the watchdog API is "stable" changing its behavioral definition (currently vendor-oriented) may require some effort.

FWIW two kernel.sched tests fail consistently on sam_e70 for me due to watchdog, but not for @ioannisg, possibly due to differences in how hardware flow control on the UART affects serial output rate.  Eventually I'll add the disable flag to those tests.

stephanosio, 2019-12-03:
@pabigot By un-intuitive, I was referring to the fact that watchdog H/W is enabled and resets SoC when the watchdog driver is not enabled on SAM 7X (`CONFIG_WDT_SAM=n`). I think the watchdog API itself is fine in this case.

For a "fix", I was thinking something like doing `wdt->WDT_MR |= WDT_MR_WDDIS` when `CONFIG_WDT_SAM=n` in SoC init code, to ensure that watchdog H/W does not misbehave when the watchdog driver is not enabled.

stephanosio, 2019-12-03:
> FWIW two kernel.sched tests fail consistently on sam_e70 for me due to watchdog, but not for @ioannisg, possibly due to differences in how hardware flow control on the UART affects serial output rate. Eventually I'll add the disable flag to those tests.

@pabigot @ioannisg Is this using SAME70 Xplained or some other board? I have tested `tests/kernel/sched/schedule_api/kernel.sched` on two Xplained boards (one from 2017 and another we just bought) and they both fail this test.

pabigot, 2019-12-03:
Changing the init code for all SOCs that have a watchdog that's enabled in hardware to disable it on platform initialization if the watchdog driver isn't enabled makes some sense initially.

But the existing behavior allows for applications to provide their own watchdog control when `CONFIG_WDT_SAM=n`.  I don't think unilaterally disabling it just because Zephyr's driver isn't enabled is unconditionally good: somebody may be depending on how it currently behaves.

Yes, it's `sam_e70_xplained` that fails those tests for me.

jukkar, 2019-12-03:
> This is extremely un-intuitive and should be fixed.

Fully agree here. For me the sam-e70 is unusable if I do not set `CONFIG_WDT_DISABLE_AT_BOOT=y`, could we set it by default for this board?

stephanosio, 2019-12-03:
> > This is extremely un-intuitive and should be fixed.
> 
> Fully agree here. For me the sam-e70 is unusable if I do not set `CONFIG_WDT_DISABLE_AT_BOOT=y`, could we set it by default for this board?

@pabigot @jukkar I will create a PR addressing this issue (i.e. disabling the watchdog when the watchdog driver is not used with Kconfig option to leave it enabled if application desires).

stephanosio, 2019-12-21:
Closing this issue as #21142 made `WDT_DISABLE_AT_BOOT` `default y`.

closed 2019-12-21 by stephanosio