zephyrproject-rtos/zephyr #21048: timer case fail on qemu_xtensa and mps2_an385

wentongwu, 2019-11-28:
qemu_xtensa

tests/kernel/sched/schedule_api/kernel.sched.multiq
```
starting test - test_slice_reset

    Assertion failed at ZEPHYR_BASE/tests/kernel/sched/schedule_api/src/test_sched_timeslice_reset.c:93: thread_time_slice: (t <= expected_slice_max is false)
timeslice too big, expected 1100000 got 1138522

    Assertion failed at ZEPHYR_BASE/tests/kernel/sched/schedule_api/src/test_sched_timeslice_reset.c:90: thread_time_slice: (t >= expected_slice_min is false)
timeslice too small, expected 900000 got 14681

    Assertion failed at ZEPHYR_BASE/tests/kernel/sched/schedule_api/src/test_sched_timeslice_reset.c:90: thread_time_slice: (t >= expected_slice_min is false)
timeslice too small, expected 900000 got 6788
FAIL - test_slice_reset
```

tests/kernel/tickless/tickless_concept/
```
starting test - test_tickless_slice
elapsed slice 110, expected: <100, 110>
elapsed slice 110, expected: <100, 110>
elapsed slice 90, expected: <100, 110>

    Assertion failed at ZEPHYR_BASE/tests/kernel/tickless/tickless_concept/src/main.c:57: thread_tslice: t >= SLICE_SIZE is false

elapsed slice 0, expected: <100, 110>

    Assertion failed at ZEPHYR_BASE/tests/kernel/tickless/tickless_concept/src/main.c:57: thread_tslice: t >= SLICE_SIZE is false

FAIL - test_tickless_slice
===================================================================
Test suite tickless_concept failed.
===================================================================
PROJECT EXECUTION FAILED
```

mps2_an385
tests/kernel/timer/timer_monotonic/kernel.timer.monotonic
```
starting test - test_timer
k_ticks_to_cyc_floor32(1) = 250000
sys_clock_hw_cycles_per_sec() = 25000000
starting test - test monotonic timer
diff = -249936 (t_last = 499998 : t_now = 250062);i = 18458

    Assertion failed at ZEPHYR_BASE/tests/kernel/timer/timer_monotonic/src/main.c:78: test_timer: (errors is true)
errors = 1

FAIL - test_timer
===================================================================
Test suite timer_fn failed.
```

wentongwu, 2019-11-28:
failed again when run CI. 

Sanitycheck / mps2_an385:tests/kernel/timer/timer_monotonic/kernel.timer.monotonic / tests/kernel/timer/timer_monotonic/kernel.timer.monotonic
 failure: N/A

***** Booting Zephyr OS build v2.1.0-rc2-4-g432e7ac5b9bb *****
Running test suite timer_fn
===================================================================
starting test - test_timer
k_ticks_to_cyc_floor32(1) = 250000
sys_clock_hw_cycles_per_sec() = 25000000
starting test - test monotonic timer
diff = -249936 (t_last = 499998 : t_now = 250062);i = 18458

    Assertion failed at ZEPHYR_BASE/tests/kernel/timer/timer_monotonic/src/main.c:78: test_timer: (errors is true)
errors = 1

FAIL - test_timer
===================================================================
Test suite timer_fn failed.
===================================================================
PROJECT EXECUTION FAILED

wentongwu, 2019-11-28:
what I can tell is that the failure of mps2_an385 maybe related to icount mode.

ioannisg, 2019-11-28:
timer_monotonic does not fail when I run it locally, @pabigot could you confirm?

nashif, 2019-11-28:
None of these tests fail for me. They would have been caught in CI. Maye they fail under heavy load, a well known issue with Qemu?

nashif, 2019-11-28:
https://app.shippable.com/github/zephyrproject-rtos/zephyr/runs/57173/3/tests

nashif, 2019-11-28:
Looks like some recent changes to the timer drivers are causing this?

nashif, 2019-11-28:
I can now reproduce them locally, a retry get them to pass.

nashif, 2019-12-02:
Can't reproduce anymore.

closed 2019-12-02 by nashif