zephyrproject-rtos/zephyr #21058: BLE: Enable/Disable Automatic sending of Connection Parameter update request on Timeout.

kipm-ot, 2019-11-28:
Problem: 
We want our peripheral to have the Peripheral Preferred Connection Parameters characteristic in the Generic Access service, but we do NOT want the Zephyr stack of our peripheral to automatically send an L2CAP Connection Parameter Update Request if the central doesn't change the connection parameters within a certain time (this behavior is optional according to the spec). We want our peripheral application layer to decide if/when such requests are sent.

Currently, it seems that the Zephyr configuration setting CONFIG_BT_GAP_PERIPHERAL_PREF_PARAMS controls both functionalities at once. If and only if the setting is enabled, the characteristic will be present (in gatt.c), and the update request will automatically be sent (in conn.c). So there doesn't seem to be a way to get one without the other.
In conn.c, when the time out happens, conn_update_timeout() is invoked and Zephyr will Automatically sends out the Connection Parameter update request. If CONFIG_BT_GAP_PERIPHERAL_PREF_PARAMS is enabled it will take the values from Characteristics else internal values. 

Solution:
Add a new KConfig flag, say CONFIG_BT_GAP_AUTO_SEND_CONN_PARAMS_ON_TIMEOUT and this can be disabled if the Application layer do not want to automatially send the Update request on time out.
And update conn_update_timeout() in conn.c something like below.

#if defined (CONFIG_BT_GAP_AUTO_SEND_CONN_PARAMS_ON_TIMEOUT)
	#if defined (CONFIG_BT_GAP_PERIPHERAL_PREF_PARAMS)
	.......
	.......
	.......
	#endif
#endif

aescolar, 2019-11-28:
CC: @jhedberg

carlescufi, 2019-12-03:
@kipm-ot will you send a PR for this?

kipm-ot, 2019-12-04:
@carlescufi yes I am preparing the PR :)

PR #21174 (2019-12-09): BLE: Enable/Disable Automatic Sending of Connection Parameter update
request on GAP recommended timeout of 5 seconds.

Added a new Kconfig flag to enable/disable this feature.

Fixes: #21058
closed 2019-12-09 by jhedberg