zephyrproject-rtos/zephyr #21068: Conflicting documentation for device initialization

pfl, 2019-11-28:
In `include/device.h` the following instructions are provided for the `DEVICE_INIT` macro:
```
* @def DEVICE_INIT
 *
 * @brief Create device object and set it up for boot time initialization.
 *
 * @details This macro defines a device object that is automatically
 * configured by the kernel during system initialization. Note that
 * devices set up with this macro will not be accessible from user mode
 * since the API is not specified; whenever possible, use DEVICE_AND_API_INIT
 * instead.
 *
 * @param dev_name Device name. This must be less than Z_DEVICE_MAX_NAME_LEN
 * characters in order to be looked up from user mode with device_get_binding().
```
In the details section it is said that devices set up with this macro is not accessible from user mode, but the parameter dev_name is used to look up the device from user mode. The latter is probably false and was supposed to say kernel mode instead?

andrewboie, 2020-03-11:
the comment is correct. devices specified in this way will not be assigned to a subsystem API and will not be registered as kernel objects.

closed 2020-03-11 by andrewboie

pfl, 2020-03-12:
If the comment is correct then the text looks wrong and needs an update - or did I miss something, like an update to the documentation...?