zephyrproject-rtos/zephyr #21088: Bluetooth: Mesh: Send Model Message

LingoMeng, 2019-11-30:
According Mesh Profile 3.8.6 Keys Application keys are bound to network keys. This means application keys are only used in a context of a network key they are bound to. An application key shall only be bound to a single network key. A device key is implicitly bound to all network keys.

https://github.com/zephyrproject-rtos/zephyr/blob/42c5b0a7fafa3360b8d835d4b76d7b1bebbe9237/samples/bluetooth/mesh_demo/src/main.c#L233-L251

But what the above code shows is that when we send data, we need to specify `net_idx `and `app_idx`.
The current protocol stack does not confirm whether the two parameters are `bound `together. My task doesn't have to specify `net_idx`, this parameter should be generated automatically through bound with `app_idx`

jhedberg, 2019-12-01:
You're right that the net_idx can be implicitly derived from the app_idx. Removing app_idx completely from bt_mesh_msg_ctx is a bit trickier since that struct is also used by the transport layer to send transport control messages, and those messages are only encrypted with a network key (no app key involved). So removing the struct member would require some redesign of the stack, which I'm not convinced is worth it. What we could do is resolve the app_key to a net_key in bt_mesh_model_send so that the application doesn't need to initialise net_key to anything.

carlescufi, 2019-12-05:
@LingoMeng will you send a PR for this?

LingoMeng, 2019-12-06:
well, I think it's best to add some binding query statements at the beginning of `bt_mesh_model_send` function:
``` C
    if (!BT_MESH_IS_DEV_KEY(ctx->app_idx))
    {
        app_key = bt_mesh_app_key_find(ctx->app_idx);

        if (app_key == NULL)
        {
            BT_ERR("Unable find binded netkey app_idx: 0x%04x", ctx->app_idx);
            return -EINVAL;
        }
		
        ctx->net_idx = app_key->net_idx;
    }
```

@jhedberg What do you think?

carlescufi, 2020-01-09:
@trond-snekvik Can you also provide feedback on the proposal above?

jhedberg, 2020-01-09:
> @jhedberg What do you think?

@moringe yes, that would probably work, though we should probably also cover the device key case (which is always used together with the primary net key of the device that the device key belongs to, IIRC). Do you want to submit a PR for this?

moringe, 2020-01-12:
> > @jhedberg What do you think?
> 
> @moringe yes, that would probably work, though we should probably also cover the device key case (which is always used together with the primary net key of the device that the device key belongs to, IIRC). Do you want to submit a PR for this?

But,according Mesh Profile 3.8.6.1 Device key The device key (DevKey) is an access layer key known only to the node and a Configuration Client. Thedevice key shall be bound to every network key known to the node.

jhedberg, 2020-01-12:
> But,according Mesh Profile 3.8.6.1 Device key The device key (DevKey) is an access layer key known only to the node and a Configuration Client. Thedevice key shall be bound to every network key known to the node.

@moringe you're right. Somehow I had missed this part. I had (wrongly) assumed that the device key is only bound to the network key that it comes together with as part of the provisioning data.

PR #24504 (2020-04-23): Bluetooth: Mesh: Remove net_idx params when with app key
According Mesh Profile 1.0.1. A application key shall
binding single network key. And Device key shall bind all
network key, and dev key only known by cfg_cli and node self,
only used by cfg_cli & cfg_srv.

Fixes: #21088
commit be33f913 (2020-04-23): Bluetooth: Mesh: Remove net_idx params when with app key

According Mesh Profile 1.0.1. A application key shall
binding single network key. And Device key shall bind all
network key, and dev key only known by cfg_cli and node self,
only used by cfg_cli & cfg_srv.

Fixes: #21088
closed 2020-04-23 by jhedberg