zephyrproject-rtos/zephyr #21094: native_posix doesn't call main function that's defined in C++

jmgao, 2019-12-01:
native_posix uses [the following macro:](https://github.com/zephyrproject-rtos/zephyr/blob/master/arch/posix/include/posix_cheats.h#L43) 
`#define main(...) zephyr_app_main(__VA_ARGS__)` to rename the app's main to something else (which is presumably linked with a no-op weak symbol that gets overridden by the app's main if it exists).

This works fine in C, but if you define your main function in a C++ file, because the function name isn't the magically special-cased 'main' anymore, the function name gets mangled into something along the lines of `_Z15zephyr_app_mainv` (for `void main(void)`), which results in the main function getting silently ignored.

I'm not sure there's actually any way to make the macro work: you can't use `asm` to specify the name of the symbol in the definition of a function, only the declaration. If it's possible to require main to be `void(void)`, the macro could be replaced by a forward declaration of `void main(void) asm("zephyr_app_main")`, but this might break existing code. Grepping around, it looks like there's one sample that might be broken by this: https://github.com/zephyrproject-rtos/zephyr/blob/master/samples/userspace/prod_consumer/src/main.c#L50.

aescolar, 2019-12-01:
@jmgao C++ support is quite experimental in Zephyr

aescolar, 2019-12-02:
@jmgao Do you have some simple case to reproduce it?

jmgao, 2019-12-02:
Oops, missed your edit and uploaded another test case.

If you're wondering what I actually ran into, it was samples/subsys/usb/hid segfaulting 2 seconds after usbip connection because k_delayed_work_init was never called.

aescolar, 2019-12-02:
@jmgao Hopefully that PR fixes the issue for you?

jmgao, 2019-12-02:
> @jmgao Hopefully that PR fixes the issue for you?

Yep, works for me.

> And by the way, thanks for reporting the bug and providing a sample to reproduce it.

No problem, thanks for the quick response!

PR #21363 (2019-12-18): POSIX arch: Fix C++ main() linkage issue
POSIX arch: Fix C++ main() linkage issue
    
To be able to define main() in C++ code we need to have its
prototype defined somewhere visibly. Otherwise name mangling
will prevent the linker from finding it.
Zephyr assumes a void main(void) prototype and therefore
this will be the prototype after renaming:
void zephyr_app_main(void);

Alternative to #21101
Fixes #21094

~Note that this PR includes #21100, as that fix is necessary for it~
closed 2019-12-18 by carlescufi