zephyrproject-rtos/zephyr #21103: Bluetooth: host: Single callback for incoming GATT notifications

wdhpawa, 2019-12-02:
**Is your enhancement proposal related to a problem? Please describe.**
The current way of receiving GATT notifications from a remote server requires the application layer of the client device to allocate a large block of data (struct bt_gatt_subscribe_params) per subscribed characteristic per peer. A reference to this block is passed to the Zephyr host when subscribing (bt_gatt_subscribe), and it contains a callback which is called by the host when a notification is received from the given peer and characteristic.

One problem with this approach is the potentially large amount of memory needed for this housekeeping data which serves no other purpose than to allow the Zephyr host to route incoming notifications to the correct callback.

Another problem is that there is no easy way to restore this housekeeping data after a reboot. Resubscribing to everything would be one way to do so, but that would be time-consuming and defeat the purpose of persisting the subscriptions on the server side for bonded devices (as is required by the BT Core spec v5.1 vol 3 part G section 3.3.3.3).

**Describe the solution you'd like**
The Zephyr host should provide an API allowing the application layer to register a single callback to be called for any incoming notification (or indication) from any peer and any attribute, irrespective of what subscriptions have been performed previously. The parameters to the callback should allow the callback to identify the connection and attribute, so the appropriate action can be taken.

This solution solves both problems mentioned above. It effectively moves the notification routing from the Zephyr host to the application layer where it can be hardcoded as part of the application logic, thus eliminating the need for maintaining awareness of subscriptions in the host and the difficulties associated with this.

**Describe alternatives you've considered**
None.

**Additional context**
None.

jsiverskog, 2019-12-03:
related: https://github.com/zephyrproject-rtos/zephyr/issues/21131

Vudentz, 2019-12-12:
We do intend to introduce a general discovery mechanism which would subscribe to service changed instead of custom made discovery for each service, so in the future we may have the stack subscribing as well.

I guess a probably better solution here is not eliminate the subscribe list completely but allow callbacks to work for multiple subscriptions, that way we keep it flexible enough that varios parts of the code can keep their own callbacks, though this makes unsubscribe less trivial Im afraid you would have to deal with the exact same complexity at the application side so why not attempt to do it on the stack?

Vudentz, 2019-12-12:
Im having a second thought on this, the problem with book keeping doesn't go away no matter what since one still has to know all the handles, value and CCC, peer, etc, making it work for multiple callbacks would have about the same logic as with a single one so the end result might consume as much memory, so we are just pushing the probably above with this. One of the things that could perhaps remove from bt_gatt_subscribe_params is the struct bt_att_req since that is only used for writing, Ive had the idea to remove it from all the params so instead we had a pool based on CONFIG_BT_ATT_TX_MAX.

Restoring is a different problem though, one that we could perhaps resolve with a flag indicating to the stack it just need to update

wdhpawa, 2019-12-12:
@Vudentz, yes, if we want the stack to retain any knowledge at all about individual subscriptions, we'll have to consider how to restore this knowledge after a reboot without having to go through the subscription process again. I'm not sure how all the information in the struct gets populated, but that would have to happen without any communication with the peer, and without requiring any local persistence of the values. And, as you say, it doesn't really solve the memory problem.

It is much easier for the application to do characteristic specific handling for the correct peer without having to store the peer identities again multiple times (the application already has a single list of peers), nor having to store the value handles again (the application already has a list of these for the purpose of reading/writing characteristics), nor having to remember the subscribe values (these are already hardcoded in the application logic).

Due to our platform limitations we're really focused on having as small a solution as possible, rather than having a very flexible, all-inclusive stack.

Vudentz, 2019-12-12:
> @Vudentz, yes, if we want the stack to retain any knowledge at all about individual subscriptions, we'll have to consider how to restore this knowledge after a reboot without having to go through [...]

I don't think that would work actually, you would need to remember the handles, peer, etc somehow, for instance if service changed is indicated. 
 
> It is much easier for the application to do characteristic specific handling for the correct peer without having to store the peer identities again multiple times (the application already has a [...]

What you are describing is exactly the reason we have this logic on the stack, so each and every application don't have to replicate it. That said I do see you can optimize some structures since you don't need to be generic enough to handle any situation.

> Due to our platform limitations we're really focused on having as small a solution as possible, rather than having a very flexible, all-inclusive stack.

Diverging the stack at this scale would not be beneficial, besides you would loose support to any profiles that do require subscription support, so you would need to rewrite them to work if your own method of subscriptions.

Vudentz, 2019-12-15:
Here is some idea how we could accomplished a the multiple callback notification with very small footprint:

1. Make a pool of bt_att_req so that doesn't need to be embedded on the params anymore
2. Make subscription list per peer, this increases a little the footprint on the stack side since we need an array of at least MAX_CONN + MAX_PAIRED but with this we can remove the peer member form the subscription params leaving it with just handle, value and liste node, which should be compact enough to scale even if the number of subscription is big.

wdhpawa, 2019-12-17:
@Vudentz, with your proposal it sounds like the client application might then be able to repopulate the subscription info after a reboot without communicating with the peer? Handles and subscribe values should be known to the application already, so if that is all that is needed, the restoring problem would be solved? Unless the pool of bt_att_req requires something more (I'm not sure what that is used for).

wdhpawa, 2019-12-17:
@Vudentz , by the way, regarding your previous comment, the client application's list of handles (not just for subscriptions) would of course be updated by a service discovery, if the server were to send a service changed indication, so that should work OK.

wdhpawa, 2020-01-09:
I've created a separate issue #21798 to deal with the problem of restoring the subscription info after a reboot.

commit d278cdc3 (2020-01-09): Bluetooth: GATT: Allocate request from a memory slab

This should reduce the footprint on applications that do a lot of
requests i.e have a lot of subscriptions.

Fixes #21103
closed 2020-01-09 by jhedberg