zephyrproject-rtos/zephyr #21113: k_sem_give reschedules cooperative threads unexpectedly

pabigot, 2019-12-02:
The [described behavior of cooperative threads](https://docs.zephyrproject.org/latest/reference/kernel/scheduling/index.html#cooperative-time-slicing) which specifically disallows a context switch until the thread is made unready or `k_yield()` is invoked.  

However, `k_sem_give()` apparently does this.  https://github.com/zephyrproject-rtos/zephyr/pull/20919#issuecomment-560532206 indicates this used to only happen for pre-emptible threads but that check was removed.

Reschedule of cooperative threads should match the documented behavior.

andrewboie, 2019-12-02:
k_sem_give() is not unique in this respect, all the IPC APIs which can wake another thread (except the bug in k_mutex) act in this way, as discussed in #20919 

Whatever solution we decide on, I feel the policy for this should be applied consistently to all the IPC APIs we have without special cases, and if we want to change it, need to fully understand what the implications are for end users.

andrewboie, 2019-12-02:
I think we can close this -- the kernel does the right thing with z_reschedule() calls if the caller was a co-op thread.

pabigot, 2019-12-03:
Agreed, both about consistency and about current cooperative thread behavior.

closed 2019-12-03 by pabigot