zephyrproject-rtos/zephyr #21119: VS Code integration

andrewboie, 2019-12-03:
Visual Studio Code is a really nice development environment, even someone such as myself who has stuck with vim for many years is considering switching over to it.

It would be cool if we could provide some documentation and/or VS Code extensions to make VS Code more zephyr-aware. This ticket can be a catch-all for integration tasks/ideas. So far I can think of:

- Best way to get tab settings correct across different file types in Zephyr
- Somehow populate header information to provide those definitions only provided by Kconfig, so we can correctly see what preprocessor directives are active, right now they are all greyed out for CONFIG_* variables. Probably need to be able to set what board/project configuration we are building for.
- Integration for interacting with west
- Integration for running CMake and doing builds for a particular project directory
- Interactive debugging to connected hardware
- Integration to run an emulator (and interactively debug it if desired) for QEMU or other emulation targets
- Some kinds of lintian checks can be shown directly in the editor GUI - wonder if we could integrate checkpatch into this

Other ideas, please comment!

ulfalizer, 2019-12-03:
For Kconfig, it's because `autoconf.h` is included automatically:

https://github.com/zephyrproject-rtos/zephyr/blob/9d5d2528947b512d85c212decaff7c1822d2e69d/CMakeLists.txt#L214

Would need to either include it manually where it's needed or tell VS Code about it somehow.

ulfalizer, 2019-12-03:
`cmake/ide/eclipse_cdt4_generator_amendment.cmake` seems to already have some stuff related to `autoconf.h`.

andrewboie, 2019-12-03:
> Would need to either include it manually where it's needed or tell VS Code about it somehow.

We'd need to generate it too, since it's a build artifact that depends on the selected board and defconfig.

Same deal with DTS #defines

nashif, 2019-12-03:
On Kconfig: https://github.com/luveti/kconfig-vscode

erwango, 2019-12-03:
Another nice improvement would be to enable search across modules, which I haven't been able to get working so far (while it is somewhat documented).

nashif, 2019-12-03:
>     * Best way to get tab settings correct across different file types in Zephyr

This should already be supported using the .editorconfig file we have in the tree. That works for me. Alternativly, you could set this directly:

https://code.visualstudio.com/docs/getstarted/settings#_language-specific-editor-settings

markusr, 2019-12-03:
There is a closed issue https://github.com/zephyrproject-rtos/zephyr/issues/17645 with some configs for vscode. Maybe it helps.

sigvartmh, 2019-12-05:
I would also add that some default configuration for https://marketplace.visualstudio.com/items?itemName=marus25.cortex-debug would be helpful for the people developing with ARM architecture.

I'm pretty sure this is what platform.io uses with some improvements https://platformio.org/platformio-ide

So having some support for 
- OS level awareness for debugging in VS code is also an improvement.

joerchan, 2019-12-09:
CC: @trond-snekvik  Maybe you have some input here?

trond-snekvik, 2019-12-09:
I actually made a couple of extensions that I'm not sure are quite ready for prime time yet, but they're working quite well for my day to day stuff:

- [vscode-DeviceTree](https://github.com/trond-snekvik/vscode-devicetree) adds syntax highlighting, type checking, code completion, go to defintion and syntax checking to deviceTree files, written specifically for the Zephyr RTOS.
- [vscode-Kconfig](https://github.com/trond-snekvik/vscode-kconfig) adds highlighting, code go to definition and completion for Kconfig and .conf files, as well as type checking, basic linting and syntax validation for .conf files. It has some significant performance issues still (it reparses the entire kconfig tree on every change), so use at your own risk :)
- [vscode-doxygen-generator](https://github.com/trond-snekvik/vscode-doxygen-generator) is a generic Doxygen header API generator for C. It improves syntax highlighting for doxygen tags, and adds a snippet generator for doxygen comments (Place your cursor on a function definition, and press Ctrl+Q).

The extensions can be installed by cloning them under ~/.vscode/extensions, and calling `npm install .` from inside the folder (let me know if there are any issues with this, as I've just installed this on my personal computers).

As mentioned, I've only used them for my personal setup, and they're not quite ready for public usage yet. They're all kind of personal pet projects, so they come with no guarantees. I'd like to make packages out of them and publish them to the VS Code extension repo eventually, so if anyone would like to try them out and provide some feedback, that would speed up the process! 
They're not very intrusive extensions (apart from the syntax highlighting, I suppose), so I think they're fairly safe to install :)

carlescufi, 2020-01-24:
One more potentially useful repo with info by @bus710:
https://github.com/bus710/zephyr-rtos-development-in-linux

brooksprumo, 2020-01-24:
I've had some good luck using VS Code and its support for Language Server to doing intellisense stuffs by having CMake generate a compilation database (that VS Code then consumes automatically, assuming the build directory is in the project).

```
west build -b <board> sample -- -DCMAKE_EXPORT_COMPILE_COMMANDS=1
```
```

sslupsky, 2020-03-14:
Here is a helpful setting in my workspace c_cpp_properties.json:
```
"compileCommands": "~/Documents/source/blinky/build/compile_commands.json"
```

I also found some helpful information here:
https://gitlab.endian.se/simon/vscode-zephyr-intro

In particular, the forced includes and defines to get IntelliSense to work properly:
```
"forcedInclude": [
                "~/Documents/source/blinky/build/zephyr/include/generated/autoconf.h",
                "~/Documents/source/zephyrproject/zephyr/include/toolchain/zephyr_stdint.h",
                "~/Documents/source/blinky/.vscode/c_cpp_undef.h"
],
"defines": [
                "BUILD_VERSION=v2.1.0-rc1-141-g3cc5bda2fa10",
                "KERNEL",
                "_FORTIFY_SOURCE=2",
                "__PROGRAM_START",
                "__ZEPHYR__=1"
],
```

sslupsky, 2020-03-14:
Also add this to your CMakeLists.txt for your project:
```
set(CMAKE_EXPORT_COMPILE_COMMANDS ON)
```

sslupsky, 2020-03-14:
CMake can be configured to generate a compile_commands.json.  However, the file is for the application which leads to problems when you open a component of the Zephyr RTOS into vscode.  For instance, if I open drivers/i2c/i2c_sam0.c I will see an error that there is no target for that file.  This is true since the compilation of the OS is controlled by other CMake processes.  Vscode opens the file and parses using the include paths and forced includes and defines specified in the config file.

What would be useful here is to be able to generate the compile_commands.json file for all the other Zephyr RTOS related files.  That way, vscode would be able to open and parse any of the OS files (drivers, kernel, subsystems, modules, etc).  At the moment, I have to rely on the Include paths and forced includes and defines in my config.  These could be eliminated all together if I could get the compile_commands.json for all the components of Zephyr.

sslupsky, 2020-03-14:
Here is an example of a task I use to invoke west to do a build:

```
{
	"label": "west build auto",
	"type": "shell",
	"group": "build",
	"command": "west",
	"args": [
	    "build",
	    "-p",
	    "auto",
	    "-b",
	    "scanimetrics_witap_isb",
	    "${workspaceFolder}"
	],
	"options": {
	    "cwd": "${workspaceFolder}",
	    "env": {
		"ZEPHYR_BASE": "/Users/stevenslupsky/Documents/source/zephyrproject/zephyr",
		"ZEPHYR_TOOLCHAIN_VARIANT": "gnuarmemb",
		"GNUARMEMB_TOOLCHAIN_PATH": "/usr/local/",
		"VS_CODE_BOARD_TO_BUILD": "scanimetrics_witap_isb",
		"VS_CODE_ZEPHYR_BIN_NAME": "zephyr.elf",
	    }
	},
	"dependsOn": [],
	"problemMatcher": [ "$gcc" ]
}
```

sslupsky, 2020-03-16:
I found another way to integrate Zephyr with vscode using the CMake-tools extension.  I had been playing around with this extension a while back but really did not understand how to use it.  The key to use it for Zephyr is the "configurationProvider" property in c_cpp_properties.json.

Install the vector-of-bool (now microsoft) Cmake-tools extension.  Then, edit your c_cpp_properties.json in your project's .vscode directory and add the following configuration:

```
{
	"name": "Cmake",
	"configurationProvider": "vector-of-bool.cmake-tools"
}
```

Select the "Cmake" configuration that was added by clicking the configuration button in the bottom right portion of the status bar beside the "Tweet Feedback" button (it is typically set to "Mac" or "Win").  When you select the "Cmake" configuration (or whatever name you gave it), this automatically finds compile_commands.json according to the CMake extension settings (you can change the default if you need to).

To get the Zephyr components to parse properly, edit the CMakeLists.txt files in each of the Zephyr directories that contains source files you want to be able to open with your project.  Add the following line to each file:
```
set(CMAKE_EXPORT_COMPILE_COMMANDS ON)
```

In my case, I edited CMakeLists.txt in the following directories:
arch/arm
boards
drivers
kernel
soc
subsys

Build your project.  Check your compile_commands.json file and now you will see a number of additional entries for all the Zephyr components.

Now, when you open your Zephyr source files, they should be parsed properly by the c_cpp extension.

sslupsky, 2020-03-16:
Some other useful extensions:

ARM, Dan Underwood
CMake Tools, Microsoft
Cortex-Debug, marus25
DeviceTree, Pietro Lorefice
Doxygen Documentation Generator, Christoph Schlosser
Git Graph, mhutchie
Intel HEX format, keroc
kconfig, luveti

sslupsky, 2020-03-19:
There is one problem with using the compile_commands.json workflow with VSCode.  When you apply the CMAKE_EXPORT_COMPILE_COMMANDS setting, it doesn't seem to propagate to all the required files.

For instance, sam0_rtc_timer.c does not appear in my compile_commands.json.  I have not determined why it is not being included.

joerchan, 2020-03-20:
@SebastianBoe Any idea why the compile commands would not be consistent with the ninja build file?

SebastianBoe, 2020-03-20:
I couldn't say why, but it seems that setting the variable from CMakeLists.txt is unofficially not supported. I would try setting it from CLI to ensure it is always set as a cache variable.

https://gitlab.kitware.com/cmake/cmake/issues/16588

joerchan, 2020-03-20:
@sslupsky Given the answer from Sebastian, try setting the compile command as was explained here:
https://github.com/zephyrproject-rtos/zephyr/issues/21119#issuecomment-578282316

beriberikix, 2020-03-20:
Awesome discussion all around! I can't help but thinking that we can capture this with some automation via a [new VS Code extension](https://code.visualstudio.com/api). If some of the scaffolding and configuration was implemented in `west` (like task generation) and the extension wrapped `west`, installed other extensions as deps - you could have a very nice developer experience out of the box. We did something similar when I was at [Particle](https://marketplace.visualstudio.com/items?itemName=particle.particle-vscode-core). Anyone with JS experience interested in collaborating? I know their APIs pretty well but I'm much more a Product guy than a good developer :)

sslupsky, 2020-03-20:
@joerchan Thank you for the tip.  I can confirm that setting the CMAKE_EXPORT_COMPILE_COMMANDS from the command line works better than using the setting in the CMakeLists.txt file.  Now all the zephyr components are included in the compile_commands.json file.  Hooray!  :-)

One problem remains though.  Some of the include paths do not resolve properly.  For instance, #include <soc.h> comes up with a list of 76 definitions.  This prevents anything that includes soc.h from linting properly with intellisense.  Unfortunately, soc.h is quite a popular file to include ;-).

I have raised an issue with the VSCode team about this problem since it appears to be specific to VSCode at the moment:
https://github.com/microsoft/vscode-cmake-tools/issues/1120

Has anyone here been able to resolve this?

sslupsky, 2020-03-20:
For clarification for anyone reading this, I am using the Multi-root workspace in VSCode.  This allows me to configure VSCode to work with projects that have multiple "sub projects".  In my use case, I have root folders for:
application (project)
custom board
zephyr tree (base)
zephyr modules

The multi root capability of VSCode is really quite helpful for Zephyr projects.  Here is a copy of the .code-workspace file:
```
{
	"folders": [
		{
			"path": "isb",
			"name": "app"
		},
		{
			"path": "zephyrproject/zephyr/boards/arm/scanimetrics_witap_isb",
			"name": "board"
		},
		{
			"path": "zephyrproject/zephyr",
			"name": "zephyr"
		},
		{
			"path": "zephyrproject/modules",
			"name": "modules"
		}
	],
	"settings": {
		"C_Cpp.default.includePath": [
			"${workspaceFolder:zephyr}/include",
			"${workspaceFolder:zephyr}/include/drivers",
			"${workspaceFolder:zephyr}/soc/arm/atmel_sam0/samd21",
			"${workspaceFolder:zephyr}/soc/arm/atmel_sam0/common/.",
			"${workspaceFolder:zephyr}/drivers",
			"${workspaceFolder:zephyr}/ext/lib/fnmatch/.",
			"${workspaceFolder:zephyr}/ext/hal/cmsis/Core/Include",
			"${workspaceFolder:zephyr}/subsys/usb",
			"${workspaceFolder:modules}/hal/atmel/asf/sam0/include/samd21",
			"${workspaceFolder:modules}/fs/littlefs",
			"${workspaceFolder:modules}/fs/fatfs/include",
			"${workspaceFolder:modules}/fs/nffs/include/nfss",
			"${workspaceFolder:zephyr}/lib/libc/minimal/include",
			"/usr/local/Cellar/arm-none-eabi-gcc/8-2018-q4-major/gcc/bin/../lib/gcc/arm-none-eabi/8.2.1/include",
			"/usr/local/Cellar/arm-none-eabi-gcc/8-2018-q4-major/gcc/bin/../lib/gcc/arm-none-eabi/8.2.1/include-fixed",
			"${workspaceFolder:app}/build/zephyr/include/generated"
		],
		"C_Cpp.default.forcedInclude": [
			"${workspaceFolder:app}/build/zephyr/include/generated/autoconf.h",
			"${workspaceFolder:zephyr}/include/toolchain/zephyr_stdint.h",
			"${workspaceFolder:app}/.vscode/c_cpp_undef.h"
		],
		"C_Cpp.default.defines": [
			"BUILD_VERSION=v2.1.0-rc1-141-g3cc5bda2fa10",
			"KERNEL",
			"_FORTIFY_SOURCE=2",
			"__PROGRAM_START",
			"__ZEPHYR__=1"
		],
		"C_Cpp.default.compilerPath": "/usr/local/bin/arm-none-eabi-gcc",
		"C_Cpp.default.cStandard": "c11",
		"C_Cpp.default.cppStandard": "c++17",
		"C_Cpp.default.intelliSenseMode": "gcc-x64",
		"C_Cpp.default.compileCommands": "${workspaceFolder:app}/build/compile_commands.json",
		"files.associations": {
			"zephyr.h": "c",
			"kernel.h": "c",
			"kernel_includes.h": "c",
			"log.h": "c",
			"printk.h": "c",
			"littlefs.h": "c",
			"i2c.h": "c",
			"usb_device.h": "c",
			"soc.h": "c",
			"samd21g18a.h": "c",
			"power.h": "c",
			"watchdog.h": "c",
			"i2c_ll_stm32.h": "c",
			"init.h": "c",
			"device.h": "c",
			"cpu.h": "c",
			"generated_dts_board.h": "c",
			"system_timer.h": "c",
			"clock_control.h": "c",
			"sys_clock.h": "c",
			"dma.h": "c",
			"hwinfo.h": "c",
			"gpio.h": "c",
			"errno.h": "c",
			"stdio.h": "c",
			"version.h": "c"
		},
		"terminal.integrated.env.osx": {
			"ZEPHYR_BASE": "${workspaceFolder:zephyr}/",
			"ZEPHYR_TOOLCHAIN_VARIANT": "gnuarmemb",
			"GNUARMEMB_TOOLCHAIN_PATH": "/usr/local/",
			"PATH": "$ZEPHYR_BASE/scripts:${env:PATH}"
		},
		"editor.tabSize": 8,
		"editor.insertSpaces": false
	}
}
```

sslupsky, 2020-03-20:
@SebastianBoe Thank you, the command line option resolved my issue with the compile_commands.json file not including all the zephyr targets.

sslupsky, 2020-03-20:
@beriberikix That is a great idea ... that would simplify the setup process enormously by eliminating the need for someone to learn about the details of configuring VSCode.  I have grown to admire VSCode tremendously and it keeps getting better.  I have one nagging issue related to git that I haven't been able to figure out.  Seems like VSCode will occasionally max out the CPU in one of its' threads.  The problem seems related to staging selected code changes but I am not completely certain that is the cause; rather it is likely just a coincidence.  VSCode has a helpful command when this happens, "Developer: reload window".  This is a lifesaver.  You loose your undo history but pretty much everything else comes back to where you were minus the laggy responsiveness due to the high cpu usage.

trond-snekvik, 2020-08-12:
I published the [Kconfig extension](https://marketplace.visualstudio.com/items?itemName=trond-snekvik.kconfig-lang) on the marketplace last week. It provides full language support for all Kconfig and .conf files out of the box, including code completion, go to definition, syntax highlighting and linting. It should improve the VS Code experience for Kconfig quite a bit for both application developers and stack maintainers.

The extension depends on West v0.7.0 or newer. Older versions should work, but there may be some manual configuration required. Feel free to contact me on slack if you have any questions, issues or feature requests :)

rosterloh, 2020-08-12:
@trond-snekvik Thanks for this! I've been using it for the last few days and it's great. Hoping to see more zephyr support for VSCode in the future.

aunsbjerg, 2020-09-10:
Just stumbled on this document by segger on how to integrate J-link debugging with vscode. It's not zephyr specific, but the general setup is applicable to zephyr as well, so posting here for reference.

https://wiki.segger.com/J-Link_Visual_Studio_Code

trond-snekvik, 2020-10-05:
I have published the [Devicetree extension](https://marketplace.visualstudio.com/items?itemName=trond-snekvik.devicetree) now as well. 

Before publishing, I rewrote most of it to provide full language support for all Devicetree files, and process the C preprocessor symbols, allowing overlays, board files and shields to be included in the processing. It has quite a bit more features than the kconfig extension, but should be less intrusive.

Again, feel free to contact me for feature requests and bug reports ☺️

beriberikix, 2020-10-05:
Someone should create a Zephyr [extension pack](https://code.visualstudio.com/api/references/extension-manifest#extension-packs)! Perhaps it can be officially managed by a maintainer :)

func4plus1, 2020-11-03:
> Someone should create a Zephyr [extension pack](https://code.visualstudio.com/api/references/extension-manifest#extension-packs)! Perhaps it can be officially managed by a maintainer :)

should def do this - so those of us in vim can benefit from it in Coq.

manoj153, 2021-01-11:
> I've had some good luck using VS Code and its support for Language Server to do intellisense stuffs by having CMake generate a compilation database (that VS Code then consumes automatically, assuming the build directory is in the project).
> 
> The CMake parameter is `CMAKE_EXPORT_COMPILE_COMMANDS`, and here's examples of building an app:
> 
> ```
> west build -b <board> <app directory> -- -DCMAKE_EXPORT_COMPILE_COMMANDS=1
> ```
> 
> If you always want to generate the compilation database, you can also do:
> 
> ```
> west config build.cmake-args -- -DCMAKE_EXPORT_COMPILE_COMMANDS=ON
> ```

Hi does this automatically works if I have /build dir on my vs code workspace? 

any further config I need to do to make the IntelliSense stuff works?

manoj153, 2021-01-19:
Hi, v0.12.0 SDK open-ocd has RTOS awareness, verified by https://github.com/zephyrproject-rtos/openocd/issues/42#issue-788565903
```
The following RTOS values are supported by OpenOCD: eCos ThreadX FreeRTOS ChibiOS embKernel mqx uCOS-III auto
```
so what's the way to inform cortex-debug about zephyr awareness?

marnold-b, 2021-07-27:
I just watched the discussion about [Zephyr Developer Environments](https://www.youtube.com/watch?v=QQ4cgkAZs1E).
In the discussion "Docker with Remote Plugin for VScode" was mentioned, that is what I use to setup
my development environment on Windows for zephyr.
I would like to share that setup.

Requirements:
- Windows with WSL2 installed
- Docker with WSL2 backend
- Pyocd on Windows
- Ozone installed on Windows

I start the container always from WSL2 (for example Ubuntu Distro),
because if I start it from a Windows path directly, I experienced a
significantly performance drop.

My config files for vscode:

**.devcontainer/devcontainer.json**

```json
{
	"name": "Zephyr development environment",
    // I use a specific tag for the docker image, that vscode can detect an update from docker image
	"image": "docker.io/zephyrprojectrtos/zephyr-build:v0.17.5",
    // It is pretty nice that you can define the plugins you want 
    // and it will be automaticly installed when you open the project.
	"extensions": [
		"ms-python.python",
		"ms-vscode.cpptools",
		"twxs.cmake",
		"marus25.cortex-debug",
		"trond-snekvik.devicetree",
		"trond-snekvik.kconfig-lang",
		"editorconfig.editorconfig",
		"jebbs.plantuml",
		"pkief.material-icon-theme",
		"eamodio.gitlens"
	],
	"remoteUser": "user",
	"remoteEnv": {"ZEPHYR_BASE":"/home/user/workspace/zephyrproject/zephyr","SHELL":"/bin/bash"},
	"workspaceMount": "source=${localWorkspaceFolder},target=/home/user/workspace/zephyrproject,type=bind,consistency=delegated",
	"workspaceFolder": "/home/user/workspace/zephyrproject/",
}
```

**.vscode/c_cpp_properties.json**

```json
{
    "configurations": [
        {
            "name": "Linux",
            "includePath": [
                "${workspaceFolder}/**"
            ],
            "defines": [],
            "compilerPath": "/usr/bin/gcc",
            "cStandard": "c11",
            "cppStandard": "c++14",
            "intelliSenseMode": "gcc-x86",
            "compileCommands": "${workspaceFolder}/build/compile_commands.json"
        }
    ],
    "version": 4
}
```

**.vscode/launch.json**

```json
[... 222 lines cut ...]
	"cortex-debug.armToolchainPrefix": "arm-zephyr-eabi",
	"cortex-debug.armToolchainPath": "/opt/toolchains/zephyr-sdk-0.12.4/arm-zephyr-eabi/bin/",
}
```

But for me the best solution so far was to debug cortex-m targets with Segger Ozone.
It has a bit more embedded specific features compared with vscode and it is easier to setup.

Ozone has the possibility to redirect the path in the elf file to a windows path in your project configuration:

**.ozone/debug.jdebug**
```
Project.AddPathSubstitute ("//wsl$/Ubuntu-20.04/home/%USERNAME%/workspace/zephyrproject/.ozone", "$(ProjectDir)");`
Project.AddPathSubstitute ("/home/user","//wsl$/ubuntu-20.04/home/%USERNAME%");
```

With this configuration Ozone on windows works just fine with the in the container built elf.

Segger has a similar feature like pyocd server for their [jlink](https://wiki.segger.com/J-Link_Remote_Server), which I have not tested yet.

It is also possible to share the credentials for git into the docker container.
Further information for that:
https://code.visualstudio.com/docs/remote/containers#_sharing-git-credentials-with-your-container

What I also really like is the same docker image can be used for ci.

carlescufi, 2021-08-30:
There's a difference in concept here:

- Zephyr does not support "in-tree" builds means that you cannot build and generate the object files scattered around in the corresponding source directories, all the build files need to be under one common `build/` (or whatever name you choose) folder
- nRF Connect SDK using "out-of-tree" code means that it adds functionality to Zephyr in a separate Git repository by using west and its extension mechanism

wbober, 2021-08-31:
Just to let you know that Nordic has released a preview of an extension bundle for vscode. It covers a number of features discussed in this thread.

More details in the blog post:
https://devzone.nordicsemi.com/nordic/nordic-blog/b/blog/posts/nrf-connect-for-visual-studio-code-preview

There is also a few videos that demonstrate the usage:
https://www.youtube.com/playlist?list=PLx_tBuQ_KSqEt7NK-H7Lu78lT2OijwIMl

hongshui3000, 2021-09-01:
@wbober This plug-in seems to be limited to nrf, not a general plug-in for zephyr? Is the plug-in implementation open source?

wbober, 2021-09-01:
@hongshui3000 The plugin is not limited to nrf. Our SDK is based on Zephyr and you can work with any upstream sample or hardware platform without limitations. We have not made a decision to open-source it yet. Is there any particular functionality that you would like to change or expand?

hongshui3000, 2021-09-02:
> @hongshui3000 The plugin is not limited to nrf. Our SDK is based on Zephyr and you can work with any upstream sample ~or hardware platform without limitations~. Some functions (like creating a new board) are tailored to Nordic SoC but majority should work for other hardware platforms. We have not made a decision to open-source it yet. Is there any particular functionality that you would like to change or expand?

I hope there are some interfaces that can connect (call) commercial debuggers (keil mdk or DS5)

beriberikix, 2021-09-02:
Reading the post it sounds like this extension is leveraging [Cortex-Debug](https://github.com/Marus/cortex-debug) for the debugging support (which many of us use!) It currently doesn't support commercial debuggers but since it is open source anyone could contribute to it. Perhaps that Nordic team might even do so ;)

hongshui3000, 2021-09-02:
> Reading the post it sounds like this extension is leveraging [Cortex-Debug](https://github.com/Marus/cortex-debug) for the debugging support (which many of us use!) It currently doesn't support [...]

The serial terminal seems to have no input function, I cannot send data to MCU.
It would be better if people can customize some trace functions by themselves. I hope to parse some log trace information according to my own rules.

hongshui3000, 2021-09-02:
I also look forward to open source implementation, @3Nigma

wbober, 2021-09-02:
> > Reading the post it sounds like this extension is leveraging [Cortex-Debug](https://github.com/Marus/cortex-debug) for the debugging support (which many of us use!) 

Correct, the extension is using cortex-debug. Our extension simply provides a context-aware configuration to cortex-debug, so you don't have to configure it manually.

> The serial terminal seems to have no input function, I cannot send data to MCU.
> It would be better if people can customize some trace functions by themselves. I hope to parse some log trace information according to my own rules.

Which terminal you refer to? The nRF Terminal extension that is bundled does support input for both serial and RTT. I've seen requests to add customization to log parsing before. What do you think about a list of regex rules that would filter out log input?

hongshui3000, 2021-09-02:
Serial terminal

> > > Reading the post it sounds like this extension is leveraging [Cortex-Debug](https://github.com/Marus/cortex-debug) for the debugging support (which many of us use!)
> 
> Correct, the extension is using cortex-debug. Our extension simply provides a context-aware configuration to cortex-debug, so you don't have to configure it manually.
> 
> > The serial terminal seems to have no input function, I cannot send data to MCU.
> > It would be better if people can customize some trace functions by themselves. I hope to parse some log trace information according to my own rules.
> 
> Which terminal you refer to? The nRF Terminal extension that is bundled does support input for both serial and RTT. I've seen requests to add customization to log parsing before. What do you think about a list of regex rules that would filter out log output?

Serial terminal。It is best to customize the parsing script outside the plug-in, so that I can save the log file in the way I want. For example, I can use log to generate a test report, etc.

In addition, it is best to support serial shell terminal, interactive debugging is also very useful

hongshui3000, 2021-09-02:
Another issue seems to be that after installing the plug-in under Windows, running west under the command line under vscode does not work.I still suggest that there is a place to raise an issue, even if the project is not open source . :)

wbober, 2021-09-09:
@hongshui3000 Yes, that was supposed to be a feature but turned out to be an issue that has hit users hard. We already have a workaround: if you use a manual toolchain installation we won't override the terminal environment. For a more complete solution we would like to introduce a separate terminal profile for NCS and leave the default one unmodified.

Feel free to open issues on https://devzone.nordicsemi.com/

carlescufi, 2021-09-09:
https://www.rajeevpiyare.com/posts/nrfconnect-vs-code/

hongshui3000, 2021-11-12:
espressif has an open source implementation, but it does not seem to be based on zephyr 
https://github.com/espressif/vscode-esp-idf-extension

marnold-b, 2021-11-12:
Might be interesting for people who are using WSL2 with vscode:
https://devblogs.microsoft.com/commandline/connecting-usb-devices-to-wsl/

ddavidebor, 2022-02-05:
Would the Nordic folks be so kind as to open-source their VSCode extensions?

wbober, 2022-02-05:
@ddavidebor Does the closed source nature of the extensions limits your usage in any way?

ddavidebor, 2022-02-05:
@wbober not for Nordic products, but of course the extensions could be expanded to encompass a larger number of toolchains over time - acting as the starting point for full VSCode integration

I totally understand if that's not viable for business reasons.

hongshui3000, 2022-02-06:
I wish there was a zephyr community maintained vscode extension

wbober, 2022-02-07:
@ddavidebor There is no plan to open source the extension. 

We are open to providing a *limited* support for development on other platforms. If you are able to provide more concrete examples of things that are missing I will take that feedback to the team. Perhaps there are ways to support with opening some APIs for extension with plugins or alike. 

Can't make any promises but I'm open for discussion.

hongshui3000, 2022-02-20:
import vsc 
https://github.com/smrtos/Zephyr2VSC

mbrunnen, 2023-02-08:
I still see this issue: https://github.com/microsoft/vscode-cpptools/issues/9187

E.g. whenever I use `LOG_INF("Hello World")` with only one argument, I get:
> too few arguments in invocation of macro "Z_FOR_LOOP_1"

trond-snekvik, 2023-02-08:
@mbrunnen you can get around this by adding `__CDT_PARSER__` to your list of defines for CppTools. It's not a proper fix, but at least it makes the C++ extension usable.

aunsbjerg, 2023-02-08:
> @mbrunnen you can get around this by adding `__CDT_PARSER__` to your list of defines for CppTools. It's not a proper fix, but at least it makes the C++ extension usable.

@trond-snekvik, I just tried that since I have the same issue, but I does not seem to have any effect, I still get the "too few arguments" warnings.

My configuration looks something like this:

```json
{
    "name": "app",
    "cStandard": "c11",
    "cppStandard": "c++20",
    "intelliSenseMode": "linux-gcc-arm",
    "compilerPath": "${env:zephyrCompilerPath}",
    "compileCommands": "${workspaceFolder}/build/compile_commands.json",
    "defines": [
        "__CDT_PARSER__"
    ],
    "forcedInclude": [
        "${workspaceFolder}/build/autoconf.h"
    ]
},
```

[10 more replies]