zephyrproject-rtos/zephyr #21121: netusb: RNDIS host support

Jeepgoing, 2019-12-03:
Hi, i am testing Zephyr connect to Android Phone with Type-C, i found Android Phone only supports RNDIS Device mode, so it that mean Zephyr device should support RNDIS Host?

tuobaye1988, 2019-12-04:
Hi， I have same question

carlescufi, 2019-12-04:
@Jeepgoing Zephyr only supports the USB Device role, not USB Host.

Jeepgoing, 2019-12-05:
@carlescufi @finikorg I think RNDIS have Host and Decice itself , is it possible RNDIS host run on USB device?

finikorg, 2019-12-11:
@Jeepgoing Do you have Android phone with OTG?

Jeepgoing, 2019-12-12:
yes @finikorg  It's have OTG function

finikorg, 2019-12-12:
> yes @finikorg It's have OTG function

Then Zephyr is USB device and RNDIS should work

Jeepgoing, 2019-12-12:
@finikorg But  Android phone need DHCP by Zephyr, I can't find DHCP server support

jukkar, 2019-12-12:
If Zephyr is compiled with auto IPv4 support `CONFIG_NET_IPV4_AUTO=y`, the system should start using 169.254.xxx.xxx addresses so there should be no need for DHCP server support.

Jeepgoing, 2019-12-12:
@jukkar you mean Zephyr Device and Android Phone all use default IP?

jukkar, 2019-12-12:
> you mean Zephyr Device and Android Phone all use default IP?

No, I mean that if devices are configured to use auto IPv4 addresses, then they will select suitable IPv4 addresses from 169.254/16 block in order to communicate with each other. Linux typically has this support automatically enabled but in zephyr you need to manually enable this.

finikorg, 2019-12-23:
Closing per comments

closed 2019-12-23 by finikorg