zephyrproject-rtos/zephyr #21136: ARC: Add support for reduced register file

abrodkin, 2019-12-03:
ARC cores could be configured with reduced set of core registers and in the deeply embedded world that's pretty typical configuration.

Given GCC for ARC supports such a configuration via `-mrf16` option we just need to accommodate quite a few missing registers in hand-written assembly.

It's not that we're going to do it with the highest priority right now but whenever we have an empty time-slot or real use-case this could be done then.

YuguoWH, 2020-10-15:
Please note that the ARC secureshield implement uses some registers (i.e. r4~r6) that are not available with `-mrf16` option. Need to take this into consideration if we adopt `-mrf16` option in the future.

ruuddw, 2021-05-18:
Closing this one - no current plans to support ARC reduced register file for Zephyr (when using an RTOS, having 32 registers makes more sense).

closed 2021-05-18 by ruuddw