zephyrproject-rtos/zephyr #21139: west: runners: Keyboard Interrupt shouldn't kill the process

SuperPrower, 2019-12-03:
**Describe the bug**
Currently, when using `west {debug,attach}` (at least with blackmagicprobe runner), pressing Ctrl-C will kill the runner process and child process. However, this is not be desirable when using GDB or other utilities that catch and use interrupt as a part of debugging process. 

**To Reproduce**
```
west attach -r blackmagicprobe 
(gdb) run
[press Ctrl-C to pause the execution]
Starting program: /.../build/zephyr/zephyr.elf
^C
Program received signal SIGINT, Interrupt.
arch_cpu_idle () at /.../zephyrproject/zephyr/arch/arm/core/cpu_idle.S:78
78              bx lr
user@linux:~$ # gdb was killed
```

**Expected behavior**
Keyboard Interrupt should be passed to the child process instead of stopping runner and killing child process as a result. i.e. GDB  should normally break execution of the running program.

**Impact**
It's impossible to properly debug hardware using runner that uses, for example, GDB, thus one is forced to run and configure it manually to be able to use interrupts to break execution.

**Screenshots or console output**
see To Reproduce

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: west & blackmagicprobe runner
 West version: v0.6.1
zephyr commit  0a925bdf9c574bf57940ff41354806f851b6ccfe

**Additional context**

mbolivar, 2019-12-03:
@SuperPrower thanks for the detailed report.

> Currently, when using `west {debug,attach}` (at least with blackmagicprobe runner),

This is only a problem with the blackmagicprobe runner, so I've renamed the issue title.

Root cause details:

The other runners tend to use run_server_and_client() to properly handle this, since they start a GDB server and connect to it with a client. The BMP USB device presents itself as a serial device which speaks the GDB serial protocol instead, so there's no server/client, and thus no call to run_server_and_client().

The problem is that blackmagicprobe essentially uses subprocess.check_call() to start GDB directly, without ignoring SIGINT in the python process. Easy fix.

@SuperPrower can you please test https://github.com/zephyrproject-rtos/zephyr/pull/21149 and confirm the fix? I have a BMP somewhere in one of my boxes, but I don't use it regularly, so I'd appreciate your help.

SuperPrower, 2019-12-03:
@mbolivar thanks for explaining. I am unfamiliar with other flashing/debugging methods and tried to make no assumptions in the issue description.

Yes, the fix works for me, thanks. I can Ctrl-C the execution without killing the GDB.

mbolivar, 2019-12-03:
Your report was excellent. Thanks again, and thanks for testing!

PR #21149 (2019-12-03): scripts: runners: fix blackmagicprobe SIGINT behavior
Fixes: #21139
commit 2024fb53 (2019-12-03): scripts: runners: fix blackmagicprobe SIGINT behavior

The blackmagicprobe runner's Python process fails to ignore SIGINT
when it runs GDB from the debug and flash callbacks, which is wrong.

The other runners tend to use run_server_and_client() to properly
handle this, since they start a GDB server and connect to it with a
client. The BMP USB device presents itself as a serial device which
speaks the GDB serial protocol instead, so there's no server/client,
and thus no call to run_server_and_client().

The problem is that blackmagicprobe essentially uses
subprocess.check_call() to start GDB d [... cut]
closed 2019-12-03 by dleach02